dresden-elektronik/deconz-rest-plugin #1167: [request device support] Plugwise smart plug (not circle) Zigbee HA Qupit model 160-01

herwarth, 2019-01-21:
Hello,
Plugwise has created a new smart plug which is now Zigbee HA compatible. It is not the Plugwise Circle but the successor of it (although they look the same).
I have added it to deCONZ and I can read the power usage. It also measures several other power related stuff like received power (solar panels?).
The requested info:
[image]
[image]

herwarth, 2019-02-22:
adding additional screenshots
[image]
[image]
[image]
[image]

herwarth, 2019-02-22:
simple on/off switching works, but power metering is a must. please help me out, because i am waiting for this to be implemented

kasteleman, 2019-02-22:
Keep in mind the plug from time to time reports -2 Watt although nothing
connected.

alexdepalex, 2020-01-20:
Does this plug properly measure power? Or is there still some work that needs to be done?

alexdepalex, 2020-02-13:
Do we need any additional information to get this forward?

Mimiix, 2020-06-05:
@alexdepalex is this still open?

alexdepalex, 2020-06-16:
Yeah. I think it is. The plug works, but HA detects it as a light, so it doesn't expose for example power measurement. And that's something that I need. Not many switches that support power measurement and 16A.

Mimiix, 2020-06-16:
Can you provide the debug logs of the addon in home-assistant? ? Info on how to enable that are on the info page of the addon. 

If you are in discord, use the .hadebug command and it will show you.

SwoopX, 2020-06-16:
Somebody just needs to read the electrical measurement and simple metering clusters attributes while a load is allied where you have an estimate of how much that is (e.g. 15W charger for mobiles). That helps.

alexdepalex, 2020-06-16:
8W led bulb
[image]
40W filament bulb
[image]
100W filament bulb
[image]

alexdepalex, 2020-06-16:
I do see that sometimes negative values are reported for `Total Active Power`.

SwoopX, 2020-06-16:
Why have you cut off the 500-series attributes? Usually, this is where the misuc plays for that cluster.

The other cluster is missing as well! ;)

alexdepalex, 2020-06-16:
[image]

Which cluster are you missing?

SwoopX, 2020-06-16:
Simple metering. Current summation and instantaneous demand is of interest.

alexdepalex, 2020-06-16:
Seems like they're not being used.

[image]

SwoopX, 2020-06-18:
Closing this as PR has been merged. Supported with .78

closed 2020-06-18 by SwoopX