dresden-elektronik/deconz-rest-plugin #1169: HomeAssistant randomly lost connection to Conbee Gateway.

Shaolo, 2019-01-22:
I'm not sure if this is a deCONZ issue or a Home Assistant issue. I suspect it might be a combination of the two. For some reason in the middle of the night after restarting the raspberry PI housing both the two will no longer communicate. The conbee comes up fine, I can see the devices it controls via Phoscon. It will not communicate with the HomeAssistant app though.

I'm running HassIO 0.85.1 (same HA revision). 

Hoping someone can help me root cause this before I blow it away and restart. Really don't want to see this happen again in the future.

rich710se, 2019-01-23:
I have the same issue, every time I restart my Nuc with Ubuntu, HA and Deconz the integration in HA for deconz stop working. The Entities show up in integrations, but says "not reachable".
I thought maybe it was because deconz goes up after HA, but even if I restart HA it doesn't connect to deconz. Every time I have to delete the inegration, open the gateway in Phoscson and add a new integration in HA. Then restart HA again, and all entities from Deconz is back and working.
Don't know if it's because of Deconz or HA..

Kane610, 2019-01-23:
Does the Ipaddress change on restart? That would cause this issue

Kane610, 2019-01-23:
People had issues a few months ago. I then fixed it by adding a retry on startup which solved it then. What the reason should be now is hard to say

Shaolo, 2019-01-23:
No, everything is static. I removed the integration, added it back. Kept getting an error about the integration already existing (after removing it and while trying to add it back), gave up. Rebooted the pi four or five times working on other things and eventually it started communicating on it's own at random.

I've found the Conbee to be unreliable at best though. Half of my lights are randomly phasing out of the network, half don't change color and half don't change brightness. None of the lights are more than 50ft from the Conbee. Even leaving them all on so they can route to each other they randomly drop out. Latest firmware seems to have actually made things worse. Getting an unpaired light to pair is difficult too. Even with the light literally 12 inches from the Conbee.

I'm about to go back to the Tradfri gateway. Which I really didn't want to do.

Kane610, 2019-01-23:
Have you added a USB extension cord between conbee and computer?

I am sad to hear about your experience since it's rock solid for me

Shaolo, 2019-01-23:
There is one showing up today. 

The Conbee isn't stuffed behind a computer though. It's on a PI B3+ sitting on an open desk.

Can I ask what lights you use? From what I've read in the issues section Ikea lights are not well supported and there is no intent behind changing that.

Kane610, 2019-01-23:
Mainly hue lights

tubalainen, 2019-01-28:
I have the same problem.

I am running a docker setup under Debian. I do believe it is the docker IP of the deconz instance that changes for some reason. I have to remove the "Integration" in Home Assistant and add it again to get it going.

Kane610, 2019-01-28:
I don't think the IP the component uses is the docker IP, rather the host IP. Can you verify this by looking in .storage/core.config_entries?

tubalainen, 2019-01-28:
> I don't think the IP the component uses is the docker IP, rather the host IP. Can you verify this by looking in .storage/core.config_entries?

When discovered by the `discovery` component of HASS it presents the 172.x ip to HASS. I have personally edited the IP in the core.config_entites to the host address to mitigate this problem.

tubalainen, 2019-01-28:
> I will add a print when connection fails that will show what IP it tries to connect to

Great. It would be very interesting to see where we differ here... 

For reference.
I run the latest Debian 9 with docker-ce .. Docker is installed with the recommended settings from dockers home page.

This is my deconz docker string (i do not use compose):
`docker run -d --name=deconz -p 8882:80 -p 4434:4434 -p 5900:5900 -v /home/my_user/deconz:/root/.local/share/dresden-elektronik/deCONZ -v /etc/localtime:/etc/localtime:ro --device=/dev/ttyUSB0 -e DECONZ_WEB_PORT=80 -e DECONZ_WS_PORT=4434 -e DECONZ_VNC_MODE=1 -e DECONZ_VNC_DISPLAY=0 -e DECONZ_VNC_PASSWORD=supersecret -e DECONZ_VNC_PORT=5900 marthoc/deconz:latest`

This is my hass docker string:
`docker run -d --name="hass_new" -v /home/my_user/haconfig_new:/config -v /etc/localtime:/etc/localtime:ro --device=/dev/ttyACM0:/zwaveusbstick:rwm --device=/dev/ttyUSB1 --net=host homeassistant/home-assistant:latest`

Noteworthy is that I use the `--net=host` option for hass and not for deconz (if that can be the root cause of the IP discovery??)

Ritte88, 2019-06-02:
Is this solved for everybody? I still have the problem. What I notice is that I can’t log into HA when having the problem. HA cant connect to conbee. But I can access conbee through the webgui.

Kane610, 2019-06-02:
I added functionality to Deconz integration to update the IP if it gets discovered

Caboz, 2019-06-04:
Do one need to do anything to add the functionality to the integration?

My Home Assistant is not able to display information from my sensors that are connected to Deconz GW since I restarted the Deconz GW. 

If I restart HA after Deconz is up and running everything works as it should but there is something wrong with the integration.

Kane610, 2019-06-04:
@Caboz make sure that you have ssd component running after upgrading to hass 0.94

Caboz, 2019-06-04:
I'm not certain that I understand what you mean.

I have two Raspberry Pie devices, one with HA and one with Deconz/Phoscon.

I use the Hass.io to run HA updated to 0.93.2.

I have the integration configured in Integrations. 

Is there anything I'm missing?

Kane610, 2019-06-04:
@Caboz the issue you have is probably that deconz has got a different network address on restart.

With 0.94 there are improvements to how discovery is made, this can be enabled by having ssdp integration specified in you configuration.yaml

SSDP will discovery deconz on its new address and update integration with it

Caboz, 2019-06-04:
Thanks, 

The IP dress is the same, as it is fixed at DHCP level on the router.

I'll try 0.94

Kane610, 2019-06-04:
@Caboz ok, then the issue could be a different one. Next step would be to enable debug for deconz component and look at the logs

Caboz, 2019-06-18:
The problem is still there...
If I restart the Deconz/Phoscon, Home Assistant must be restarted to find it again.
HA then recognises Deconz as a new gateway despite the IP being the same.
I'm getting a little frustrated with this, everything else works perfectly.

Kane610, 2019-06-18:
Can you enable debug for ssdp and deconz?

Caboz, 2019-06-18:
`2019-06-18 12:33:23 INFO (MainThread) [homeassistant.setup] Setup of domain logger took 0.0 seconds.
2019-06-18 12:33:23 INFO (MainThread) [homeassistant.setup] Setting up http
2019-06-18 12:33:23 INFO (MainThread) [homeassistant.setup] Setup of domain http took 0.0 seconds.
2019-06-18 12:33:23 INFO (MainThread) [homeassistant.setup] Setting up system_log
2019-06-18 12:33:23 INFO (MainThread) [homeassistant.setup] Setup of domain system_log took 0.0 seconds.
2019-06-18 12:33:24 INFO (MainThread) [homeassistant.setup] Setting up recorder
2019-06-18 12:33:26 INFO (MainThread) [homeassistant.setup] Setup of domain recorder took 2.4 seconds.
2019-06-18 12:33:26 INFO (MainThread) [homeassistant.setup] Setting up updater
2019-06-18 12:33:27 INFO (MainThread) [homeassistant.setup] Setting up xiaomi_aqara
2019-06-18 12:33:27 INFO (MainThread) [homeassistant.setup] Setting up homekit
2019-06-18 12:33:27 INFO (MainThread) [homeassistant.setup] Setting up deconz
2019-06-18 12:33:27 INFO (MainThread) [homeassistant.setup] Setup of domain deconz took 0.0 seconds.
2019-06-18 12:33:28 DEBUG (MainThread) [pydeconz.utils] Sending {} to http://192.168.1.223:80/api/07856A3797
2019-06-18 12:33:28 INFO (MainThread) [homeassistant.setup] Setting up zone
2019-06-18 12:33:28 INFO (MainThread) [homeassistant.setup] Setup of domain zone took 0.0 seconds.
2019-06-18 12:33:28 INFO (MainThread) [homeassistant.setup] Setting up sun
2019-06-18 12:33:28 INFO (MainThread) [homeassistant.setup] Setup of domain sun took 0.5 seconds.
2019-06-18 12:33:28 INFO (SyncWorker_5) [homeassistant.loader] Loaded workday from homeassistant.components.workday
2019-06-18 12:33:31 INFO (MainThread) [homeassistant.setup] Setting up ssdp
2019-06-18 12:33:31 INFO (MainThread) [homeassistant.setup] Setup of domain ssdp took 0.0 seconds.
2019-06-18 12:33:31 INFO (SyncWorker_2) [homeassistant.loader] Loaded moon from homeassistant.components.moon
2019-06-18 12:33:32 INFO (MainThread) [homeassistant.setup] Setting up tradfri
2019-06-18 12:33:32 INFO (MainThread) [homeassistant.setup] Setup of domain tradfri took 0.0 seconds.
2019-06-18 12:33:32 INFO (MainThread) [homeassistant.setup] Setting up lovelace
2019-06-18 12:33:32 INFO (MainThread) [homeassistant.setup] Setup of domain lovelace took 0.1 seconds.
2019-06-18 12:33:32 INFO (MainThread) [homeassistant.setup] Setting up group
2019-06-18 12:33:32 INFO (MainThread) [homeassistant.setup] Setting up ps4
2019-06-18 12:33:32 INFO (MainThread) [homeassistant.setup] Setup of domain ps4 took 0.0 seconds.
2019-06-18 12:33:32 INFO (MainThread) [homeassistant.setup] Setting up input_boolean
2019-06-18 12:33:32 INFO (SyncWorker_16) [homeassistant.loader] Loaded webostv from homeassistant.components.webostv
2019-06-18 12:33:32 INFO (MainThread) [homeassistant.setup] Setting up discovery
2019-06-18 12:33:33 INFO (MainThread) [homeassistant.setup] Setup of domain discovery took 0.0 seconds.
2019-06-18 12:33:33 INFO (MainThread) [homeassistant.setup] Setting up zwave
2019-06-18 12:33:33 INFO (MainThread) [homeassistant.setup] Setup of domain zwave took 0.0 seconds.
2019-06-18 12:33:38 INFO (MainThread) [homeassistant.components.zwave] Z-Wave USB path is /dev/ttyACM0
2019-06-18 12:33:38 INFO (MainThread) [homeassistant.setup] Setting up wake_on_lan
2019-06-18 12:33:38 INFO (MainThread) [homeassistant.setup] Setup of domain wake_on_lan took 0.0 seconds.
2019-06-18 12:33:38 INFO (MainThread) [homeassistant.setup] Setting up ipma
2019-06-18 12:33:38 INFO (MainThread) [homeassistant.setup] Setup of domain ipma took 0.0 seconds.
2019-06-18 12:33:38 WARNING (MainThread) [homeassistant.setup] Setup of updater is taking over 10 seconds.
2019-06-18 12:33:38 WARNING (MainThread) [homeassistant.setup] Setup of xiaomi_aqara is taking over 10 seconds.
2019-06-18 12:33:38 WARNING (MainThread) [homeassistant.setup] Setup of homekit is taking over 10 seconds.
2019-06-18 12:33:38 INFO (SyncWorker_8) [homeassistant.components.xiaomi_aqara] Discovering Xiaomi Gateways (Try 1)
2019-06-18 12:33:38 ERROR (MainThread) [.] Error connecting to deCONZ gateway at 192.168.1.223
2019-06-18 12:33:38 WARNING (MainThread) [homeassistant.config_entries] Config entry for deconz not ready yet. Retrying in 5 seconds.
2019-06-18 12:33:38 INFO (MainThread) [homeassistant.setup] Setup of domain updater took 12.1 seconds.
2019-06-18 12:33:39 INFO (MainThread) [homeassistant.setup] Setting up binary_sensor
2019-06-18 12:33:39 INFO (MainThread) [homeassistant.setup] Setup of domain homekit took 11.7 seconds.
2019-06-18 12:33:39 INFO (SyncWorker_13) [homeassistant.loader] Loaded yr from homeassistant.components.yr
2019-06-18 12:33:39 INFO (MainThread) [homeassistant.setup] Setting up notify
2019-06-18 12:33:39 INFO (SyncWorker_8) [xiaomi_gateway] Xiaomi Gateway f0b4299a8xxx found at IP 192.168.1.189
2019-06-18 12:33:39 INFO (SyncWorker_8) [xiaomi_gateway] Discovering Xiaomi Devices
2019-06-18 12:33:39 INFO (SyncWorker_8) [xiaomi_gateway] Found 17 devices
2019-06-18 12:33:39 INFO (MainThread) [homeassistant.setup] Setting up system_health
2019-06-18 12:33:39 INFO (MainThread) [homeassistant.setup] Setup of domain system_health took 0.0 seconds.
2019-06-18 12:33:39 INFO (SyncWorker_1) [homeassistant.loader] Loaded weather from homeassistant.components.weather
2019-06-18 12:33:39 INFO (SyncWorker_19) [homeassistant.loader] Loaded google_translate from homeassistant.components.google_translate
[... cut ...]
2019-06-18 12:36:02 INFO (Thread-5) [pyhap.hap_server] Got connection with ('192.168.1.176', 54593).
2019-06-18 12:36:02 INFO (Thread-8) [pyhap.hap_server] 192.168.1.176 - "POST /pair-verify HTTP/1.1" 200 -
2019-06-18 12:36:03 INFO (Thread-8) [pyhap.hap_server] 192.168.1.176 - "POST /pair-verify HTTP/1.1" 200 -
2019-06-18 12:36:03 INFO (Thread-8) [pyhap.hap_server] 192.168.1.176 - "GET /accessories HTTP/1.1" 200 -
2019-06-18 12:36:24 INFO (Thread-8) [pyhap.hap_server] 192.168.1.176 - "PUT /characteristics HTTP/1.1" 204 -`

HackThatNow, 2019-07-16:
Wow, this is exactly happening to me as well.
I'm using 0.95.4 Hass.io on rpi3 b+.

Conbee plugin installed on the same pi. So Conbee 2 + hassio is running on the same pi. 

After each reboot, sometimes the devices show up in hassio, and sometimes it doesn't show up. If I go to the phoscon web interface, everything works if I control it using their environment. 

Rebooting only the plugin, helped me once to have it working again, but tried it again later and looks like the only thing is to keep rebooting till it shows up. 

Usually it needs just a single reboot and then it is fixed again. And if it is working, it works Rock solit. Unfortunately, as I modify things in hassio configuration, I need to reboot often and then most of the time the conbee doesn't show up anymore.

alundstroem, 2019-07-20:
I have a similar situation as described by @HackThatNow.

My setup is on a Raspberry Pi 3, where hass is deployed in a python venv (as per the manual installation instructions) and deCONZ via the community docker image.

Sometimes after a reboot the deCONZ integration isn't available, sometimes it is...

Could a solution be to have hass wait on deCONZ to be fully started before starting up?

ualex73, 2019-09-02:
I had similar issues with DeCONZ under normal docker ("host" option is set properly), it shows the 172.x.x.x IP address in the DeCONZ GUI. A restart of DeCONZ always solves the issue. I added a small delay after reboot, and now it looks to be much more stable (fingers crossed).

M-Gregoire, 2019-09-19:
I had the same issue as described above. I use docker-compose to manage both homeassistant and deCONZ containers.

I've enabled `ssdp` in my `configuration.yaml` as advised, deleted and recreated the deconz integration from HA. Since then, the deconz IP gets automatically updated in `config/.storage/core_entries`.

Don't forget to add deconz container as a dependency to the HA container.

HackThatNow, 2019-10-01:
> I had the same issue as described above. I use docker-compose to manage both homeassistant and deCONZ containers.
>
> I've enabled `ssdp` in my `configuration.yaml` as advised, deleted and recreated [...]

Could you let me know how to enable ssdp? I cannot find information about this on google. Also, what you mean about deconz container as dependency to the HA container? Is this something not related to users of HASS.IO?

M-Gregoire, 2019-10-08:
Unfortunately, It seems my solution doesn't work, homeassistant was able to discover deconz when it changed ip but didn't replaced the existing instance with the new one.

Concerning ssdp, simply add `ssdp:` in your config as stated here:
https://www.home-assistant.io/integrations/ssdp/

The only working way I found is setting both deconz and homeassistant to use the `host` network mode.

I don't use HASS.IO.

Kane610, 2019-10-08:
@M-Gregoire upgrade to hass beta, there are fixes to discovery

M-Gregoire, 2019-10-19:
@Kane610 Oh, indeed! Thank you for pointing that out

SamB-GB, 2019-12-22:
I have the same issue but I found that when watching the devices in /dev/serial/by-id on Ubuntu that after the start up the Conbee II disappears for a few seconds and then comes back and then goes away again and then comes back. By the time that's happened the system has started up and is ready.

chbndrhnns, 2020-01-20:
I also noticed the behavior of having an outdated IP in `core.config_entries` that I had to update manually to make it work after a reboot of my Sonology NAS

closed 2020-05-26 by stale[bot]