dresden-elektronik/deconz-rest-plugin #1172: deCONZ with ConBEE  autonomously creates groups

nodefeet, 2019-01-24:
I noticed a strange behavior. As soon as I have one or more groups created something autonomously creates a group with the same lights and an id of original id plus 100.

So when I have a group: “MyGroup” with the id of 20 after a few hours later there is another group called “Group 110”. After unplugging every member in the network besides the ConBee stick running with deCONZ headless on a RPi or any other program talking to the API there is yet another Group with again plus 100 of the self-created group 120: which makes it “Group 220”. Although it didn’t copy all the lights of group 120 or rather group 20 this time. 

I think this is happening since I updated the ConBEE Firmware to 262F0500 recently but I could be wrong on that. I did a reset on ConBEE and updated deCONZ to the latest stable version 2.05.57 / 1/20/2019 but the behavior is still the same (On a side note: after the firmware update all the lights were always showing up as reachable even when they weren’t plugged in, but this seems to work again after I updated deCONZ ).

Does anyone have an idea what might be the problem?

Here is what get all groups currently shows: I only created group 18, 19, 20.

> {

    "18": {
        "action": {... },
        "devicemembership": [],
        "etag": "d13368c6063bbf53abf6d1b4d0066aec",
        "id": "18",
        "lights": [
            "1"
        ],
        "name": "Gruppe A",
        "scenes": [],
        "state": {
            "all_on": false,
            "any_on": false
        },
        "type": "LightGroup"
    },
    "19": {
        "action": {... },
        "devicemembership": [],
        "etag": "d13368c6063bbf53abf6d1b4d0066aec",
        "id": "19",
        "lights": [
            "1"
        ],
        "name": "Gruppe B",
        "scenes": [],
        "state": {
            "all_on": false,
            "any_on": false
        },
        "type": "LightGroup"
    },
    "20": {
        "action": {...},
        "devicemembership": [],
        "etag": "d13368c6063bbf53abf6d1b4d0066aec",
        "id": "20",
        "lights": [
            "3"
        ],
        "name": "Gruppe C",
        "scenes": [],
        "state": {
            "all_on": false,
            "any_on": false
        },
        "type": "LightGroup"
    },
    "118": {
        "action": {...},
        "devicemembership": [],
        "etag": "d13368c6063bbf53abf6d1b4d0066aec",
        "id": "118",
        "lights": [
            "1"
        ],
        "name": "Group 118",
        "scenes": [],
        "state": {
            "all_on": false,
            "any_on": false
        },
        "type": "LightGroup"
    },
    "119": {
        "action": {...},
        "devicemembership": [],
        "etag": "d13368c6063bbf53abf6d1b4d0066aec",
        "id": "119",
        "lights": [
            "1"
        ],
        "name": "Group 119",
        "scenes": [],
        "state": {
            "all_on": false,
            "any_on": false
        },
        "type": "LightGroup"
    },
    "120": {
        "action": {... },
        "devicemembership": [],
        "etag": "d13368c6063bbf53abf6d1b4d0066aec",
        "id": "120",
        "lights": [
            "1",
            "2",
            "3",
            "4"
        ],
        "name": "Group 120",
        "scenes": [],
        "state": {
            "all_on": false,
            "any_on": false
        },
        "type": "LightGroup"
    },
    "220": {
        "action": {...},
        "devicemembership": [],
        "etag": "d13368c6063bbf53abf6d1b4d0066aec",
        "id": "220",
        "lights": [
            "1",
            "4"
        ],
        "name": "Group 220",
        "scenes": [],
        "state": {
            "all_on": false,
            "any_on": false
        },
        "type": "LightGroup"
    }

Shaolo, 2019-01-25:
Have you set names and groups in Phoscon specifically?

I read one of the earlier issues today that said they added auto groups for things like the button remote that comes with the Ikea outlet. Possible this is related in some way.

nodefeet, 2019-01-26:
Yes I created them in Phoscon and later through the API with Postman directly, same result. 

But I do now suspect an LED Spot after all. It must have been creating the group while I was unplugging everything. That’s why I thought it must have been ConBee. There were no new groups the whole day. I then started plugging in light by light and waited some time. After the LED Spot was plugged in (which I had for a while now but did not assign a group until recently) a new copy group was autonomously created after a few minutes. 

So I am sorry for the fuss. I am checking it more closely today, but it has probably nothing to with ConBee or deconz.

But if anyone has an idea why it would do this, I’d be interested to hear that.

manup, 2019-01-26:
>But I do now suspect an LED Spot after all. It must have been creating the group while I was unplugging everything.

I've seen similar things with Ikea lights but didn't dig deeper yet.

deCONZ queries all groups on a light periodically, and will add them to the REST-API if not already known.

Which light type is this exactly?

closed 2019-06-03 by stale[bot]