dresden-elektronik/deconz-rest-plugin #1174: Cannot add action with URL /rules/<id> to rule

Thomas-Vos, 2019-01-24:
I'm not able to add the following action to a rule:
```
{
  "address": "/rules/1",
  "body": {
    "status": "disabled"
  },
  "method": "PUT"
}
```
deCONZ answers with the following response:
```
[
  {
    "error": {
      "address": "/rules/1",
      "description": "Rule actions contain errors or an action on a unsupported resource",
      "type": 608
    }
  }
]
```
I think this is because an action with URL `/rules/<id>` is not allowed on a rule. This is allowed on the Hue bridge. I need this so I can disable a rule when a rule is activated. Is this something that could be supported?

manup, 2019-01-27:
Good catch, it is fixed now and will be part of version 2.05.58.

Thomas-Vos, 2019-02-03:
Thanks! Just tested this and it's working now.

closed 2019-02-03 by Thomas-Vos