dresden-elektronik/deconz-rest-plugin #1175: Zigbee Network very slow and not connection properly

PrimaryInc, 2019-01-24:
Hi, 

Today my network started slowing down extremely and now only partially works.
Updated to .57 to see if that helped, no luck.

@manup any ideas?

[image]
[image]

manup, 2019-01-24:
Few Questions:
* How long is it running like this?
* Which firmware version are you using (Menu > Help > About deCONZ)
* Have you tried restart deCONZ and perhaps unplug ConBee for a few seconds?

manup, 2019-01-24:
Hmm indeed strange, versions look fine.

Do you have a usb extension cable available to put between ConBee and PC?

Can you also please create a backup in the Phoscon App and send it to mpi@dresden-elektronik.de, I can check if the Zigbee network parameters are ok.

PrimaryInc, 2019-01-24:
I did use a USB extension cord before but removed by your advice.
2.4GHz use channel 1,6 and 11. deCONZ is on 25.

Email sent :)

manup, 2019-01-24:
Thanks, config looks fine all parameters are stable and same as previous versions.

>I did use a USB extension cord before but removed by your advice.

Advice is to actually use a extension cable :) Without it on some machines the signal quality gets really messed up when the ConBee is connected directly.

>2.4GHz use channel 1,6 and 11. deCONZ is on 25.

Note Zigbee channels are different from WiFi channels as described in the following article:

https://support.metageek.com/hc/en-us/articles/203845040-ZigBee-and-WiFi-Coexistence

However 25 is totally fine here, I would advice to keep it.

PrimaryInc, 2019-01-24:
Ok, at the moment my USB extention is in use by another USB stick (z-wave) so I need to get a new one.
But it has been working for weeks without issues until today.
Status now :)

[image]

PrimaryInc, 2019-01-24:
Just tried to move the USB ConBee stick by adding a USB splitter approx 2m from server, same results.

PrimaryInc, 2019-01-25:
Hi,

No there is no change today unfortunately. Guess I should start deCONZ with debugging to see if it can show something?

manup, 2019-01-25:
>Guess I should start deCONZ with debugging to see if it can show something?

Yes that would be helpful, do you know how to do it on Windows?

I'll put up a small Wiki page for debugging later today, which also describes how to enable debugging on Windows.

PrimaryInc, 2019-01-25:
Yes, Iv'e started debug now, at the same time I did reboot all my AP and it seems that it helped a little bit. So I've stated a downgrade of FW om all AP, it might be that it cloud be something there. I did upgrade them a few days ago but I could not connect the dots to this issue.

I'll send log in a little bit, but ca see a lot of these messages:
[5256] 12:12:28:802 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:28:894 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:28:989 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:29:000 APS-DATA.confirm id: 233, status: 0xA7 NO_ACK
[5256] 12:12:29:082 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:29:178 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:29:270 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:29:364 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:29:457 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:29:552 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:29:646 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:29:738 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:29:832 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:29:927 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:30:020 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:30:113 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:30:208 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:30:301 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:30:394 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:30:488 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:30:583 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:30:676 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:30:770 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:30:863 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:30:957 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:31:051 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:31:147 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:31:238 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:31:362 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:31:425 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:31:521 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:31:615 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:31:708 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:31:801 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]
[5256] 12:12:31:896 ZDP skip fetch, node 0xB57FFFEB957BA total unconfirmed zdp request count 2 [2]

PrimaryInc, 2019-01-25:
It does seem that FW on AP could been the case here, this is landscape after downgrade.

[image]

ebaauw, 2019-01-25:
If the AP firmware changed the WiFi radio settings (e.g. more transmit power, broader channels, more parallel channels, more aggressive resends on collisions), it might very well cause more interference with the ZigBee radio.  How closely is the AP placed to the ConBee?

PrimaryInc, 2019-01-25:
@ebaauw, I haven't looked at the release info in details, but it does not change ratio or power setup. 
I use Ubiquiti Unifi AP-AC-Pro 3 x located on loft. AP covers Wifi for the entire house and then some. Coverage extends outside the house. ConBee is located in basement, most light in the roof on main floor. It is something that made the AP "more" aggressive somehow, but I do not know that.

manup, 2019-01-25:
Interesting, maybe a WiFi scanner app brings some light what changed about the WiFi signal.

manup, 2019-01-27:
Closing the issue for now.

closed 2019-01-27 by manup