dresden-elektronik/deconz-rest-plugin #1176: Not able to control GLEDOPTO devices any longer

saxn-paule, 2019-01-24:
Today I updated to the .57 firmware. Its not possible to control GLEDOPTO devices via Phoscon any longer.

The controllers are shown and could be added to groups but the sliders don't appear.

manup, 2019-01-24:
I have only the GLEDOPTO RGBW version with SMILB brand here for testing, this one works fine.

Version: 1.0.2
Datecode: 20131206

You may try to force Phoscon App cache refresh by appending some random search query in the URL bar.

    http://<gateway-ip-address>/pwa/index.html?12345

saxn-paule, 2019-01-24:
Still no change. If I click the lights icon of a group, the dim slider jumps to ~50% but the lights stay off.

Do you live in DD? The I could give you a 009 for testing.

saxn-paule, 2019-01-24:
007 and 008 still work like before.

saxn-paule, 2019-01-27:
I use Deconz in headless mode, so I can't make any screenshots.
With .55 controlling the 009 was still possible.

I only updated the firmware. Nothing else.

saxn-paule, 2019-01-30:
With the .58 version the 009 isn't detected at all any longer:

[image]

saxn-paule, 2019-01-30:
I just tried a second 009 controller. It isn't detected at all. Nothing shown in Phoscon.

saxn-paule, 2019-01-30:
I changed the controller. The new one is controllable again. So, my fault. Sorry.

manup, 2019-01-30:
:) I guess these require a bit patience, the missing modelid and manufacturer names are populated in an idle timer which takes a bit. But once set these should be directly available in further restarts.

ebaauw, 2019-01-31:
As I understand, the GL-C-009 is a Dimmable Light (no colour nor colour temperature)?  I suspect it has both a ZLL and a ZHA endpoint doing the same, like the GL-C-007 (RGBW), see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/952.  Could you post a screenshot from the deCONZ GUI and the resources as exposed by the REST API?

The GL-C-008 (RGB+CCT) doesn't seem to react to _Move to Level (with On/Off)_, see https://github.com/ebaauw/homebridge-hue/issues/424.  If the 009 does the same, PUTting `{"bri": 0}` wouldn't turn it off.  Also, it wouldn't turn on if you PUT `{"on": true, "bri": 254}`.

saxn-paule, 2019-01-31:
Yes it's only a dimmer. I use deconz in headless mode. How could I provide these screenshots?

ebaauw, 2019-01-31:
> Yes it's only a dimmer. I use deconz in headless mode. How could I provide these screenshots?

You need to run the GUI for that.  I use VNC on a headless Raspberry Pi to share the graphical screen to my Mac.

nagyL, 2019-02-23:
Some sample images related to my GL-C-009:

[image]

[image]

closed 2019-06-30 by stale[bot]