dresden-elektronik/deconz-rest-plugin #1177: How fast does deCONZ update?

Shaolo, 2019-01-25:
I'm curious how fast deCONZ gui updates.

If I turn on some lights via their switch, and it takes them 10 seconds to show mesh connections in deCONZ, are they actually taking that long to connect to the network? Are they potentially connecting faster and it's taking longer for that data to show up in the GUI?

Hoping to provide some meaningful feedback/data on the performance of my network.

manup, 2019-01-25:
> Are they potentially connecting faster and it's taking longer for that data to show up in the GUI?

This. The lines between the nodes are just visual sugar and are queried slower than other attributes like on/off brightness state.

Since 2.05.57 polling of lights has also optimized for speed, so once a lights is turned on it should be available within a few seconds. This of course depends all on network size.

One advise, if possible keep lights/routers always powered. Especially in networks with Xiaomi battery powered devices which often get stubborn when the parent node (like a light) is powered off.

Shaolo, 2019-01-25:
My setup is (at the moment) entirely z wave light switches and Ikea bulbs. Trying to leave the lights powered all the time ends up being inconvenient. Someone always comes through and turns off the light at the switch. Would adding something like a persistent router (Ikea outlet switch) help the lights re-establish their mesh faster?

manup, 2019-01-25:
>Would adding something like a persistent router (Ikea outlet switch) help the lights re-establish their mesh faster?

Persistent routers are preferable, especially in combination sensors and switches.

>My setup is (at the moment) entirely z wave light switches and Ikea bulbs. Trying to leave the lights powered all the time ends up being inconvenient. Someone always comes through and turns off the light at the switch.

Zigbee works best when most routers are powered otherwise there is no way around delays with re-establish routes.

Think of a highway where cars just stop driving, it takes time for the other cars to adapt to this situation and the first cars will just crash into the stopped car (route/lane no longer works).

When a light is powered off, the mesh will only notice this after a while (roughly 1 minute at the lowest layer).

manup, 2019-01-27:
Closing the issue for now, further discussion in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/445

closed 2019-01-27 by manup