dresden-elektronik/deconz-rest-plugin #1181: Xiaomi WXKG02LM first press not registering

cappadanna, 2019-01-25:
Hi. I’m using two Xiaomi WXKG02LM switches controlling a couple of Ikea bulbs using a RPi with Raspbee. I have them setup for toggling the bulbs on/off in the Phoscon app.

The problem is that often the first press won’t register to turn on/off the lights. I often have to press them a second time.

However, i can see the Xiaomi switches sending commands the first time since i see them ”bouncing” in the Apple home app (using homebridge hue)

Anyone have had any similar problem?

2.05.57 with fw. 26300500

bdickhaus, 2019-01-25:
I have one that does the same thing. I've also had it completely disconnect when I didn't use it for a long time. I had to rediscover it.

manup, 2019-01-26:
> I have them setup for toggling the bulbs on/off in the Phoscon app.

Was the configuration done with Phoscon App switch editor or Apple home app?
Can you please show the configuration for that button?

cappadanna, 2019-01-26:
The configuration for toggling was done in the Phoscon app. A have a double-press configured through the Apple home app but have tried deleting it, doesn't change. I have removed the switches and reconnected them. Even tried with a fresh install of my RPi.

I've found that i can fool it:
If i turn the bulb of in the apple home app and press the switch, the press is registered but the bulb isn't turned on. (and same the other way around) I't like its sending an off command again when it should be an on command. Does the software read the state of the bulb (on or off) before it sends the command?

I hope i'm making sense to you :)

[image]
[image]

ebaauw, 2019-01-26:
Doesn't Phoscon send a _Toggle_ command?

I often find that after (re-)starting deCONZ, it takes a while before the REST API correctly reflects the light states.  Typically, it reports a lot of lights as on, white they're actually off.  When looking at the state to determine whether to send _On_ of _Off_ it will send _Off_ for the first press, wrongly thinking some of the lights are on.

manup, 2019-01-26:
Nope, currently we use a CLIPGenericFlag which is toggled between true/false.
And I think this is the very problem here.

What likely happens is:

* CLIPGenericFlag = true
* Group = off
* On first button press Off command is sent... and nothing happens ;)
* Only the second button press will send the desired On command

I'll put it on the todo list for next versions, maybe 2.05.59. group.state.any_on might be a better approach here.

cappadanna, 2019-01-26:
> Nope, currently we use a CLIPGenericFlag which is toggled between true/false.
> And I think this is the very problem here.
>
> What likely happens is:
>
> * CLIPGenericFlag = true
> * Group = off
> * On first [...]

That must be what causes it. Now at least i understand what is happening :) Thank you! Keep up the good work!

johanhansen, 2019-02-12:
I have also observed the same problem. Using a Xiaomi wall switch configured to toggle, it functions as expected as long as I don't control the linked light in Home.app. Once I do that the state is out of sync between the state of the light and the CLIP-sensor.

Toggeling the light using Home.app should also trigger the CLIP-sensor to toggle. Right?

Oliviakrkk, 2019-04-07:
Hi @manup

Any updates on those switches? I have similar issue. But sometimes I have to press button multiple times to make it work.

cappadanna, 2019-08-05:
Is this still on the todo list? 😊

jo-me, 2019-08-25:
Same issue here...

manup, 2019-08-25:
The problem is fixed and will be part of the upcoming 2.05.57.
Sorry took way longer than anticipated.

manup, 2019-08-27:
https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_67

The issue is fixed in 2.05.67, but you need to update or re-create the switch button assignment.

nihj, 2019-12-22:
Hi,

I'm still having this exact issue with my WXKG11LM (I know that's not the switch the issue was reported on byt I assume the fix is somewhat generic...?), even though I'm currently using version 2.05.72.

There was a mentioning of having to redo or update the key assignments, could you clarify that a bit @manup ? I (think I) have re-done my key assignments multiple times now but to no avail.

nihj, 2020-02-08:
Ping @manup 

I'm still experiencing this problem, even though I'm currently using version 2.05.72.

There was a mentioning of having to redo or update the key assignments, could you clarify that a bit? I (think I) have re-done my key assignments multiple times now but to no avail.

Mimiix, 2020-06-05:
Is this still a issue?

nihj, 2020-06-11:
This is technically not an issue, because when using the "Toggle" action it actually works, the first press is registered.

However, the problem still occurs when using the "Toggle dimming up/down, on/off" functionality. I was using this, very nifty, feature when I "still was not able to get it working" in my comments above and have now stopped using that.

So, follow up question: was tis fix explicitly put on ONLY the code path of the "Toggle" action, and if so would it be possible to also fix it for "Toggle dimming up/down, on/off"? It's really handy and I'd like to be able to use it again.

Mimiix, 2020-06-11:
@nihj I'm going to hand this over to @YKO-de as she is the dev of Phoscon. I will move this issue to their repository. She is the perfect person for your question ;)