dresden-elektronik/deconz-rest-plugin #1182: Xiaomi Mijia (HonneyWell) gas detector

AdamWeglarz, 2019-01-25:
Hi,

I have also Xiaomi Mijia (HonneyWell) gas detector. It connects without any issue to Xiaomi Gateway so it work correctly. It does not connect at all to Deconz zigbee. It is not visible in Deconz graphical interface (In oposite to Xiaomi smoke sensor, which is visible in Deconz app).
How I can help to diagnoze issue?

Adam

manup, 2019-01-25:
The device is currently not supported.

Please provide the details as described here:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

SirJMD, 2019-01-25:
I've hit the same road block, I also have a gas detector I'd like to use.

I'd be more than happy to supply the needed details, but I have no idea where to find "Cluster Info Panel" and the rest mentioned in the link. I guess it's no via Phoscon?

AdamWeglarz, 2019-01-26:
For starters. Tomorrow I will buy lighter to start alarm and capture what is send by the device when alarm is raised and cleared:

[image]
[image]
[image]

ebaauw, 2019-01-26:
This one plugs into the mains power?  It looks a lot like the Heiman sensor I had on loan, also Jennic based.  While detecting gas, it issues an alarm from the _IAS Zone_ cluster while making a lot of noise.  Same for pressing the test button.  Unfortunately, the siren cannot be activated externally.

It’s probably enough to whitelist the sensor (twice: in the sensor whitelist and in the _IAS Zone_ logic, to create the right sensor type).  Could you press _Read_ on the cluster into panel so the attribute values get populated?

SirJMD, 2019-01-26:
It's powered by 12V and comes with a mains adapter. It also have a NO set of contacts, for hard wired alarm.

SirJMD, 2019-01-26:
There's no battery.

ebaauw, 2019-01-26:
Ah, OK.  Then it's "just" an external power adapter.

AdamWeglarz, 2019-01-26:
Basic
[image]
Ias
[image]
power
[image]
 
Rest is empty.
There are two push options 
- Test alarm
- Sensitivity of the sensor

Is it possible to sniff traffic between xiaomi gateway and gas sensor?
PS The same options are available for smoke sensor.

Adam

ebaauw, 2019-01-26:
OK, if you write the mac address of the RaspBee or ConBee to the _IAS_CIE_Address_ attribute (0x0010 in the _IAS Zone_ cluster), _Zone State_ should change to _Enrolled_ and you should see alarms in _Zone Status_ when pressing the test button.  Be sure to protect your ears ;-)

It doesn't look like the sensitivity is exposed over Zigbee.

AdamWeglarz, 2019-01-26:
Hi,

should it be visible in Rest API?
[image]
It is recognized as co2 sensor after putting MAC as you recommended. Not enrolled

Adam

SirJMD, 2019-01-27:
Is there any additional information required before it can be added? If so, I'd be happy to help.

fuzzysb, 2019-03-15:
I have one with the same issue, when updating the address of the Deconz Gateway in to the  IAS_CIE_Address i am also unenrolled, when trying to read the reporting configuration i get the following error
[image]

alahdal, 2019-07-14:
Hi,
Are you going to support this device? Or is it already compatible with RaspBee?

alahdal, 2019-08-13:
I got my new Xiaomi Mijia (HonneyWell) gas detector, it is not connecting to RaspBee. Any plan?

iamkarlson, 2019-10-05:
Any hope to get this available?

Otnow, 2019-11-04:
> 
> 
> Is there any additional information required before it can be added? If so, I'd be happy to help.

+1

I have this device and Conbee II. What information needs to be provided to add support?

WladyX, 2019-11-14:
I also have 2 of these and Conbee 2 and also can help with info.

AdamWeglarz, 2019-12-08:
So any news?

fastsonic, 2019-12-08:
hi, i also try to connect to the Gas Sensors,
when its poosibel to Help ? please contact me
Thanks

L3H0, 2019-12-13:
> 
> 
> Any hope to get this available?

+1

WladyX, 2019-12-13:
This was working on zigbee2mqtt, maybe it's possible to get some inspiration from that code? :)

SirJMD, 2019-12-23:
I've given up on deconz, zigbee2mqtt is being maintained much better.

WladyX, 2019-12-24:
Maybe, in my case it did not handle the size of my network well (+80 nodes).
On topic: If we can help in anyway to get this supported, please let us know.

mwhiegl, 2020-01-02:
I second the request and offer to help. Not a coder, but let me know if any more data is needed or so.

fastsonic, 2020-01-15:
push Also available for any Help

iamkarlson, 2020-01-15:
@SirJMD is there also a way to set up direct bindings with zigbee2mqtt?

AdamWeglarz, 2020-02-18:
Hello,

Can you tell that you plan to add support for this device?

JohCo-hassio, 2020-02-22:
Hello,
I also bought this sensor last week and unfortunately it does not come online even in the Rest-api not receiving anything to HA. What is different is that the site now supports the ZYCT-202.

AdamWeglarz, 2020-03-22:
I gave up and setup zigbee2mqtt :-(

L3H0, 2020-03-23:
> 
> 
> I gave up and setup zigbee2mqtt :-(

z2m on Conbee2 or C2538?

And3rsL, 2020-05-11:
is this device supported now? i currently looking for a gas detector to buy and i would like to buy the xiaomi one

L3H0, 2020-05-11:
@And3rsL nope, till today this device is not supported.

andriej, 2020-05-11:
It can be seen in integrations using API (like HomeAssistant) not in Phoscon.
Nor you will be able to set sensitivity as in app.
But I still recommend the sensor, I had tested it. I just hope dresden will finally wake up some day and keep up to the market...

AdamWeglarz, 2020-05-11:
> 
> 
> It can be seen in integrations using API (like HomeAssistant) not in Phoscon.
> Nor you will be able to set sensitivity as in app.
> But I still recommend the sensor, I had tested it. I just hope dresden will finally wake up some day and keep up to the market...

No it is not. You are mixing gas sensor with smoke sensor which is really working like you explain.

andriej, 2020-05-11:
Sorry! You're right.

SwoopX, 2020-06-18:
I'll add it soon, should be a quick one

alahdal, 2020-06-18:
Very much appreciated! @SwoopX

JohCo-hassio, 2020-06-18:
Very nice 👍🏻

OldSystemMan, 2020-06-18:
At last 👍

SwoopX, 2020-06-18:
I still need someone to collect me 1h of log data so I can inspect the special attribute reporting for this device.

`sudo systemctl stop deconz-gui`
`/usr/bin/deCONZ --dbg-info=2 > xiaomi_debug`

alahdal, 2020-06-19:
I hope you find the file helpful this time. It is worth mentioning that the Gas detector is not connected to the RaspBee, because it is not supported yet.

Please let me know if there is anything else I can do.

[xiaomi_debug.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4804925/xiaomi_debug.txt)

SwoopX, 2020-06-20:
That's perfect, thanks!! Lots of stuff in there I always wanted to get a look at.

What I actually do not see it the gas sensor as far as I can tell. You got one right?

alahdal, 2020-06-20:
@SwoopX I got the Gas Sensor, but it is not connected with RaspBee. When I try to connect via Phoscon App (web), it does not connect.

SwoopX, 2020-06-20:
@alahdal I see. Will come with the next release. Do you use HA or docker? If not, you could compile a dev version of the plugin and become my lab rat ;)

alahdal, 2020-06-20:
It is installed on a Raspbian.

SwoopX, 2020-06-21:
So, you might want to try the following. Would really appreciate if you could generate another debug file when you want to play with it (then use `/usr/bin/deCONZ --dbg-info=2 > xiaomi_debug2` for step 5 mentioned below):

1. Clone my repository containing the changes
`git clone https://github.com/SwoopX/deconz-rest-plugin.git`
`cd deconz-rest-plugin`
2. Shut down deconz
`sudo systemctl stop deconz-gui`
3. Make a backup of file "general.xml"
`sudo mv /usr/share/deCONZ/zcl/general.xml /usr/share/deCONZ/zcl/bak_general.xml`
4. Copy the amended general.xml to the required location
`sudo cp general.xml /usr/share/deCONZ/zcl/`
5. Start deconz again
`sudo systemctl start deconz-gui`

For the smoke sensors, you should see 2 additional attributes in cluster `0x0500 IAS Zone`. For attribute `Selftest/Density`, double click on it and read the value (not read config!). Please share a screenshot from that.

Now, if all wen well, you should be able to initiate selftest by writing value `03010000` to that attribite. Sensitivity should be setable with values as follows:
- Low: `04010000`
- Medium: `04020000`
- High: `04030000`

alahdal, 2020-06-21:
@SwoopX These are the screenshot for the attributes in 0x0500.

[image]

[image]

SwoopX, 2020-06-21:
My bad, used the wrong manufacturer code. Can you try once again? Start with deleting the downloaded repository locally `sudo rm -rf ./deconz-rest-plugin`.

alahdal, 2020-06-21:
Reading Failed.

PS. Just to clarify, we are at the Gas detector topic not the [Smoke Detector](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2954#issuecomment-645613868)

[image]

SwoopX, 2020-06-21:
Yes, but if I understand this correctly, it should work for both, smoke and gas. I'm not in front of my desktop to double check, but at least both attributes are offered. Did sesnitivity succeed in reading? If yes, can you share the screenshot?

From what I've read, those devices are deep sleepers, waking up only every 30 secs. So you might try a bit for reading. The node should blink blue/green when ready...

alahdal, 2020-06-21:
I tried to read from another sensor that is close to the RaspBee system (I thought maybe with better connection I may get different results), I got different error:

[image]

[image]

Moreover, I tried reading other basic attributes all gives the same error (reading failed), maybe something wrong with my setup?

alahdal, 2020-06-21:
Here is a log 
[xiaomi_debug2.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4809795/xiaomi_debug2.txt)

alahdal, 2020-06-21:
I restarted deCONZ with sudo, however, it showed all nodes without names and not does not show all clusters. I will leave it for sometime and check again, I noticed the clusters are shown gradually.

Why I run as sudo, I noticed that I can read the cluster info when it is run as sudo.

alahdal, 2020-06-21:
I have stopped deCONZ and restarted without sudo, when I tried to read attribute in basic cluster info, it worked without problem. However, for the new two new attributes, I am still getting the error shared in the last screenshot

SwoopX, 2020-06-21:
Hm, maybe the 2 attributes are not required to be manufacturer specific. The additions to `general.xml` were the following

```
<attribute-set id="0xf000" description="Xiaomi specific" mfcode="0x1037">
    <attribute id="0xfff0" name="Sensitivity" type="u32" access="r" required="o" mfcode="0x1037"></attribute>
    <attribute id="0xfff1" name="Selftest/Density" type="u32" access="rw" required="o" mfcode="0x1037"></attribute>
</attribute-set>
```

You could try removing this part for the data above `mfcode="0x1037"` to see if that changes anything.

Btw, if you pull my repo now, support for the gas sensor is included.

alahdal, 2020-06-22:
@SwoopX after removing (mfcode="0x1037") from above data, I get "unsupported attribute" error.

[image]

I believe the Gas Sensor is connected now, and I could read some of its attributes. However, the item does not appear in the REST API nor discovered by Phoscon.

[image]

SwoopX, 2020-06-22:
@alahdal Please leave the `Sensitivity `attribute for further tests. Have you tried to write any of the low/medium/high values yet to `Selftest/density` yet? If not, please do so (with and without the mfcode in general.xml). You can also test the selftest value if you're brave.

For the gas sensor to work, you need to compile my repo. Instructions are in the readme.

alahdal, 2020-06-22:
@SwoopX when I remove the mfcode, both attributes are not supported, hence, I cannot read or write. With the mfcode, I get the error shared earlier [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1182#issuecomment-647144930) where I read/write.

DaveRuijter, 2020-07-07:
Hi! I'm wondering about the status of this development 🤗. @SwoopX can you share what is going on?
I can't wait to try this capability in Home Assistant (for the smoke detectors!).
Thanks!

AdamWeglarz, 2020-07-08:
> Hi! I'm wondering about the status of this development . @SwoopX can you share what is going on?
> I can't wait to try this capability in Home Assistant (for the smoke detectors!).
> Thanks!

Smoke sensor is working with test api even though it is not visible in phoscon

[14 more replies]

closed 2020-07-27 by SwoopX