dresden-elektronik/deconz-rest-plugin #1183: WXKG01LM: Long press, timing bug.

cnrd, 2019-01-25:
Hi

So with the new long press it seem that multiple events are sent: 1000 when pressing in, 1001 when triggering long press, and 1003 when lifting after long press.

It is however possible to trigger 1000, 1001, and 1002:
```
2019-01-25 22:09:01.997466 INFO deconz_events: {'event': 1001, 'id': 'lumisensor_switch_7'}
2019-01-25 22:09:02.214983 INFO deconz_events: {'event': 1002, 'id': 'lumisensor_switch_7'}
```
It requires that you remove your finger from the switch as soon as event 1001 is sent, this will then trigger event 1002 instead of the expected 1003.

If you look at the events, you can see that there is less than a second between them.

ebaauw, 2019-01-25:
> It requires that you remove your finger from the switch as soon as event 1001 is sent, this will then trigger event 1002 instead of the expected 1003.

This is conform spec.  Originally the `buttonevent` values come from the Hue dimmer switch:
- 1000 = (initial) press
- 1001 = hold
- 1002 = release (after press)
- 1003 = release (after long press)

So press / release results in a 1000 followed by a 1002; press / hold / release results in a 1000, followed by a 1001 per second while holding, followed by an 1003.

Different switches have different hardware capabilities.  Only a few other switches issue a 1000 event, and I've seen no others that repeatedly issue 1001 events while holding.  And we added events for double press, treble press, shake, etc. that the Hue dimmer does not issue.  See https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices#wireless-switches for the details.

ebaauw, 2019-01-25:
Sorry, I misread that.

cnrd, 2019-01-25:
So as we are emulating the long press for these switches, I'm expecting some kind of race condition between the 1001 event not being fully registered and me releasing the button.

closed 2019-06-01 by stale[bot]