dresden-elektronik/deconz-rest-plugin #1184: Unknown Xiaomi device showing up after updating from .55 to .57

cnrd, 2019-01-25:
I don't own any door locks, however this device is showing up after updating from .55 to .57

Anyone who can identify what it is and if it is supposed to be there?

[image]
[image]
[image]
[image]
[image]

ebaauw, 2019-01-25:
It’s a `lumi.vibration`.

manup, 2019-01-26:
Side note if it is not shown in the REST-API and Phoscon App as vibration sensor, you need to rejoin the sensor via sensor search to be fully detected.

closed 2019-06-02 by stale[bot]