dresden-elektronik/deconz-rest-plugin #1185: All devices in phoscon are not reachable

Sturmhorst, 2019-01-26:


rikroe, 2019-01-26:
I had the same issue a week ago - no devices respond to Phoscon/REST API calls. Also, switches in a group don't toggle the respective lights anymore. 

What fixed it for me (probably more of a workaround) was starting a device search in Phoscon and then factory-resetting/reconecting the light. They will be found and mapped to the groups but all group/scene config has to be redone. Same for switches.

manup, 2019-01-26:
No reset shouldn't be needed.

Have you tried logging out and in again?
Also restarted the Raspberry Pi?

Sturmhorst, 2019-01-26:
i restarted the Pi, restarted Deconz and tried the login method

manup, 2019-01-26:
Hmm strange can you please create a backup in the Phoscon App and send it to mpi@dresden-elektronik.de 

I saw in the other issue you've updated from firmware version 0x26240500, I can check that all parameters are still valid.

eddy2501, 2019-01-27:
I have somehow the same problem....Running the Deconz app in Ubuntu 16.04LTS, REST API installed too...
After restart of my VM all lights and sensors missing in the Deconz app... Phoscon shows them but says not reachable.... Firmware is the latest.....

Sturmhorst, 2019-01-27:
manup helped me alot with this issue. Thanks

manup, 2019-01-27:
The underlying bug is in the firmware, we're hunting this for quite a while now and have pin pointed it last week. The next firmware version will contain a fix for that.

Problem:
* The parameters: panid, network key and sometimes channel get invalid values in RAM. Likely due memory corruption, or bad initialization after soft reset.
* More dramatic these invalid values will be written to NVRAM and are applied also on further restarts.

The database file which is also part of the backup luckily logs all changes to the configuration. That's the reason the original working parameters are not lost and can be restored to get the network in operational state again without the need to reset all devices. Currently this can be done only manually.

So beside the firmware bugfix we'll add a page to Phoscon App to easily inspect and revert the configuration to a working state.

eddy2501, 2019-01-27:
OK..
 When will the new firmware be available?

moritzschaefer, 2019-02-12:
I have a firmware version from the 30th of January. Should this include the fix already? I still observe the same issue. @manup how is it possible to manually restore the origional working parameters?

SebSemmi, 2019-02-13:
> I have a firmware version from the 30th of January. Should this include the fix already? I still observe the same issue. @manup how is it possible to manually restore the origional working parameters?

I have the same issue. How can I fix this manually? Would another sample backup help to fix this problem?

manup, 2019-02-13:
Hi, sorry for the delay, tomorrow version 2.05.59 will be released which contains a simple page to diagnose and revert to previous working Zigbee configuration just a button press.

The page will just list all known previous configurations and by pressing the Load button it will be applied.

The backups which I requested in the last weeks showed that either PANID, Channel or Network key were scrambled, so if that is the case, just select the previous known working version.

This is a screenshot of the page, note the many different configurations here are just from my tests, it will look very uniform in your setups, so that invalid configurations can be easily spottet.

[image]

The actual firmware fix to prevent the issue for good is planned for 2.05.60.

SebSemmi, 2019-02-13:
Hallo

Vielen Dank für die schnelle Hilfe! Welche Wege gibt es, diese Projekt zu unterstützen, abseits von einmal einen Conbee für 50€ kaufen?

Mit freundlichen Grüßen
--
Sebastian Semmler
Eva-Laube-Weg 4 | 14473 Potsdam
Tel: +4915140508965
www.semmler.pro

moritzschaefer, 2019-02-14:
Ok, till then I just won't reboot my Pi :)

eddy2501, 2019-03-09:
Dumb question... How can I update my stick with ubuntu? My phoscon app says 2.05.20 is the newest

moritzschaefer, 2019-03-15:
@manup Are there any updates on this? I would like to restart my Pi for several weeks now but don't want to fight with this bug after a restart..

manup, 2019-03-15:
You can install the 2.05.60 version now with the new firmware.

The bug shouldn't occur anymore. Also the above mentioned new page allows fixing the config easily even if the bug wold occur :)

butschi, 2019-04-27:
@manup Just happened to me with these FW versions:

```
Product ConBee II
Version 2.05.64 / 22/04/2019
Firmware 26490700
```

I will try to use the hidden page recovery thingy now …

manup, 2019-04-27:
The above bug was only related to  ConBee  and RaspBee not ConBee II.

Starting from scratch usually shouldn't be needed. Does the log shown anything useful?
Is the firmware shown as connected (aka the firmware version) in Phoscon App > Menu > Settings > Gateway?

butschi, 2019-04-30:
The FW version was shown just fine, the logs did not show anything that helped me, anyway.

What I did was what @rikroe suggested and repaired all lights and switches without removing them. The group configurations worked like before but I had to redo all the scenes, just like he experienced, as well.

If it is helpful I can hook you up with a before/after backup of the gateway.

bhansson, 2019-05-11:
Same problem here, had to remove and pair everything again. Where do I find that new page to restore previous settings?

Product ConBee
Version 2.05.64 / 4/22/2019
Firmware 260B0500

manup, 2019-05-12:
>Where do I find that new page to restore previous settings?

Currently only described in the respective release note:

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

>Product ConBee
>Version 2.05.64 / 4/22/2019
>Firmware 260B0500

I'd recommend to upgrading your ConBee firmware to 0x26330500 to prevent the problem happen again. If the update isn't shown in `Phoscon App > Menu > Settings > Gateway` you can update it manually:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

lordkev, 2019-06-22:
This happened to me as well with a ConBee II after rebooting the Pi running Hassio. Haven't been able to figure out how to get them back and I've already had to re-add everything once from the same thing happening. Resetting 15 bulbs every time I reboot isn't really an option...

jordyverbeek, 2019-07-12:
I have the same problem with ConBee II.
Firmware: 0x26490700
All lights are grayed out. The only error I see in the logs is about Invalid admin password hash.

Any news on this ?

Smanar, 2019-07-12:
Have you tried a previous backup, if you can ?

jordyverbeek, 2019-07-15:
> I have the same problem with ConBee II.
> Firmware: 0x26490700
> All lights are grayed out. The only error I see in the logs is about Invalid admin password hash.
> 
> Any news on this ?

This started happening after I rebooted my server. My OS messed up the ACM assignments, so deconz and zigbee2mqtt were both listening on /dev/ttyACM1.  Problem had nothing to do with deconz.

sberryman, 2019-07-26:
Where is this hidden page? I'm having the issue where all devices are listed in PWA but unreachable. Happened spontaneously yesterday. I have managed to add another (new) device which connects fine so it has to be some id changing.

edtwodth, 2019-09-01:
I have the same problem now with Deconz now ...  Conbee II, firmware 264A0700, hassio, raspberry Pi 3 b+ 

Just installed hassio today and started from scratch, after abot 2 hous alle lights where greyed out ...

moritzschaefer, 2019-09-01:
I get 

    /usr/bin/GCFFlasher_internal.bin: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.28' not found (required by /usr/lib/libwiringPi.so)

when I try to update the firmware as described here https://hub.docker.com/r/marthoc/deconz/

privatesam, 2019-09-14:
Same problem here, conbee II,  firmware 264A0700, hassio, raspberry Pi 3 b+, all devices greyed out.

sergiocaridi, 2019-09-15:
same problem here, different times:
- conbee II, firmware 264A0700, phoscom, pc, all devices greyed out (reincluded all the devices with on-off procedure.)
- conbee II, firmware 264A0700, phoscom, pc, included 30 devices, then moved to another area included 35 devices more. Then the 30 devices grayed out. (I have the pc and the conbee backup ar SW) 
- conbee II, firmware 264A0700, phoscom, raspberry Pi3, included 1 device, I've reinstalled  deconz.
the day after the lamp was gray. I've tried to move to different channel with no result.
 
[image]

sergiocaridi, 2019-09-15:
one more thing, is there a different network settings configuration to minimize this problem.
I realli need to find out a solution ASAP

[image]

manup, 2019-09-15:
The settings look fine. Can you provide more details on your devices, are these yellow devices Xiaomi routers?

Currently we can't recommend channel changes with Xiaomi devices, they don't seem to follow the change reliably.

sergiocaridi, 2019-09-15:
Thanks manup 
the router are device from Homa Shenzen.
In my project I'll have  toma and Sunricher (Philips)

[image]

sergiocaridi, 2019-09-16:
here is the situation.
I've restored the previus image (6) in  the " hidden zigbee configuration" and it works.

[image]

I turned off and on the rasberry and the problem was back again.
same process (again n.6) an it works.

sergiocaridi, 2019-09-17:
News... bad news.

I'm testing more and more. 
This time it doesn't work anymore.

[image]

old2 c is missing in action!

sergiocaridi, 2019-09-17:
Hey guys 
sorry to borther U. 
It was the socket unplugged this time.

I'm so sorry!

cinemarene, 2019-09-21:
Hello,
I have the same problem with my ConBee II (2.05.69 / 6.9.2019). This is probably the 10th time this has happened. Otherwise I have always learned the devices again. If it happens, then mostly spontaneously, without a reboot or enything else. The "Alt+Click" hidden area shows two configurations I can load. The previous one doesn't work and also a backup also not. Can anyone help me?

AlexKalopsia, 2019-09-22:
Today all my lights are grayed out (though they appear on the list. 
Running ConBee II attached to Synology NAS and deconz running in Docker

EndlessDelirium, 2019-10-17:
Same problem here. I've sent an email to their support but haven't heard back. It's quite incredible how they can't be bothered fixing such serious issues. It seems that they don't quite understand how annoying it is to re-pair numerous ZigBee devices. I even had to buy a Hue Dimmer just for being able to reset/re-add my Hue lights.
On my side this issue seems easy reproducible if you pull the plug after adding a device. What I do now is restarting the service and creating a backup whenever I add a new device.
(Generally something doesn't seem to be quite right after adding devices. It for example isn't possible to create backups until you restart the service)

JensTimmerman, 2019-10-19:
I also have the same problem, after rebooting the server with the conbee II stick in it all my devices are grayed out, nothing seems to help except resetting the gateway, resetting all lights, sensors and remotes and adding everything again. This is now the second time so I really would like a different solution.

The new network page only shows 1 entry for me, so restoring to previous working settings is not a solution here.

D3Troy, 2019-10-19:
Same problem here. Restarted my server and now i can't control anything since everything is grayed out. I have also only 1 entry in the network restore page just like @JensTimmerman.

I wrote to the support and hopefully there will be a fix soon. It's not worth using it like that.

brisketfingers, 2019-10-29:
Same issue: all devices unreachable/greyed out after Hass.io restart. Second time this has occurred in october. I restored a working backup configuration, but that did not have any affect. Had to reset and rejoin all devices.

BrianKlDK, 2019-11-03:
Same here I am afraid...

sberryman, 2019-11-03:
@everyone: This is a really annoying bug in the software or firmware. It happens to me every month or two.

1. Navigate to the Phoscon App (web page)
1. Go to settings and gateway
1. Hold down the alt key while left clicking on the advanced icon
1. Look for Network key and PANID that appears the most frequent. It should be very easy to spot in the list and click the `Load` button for the most recent Id. It is most likely the 2nd row.
1. Wait a few minutes and you'll start to see lights, switches and sensors reconnect.

FYI, I'm on version 2.05.66 and have seen the issue since at least 2.05.64.

ketzu, 2019-12-01:
Product: Conbee II
Using Docker container on RPi 3b+
Current firmware: 264A0700

Same problem here, electricity turned off, my lights and some sensors got greyed out. My xiaomi sensors are not greyed out, but I suspect they are not read properly anymore.

I managed to readadd some lights by deleting them and resetting them.

I updated the app, loading the network values from the hidden page entered a nice "segfault" entry to the log. Is there any update on the issue?

sverleysen, 2019-12-03:
I have the same issue, all the lights and sensors are gone after my power went out.
Tried to reload the configuration using the hidden page, but no success.

Is there any other way to reconnect my lights?

This is the 5th time this happened to me... Really frustrating...

michalzadkowski, 2019-12-17:
Same here. Conbee 2, Hassio, RPi3B+, all sensors lost connection and unreachable.

pawelbura, 2020-01-11:
I have the same:
Conbee 2, RPi4B+, home assistant and deconz in docker containers. Restart of deconz container cousing all sensors (all Xiaomi) not updating :(
only one configuration in hidden page

lordkev, 2020-01-12:
Sorry to say it, but my advice is to just ditch deconz and use zha built
into Home Assistant. It's been way more stable.

ezzq, 2020-01-13:
I have the same issue with the following Settings:
Version: 2.05.71 / 12.12.2019
Firmware 264A0700
Deconz direct on a PC or within Docker or HA Add-On.
After the first reboot and power off all devices are unreachable.
New Instanz and restore backup not resolving this issue.
Just added around 15 Devices and its a big issue for me.

zenzay, 2020-01-16:
Same or likewise problem.  I had it running just fine; lost connections to all devices after an update.

Version 2.05.72.
Firmware 264A0700
Tried on Raspberry Pi3b+, Pi4, Windows and a Macbook pro.  Even tried the image on the site on a Raspberry Pi.

I've reset all lights to factory settings (several times), tried reflashing firmware and everything in the book, but it just wont find any lights.

tdesplenter, 2020-01-17:
I also have the same issue: all lights became unavailable after rebooting my RPi 3 running Home Assistant. I've emailed support a few days ago but no reply.

@manup : communication is important - it'd be good to hear that you're at least aware of the issue and looking into it. There are a lot of related issues so it seems to affect quite a few people (#2015, #2023, #2035, #2273,  #2333 etc).

osos, 2020-01-20:
Also maybe this #2349 about channel changing - could be a pointer?

MegaMatze, 2020-01-21:
Contacted support just this evening. Having the same problem. After a power outage of my Raspberry Pi everything is grey and no more "green lines" to the lights. In Deconz Sensors and switches where entirely gone. After Backup restore, still the same. Grey sensors in phoscon but in deconz some sensors start to come but still not reachable in phoscon and not possible to see status in iobroker.... This makes the Conbee2 useless, if I have to repair every device after a poweroutage of the raspy

ultimateguy, 2020-01-25:
Same issue here. This insanely infuriating. I've spent two full evenings switching over from ZHA and HUSBZB-1 because I was having awful latency issues. I finally get all of my devices paired with Conbee II and after rebooting my Pi, they are all unavailable and don't work.

I'm starting to think that Zigbee with Home Assistant is a huge waste of time.

mountainsandcode, 2020-01-25:
Have had my setup crash two times while running Deconz through Docker on a Synology DS with Conbee II. Setup becoming inoperative upon any shutdown/ halt/ reboot/ power loss. Using the same hardware and setup on a Raspberry Pi (installation in Raspbian) WITHOUT docker I have no issues, even with power losses or shutdowns.

Probably makes sense if we try to standardize the reporting here to see if there is some sort of issue, e.g., while using docker

thenightfighter, 2020-01-27:
Same shit here... unbelievable that this bug is there for a year now and still not fixed...

ultimateguy, 2020-01-27:
I got my setup up and running. I'm not sure if this will help anyone but figured I would share just in case.

I have another USB stick on my Raspberry Pi in addition to the Conbee II. That other stick is a HUSBZB-1 that I use for my Z-Wave devices. The device path for my Z-Wave integration in Home Assistant was set to something like /dev/ttyAMA0. Problem is that Linux does not have a set order in which it loads USB devices on boot, so that device path will not always point to the same device if there are 2 or more USB devices connected.

I reconfigured BOTH the Z-Wave stick and the Conbee II using their "by-id" paths instead of the generic /dev/ttyAMA0 type of path. These can found by going to the Home Assistant UI -> Hass.io in the sidebar -> System tab -> Hardware.

The Conbee II path looks something like /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2126718-if00 and the HUSBZB-1 path looks like 
/dev/serial/by-id/usb-Silicon_Labs_HubZ_Smart_Home_Controller_513000FF-if00-port0. (There is another identical entry ending with if01-port0 for the Zigbee part of the HUSBZB-1, which I'm not using.)

This works so far. Hard rebooted and Deconz is still working. Fingers crossed that it stays this way.

sverleysen, 2020-01-28:
I also use the by-ID path but I still have the issue.

psanwald, 2020-01-29:
I ran into the same problem. Greyed out Zigbee items without functionality. Also the Raspberry is hanging on reboot/shutdown. After a while I have to pull the plug, to get the system up and running again.

Version 2.05.72 (12.12.2019)
Firmware 264A0700

[22 more replies]

closed 2020-06-26 by Mimiix