dresden-elektronik/deconz-rest-plugin #1187: IKEA Trådfri light bulbs receive multiple and repeated reachable true change events

mwhrtin, 2019-01-26:
Running deCONZ 2.05.57 and Rest API 1.16.0 together with HomeAssistant.

**Issue**
Several of my IKEA Trådfri light bulbs receives repeated reachable true change events after new lights were added and existing lights moved around. The deCONZ Rest API sends repeated `"reachable: true"` change events over the websocket for the same light.

What is happening here, does the lights interfere each other? Are deCONZ and/or the lights adjusting to the changes in the environment? Is this expected behavior for Zigbee lights?

**Note**
This is a fresh behavior, started only 2 hours ago and I haven't seen it the last 40 minutes. So perhaps it will solve itself, but if so, I'm still interested in what's happening and why.

**Description**
I've had what I considered a stable setup the last 14 days using three IKEA Trådfri light bulbs _(TRÅDFRI bulb E27 CWS opal 600lm)_. I've only received reachable events when I've powered the lights on or off using a power switch.

I added five additional IKEA Trådfri lights using Phoscon and moved the existing lights around. I ended up with 4 lights online in the same room (the rest are registered but offline). Three of my existing _"TRÅDFRI bulb E27 CWS opal 600lm"_ and a new _"TRÅDFRI bulb E14 opal 400lm"_. 

Shortly after I finished setting them up one of the existing _"TRÅDFRI bulb E27 CWS opal 600lm"_ started to receive `"reachable: true"` change events over the websocket. During 1 hour the light bulb received 9 `"reachable: true"` change events. I wouldn't have noticed it if it wasn't for the light resetting the color to the default one each time it happened. 

After one hour it appears as if the light was stabilized, instead the newly installed _"TRÅDFRI bulb E14 opal 400lm"_ started to receive `"reachable: true"` change events, 5 change events during a 15 minutes period.

Anything between 2 minutes to 20 minutes between each `"reachable":  true` change events.

ebaauw, 2019-01-26:
> I've only received reachable events when I've powered the lights on or off using a power switch.

Did you recently update deCONZ?  As of https://github.com/dresden-elektronik/deconz-rest-plugin/commit/17abb815b62ba1cc15bddf68cfd6753a61d76911 (in v2.05.56), devices are marked unreachable on staring deCONZ, so you would expect one `"reachable": true` event per light after starting deCONZ.

In case of Zigbee network issues, you might see additional events, but there should be a `"reachable": false` before a `"reachable": true`

> I wouldn't have noticed it if it wasn't for the light resetting the color to the default one each time it happened.

That would indicate the light firmware reboots.  It does send a _Device Announcement_ message on startup, which causes a `"reachable": true` event (without a preceding `"reachable": false`).  As far as I've experienced the IKEA firmware can hang (requiring a power cycle to remedy), but I haven't seen a firmware crash causing a reboot.  As it's multiple lights, I would rule out hardware failures and check for issues with your mains power.  Are all effected lights on the same circuit breaker or wall switch?

mwhrtin, 2019-01-28:
v2.05.57 was installed a day or two after its release, was previously on v2.05.55. 

After the first discovered hammering of `"reachable: true"` events it went silent for the rest of the evening. 

During the last 48 hours I can only find `"reachable: true"` events for the same _"TRÅDFRI bulb E27 CWS opal 600lm"_ light bulb. So the second light that also showed similar logs appears to have settled for now.

During the last 24 hours the _"TRÅDFRI bulb E27 CWS opal 600lm"_ light bulb is continuously sending `"reachable: true"` events, it's doing it throughout the day. The events are more or less grouped, it can be nothing for several hours and then within a 5 minutes span 5-6 events are sent, sometimes with only 10 seconds apart. 

The setup is a room with 4 table/floor lamps connected to wall outlets, the outlets are all on the same circuit breaker. The wiring is rather new (house built 8 years ago) and I've never experienced any other issues with unstable circuits. So if there is any hardware failure I'd suspect the floor lamp rather than the wiring or the outlet. 

The light bulb doing this is reported having `"firmware: 13000272"`while my other lights of the same model have `"firmware: 1.3.002"`. The light doesn't start the OTAU when I try to query it, so I've assumed they're on the same version, just that it's weirdly named. 

I'm going to switch the connected light bulbs in the room, one of the other lamp/bulb combo have never done this. So if the behavior continues after the switch I should (hopefully) be able to pin-point if it's the light bulb or the lamp that's the issue. Assuming it's one of them.

mwhrtin, 2019-01-29:
I can now rule out issues with a specific bulb, light firmware or the lamp it's connected to. 

Since my last post I've had `"reachable: true"` events from multiple rooms, floors and lights. In total 7 of my 8 connected lights have sent events, the only one that haven't so far is the one that's closest to the ConBee stick (1 meter).

It appears as if the `"reachable: true"` events are only happening for lights that are turned on, I've never seen logs from a turned off light.

During the last hours 6 of my 8 connected lights, located in different rooms, have sent `"reachable: true"` events. A new thing this time was that one of the lights had 21 occurrences of `"reachable: false"` followed by a `"reachable: true"` without any color reset. The rest did what I've seen before, a single `"reachable: true"` event followed by a color reset.

The Conbee stick is located next to one of my access points (bad idea?). I've got 8 lights and one Trådfri outlet connected in various rooms spread out over two floors, and the longest distance from the ConBee stick to a light is around 7-8 meters through walls/floor. 

I've never experienced any issues or delay when I try to turn a light on/off. But could this still be a range or inference issue?

I've got two 2.4GHz access points using 20Mhz channel bandwidth set to channel 1 and channel 13, ConBee is using Zigbee channel 15. So if I'm not mistaken the WiFI and ZigBee shouldn't interfere with each other.

mwhrtin, 2019-03-09:
I read somewhere else (don't remember where now) about another user that had similar issues with IKEA Trådfri (not sure if he was using deCONZ or Hue Bridge) but after a few weeks the issues had resolved themselves. 

For me the issues have almost completely disappeared, it's as if the IKEA Trådfri bulbs have stabilized themselves. When I first reported the issue it occurred at least 15-20 every day for random Trådfri lights (multiple models, sockets, lights). But since then I think it've slowly started to happen less and less frequent. Today I can't remember it even happening the last 2-3 weeks. 

I will continue to monitor this, especially since IKEA have released new firmwares that might change something.

closed 2019-07-14 by stale[bot]