dresden-elektronik/deconz-rest-plugin #1189: Problem with adding Hue Lights

akinomeroglu, 2019-01-28:
Hello,

I am using version .57 and have a problem when I try to add hue color lamps. I have 3 colored and one white spectrum hue lamps (first generation IMHO). I reset all the lights with hue dimmer (pressed on and off at the same time and lights flashed).

When I try to add lamps my raspbee only finds white spectrum lamps and can't find others. One of the colored lamps is in the line of sight (approx. 5 meters) and white spectrum is got a wall between raspbee. So I am thinking it is not about signal strength. 

I am using Tradfri gateway in my home I tried to change the channel. I tried 11 and 20 for avoiding interference but the result did not change.

My search duration is 60 minutes.

I could send any log if you need it. 

Thanks and regards,

--
Akın

manup, 2019-01-28:
Which firmware version is the RaspBee running? 
This is visible in the *Phoscon App > Manu > Settings > Gateway*

Do you have by any chance multiple tabs with the Phoscon App open, or on multiple devices?
Currently when joining lights or sensors only one instance of the Phoscon App should be open (as the others might close the Zigbee network).

akinomeroglu, 2019-01-28:
Hey manup,

Thanks for your help. My FW version is 262F0500

I tried again and now I am sure that app is only open in one device and tab (it doesn't solve the problem) but deconz gui is also open and I have an deconz integration with Home Assistant.

manup, 2019-01-28:
Firmware version looks fine. Not sure why the lights won't join, usually Hues join very easily.

>I reset all the lights with hue dimmer (pressed on and off at the same time and lights flashed).

Did the color bulbs also turn red? Note don't release the buttons too early, after flashing the light will do a reboot after a few seconds, only then release the buttons.

Oh, one important hint, before reset the lights via remote you need to power-cycle them. For security reasons recent Hue firmware doesn't allow to reset a light when it is running for some minutes (?).

akinomeroglu, 2019-01-28:
I tried again and it does not work but as a side not light won't flash red. Only white and daylight. Am I missing something? Thanks for your help.

manup, 2019-01-28:
Have you also power cycled the light, power off for a few seconds?

As alternative you can also reset lights via the old WebApp:

* *Phoscon App > Menu > Help > Open Wireless Light Control (2016)*
* *Menu > Settings*
* scroll down to Reset Devices via Touchlink

[image]

Note the lights need to be close to the gateway to be found.

If the lights show a network id of 0xfffe it means they are already reset, otherwise they are part of a network as shown in the screenshot.

akinomeroglu, 2019-01-28:
> Have you also power cycled the light, power off for a few seconds?
> 

Yes but no luck.

> As alternative you can also reset lights via the old WebApp:
> 
> * _Phoscon App > Menu > Help > Open Wireless Light Control (2016)_
> * _Menu > Settings_
> * scroll down to Reset Devices via Touchlink
> 

Thanks for the alternative method. Unfortunately it takes a huge effort in my setup but I am going to this at weekend if I can't solve the problem.

manup, 2019-01-28:
>Thanks for the alternative method. Unfortunately it takes a huge effort in my setup but I am going to this at weekend if I can't solve the problem.

You may try to scan anyway, 5 meters line of sight might still work.

akinomeroglu, 2019-01-28:
> > Thanks for the alternative method. Unfortunately it takes a huge effort in my setup but I am going to this at weekend if I can't solve the problem.
> 
> You may try to scan anyway, 5 meters line of sight might still work.

Oh cool. I just found the one of the lights. It is interesting LQI is between raspbee and light is just 61/52. I suspect that this could be signal problem. There is a Xiaomi climate sensor at the same place and it shows 252/0.

anyway I am going to try with other lamps and then close the issue. Thanks for your kind help @manup really appreciate it.

manup, 2019-01-28:
Hmm 61/52 is pretty weak signal, 180 and up are preferred. If the Raspberry Pi is near some other devices like WiFi router you may try to keep some space to prevent interference.

akinomeroglu, 2019-01-30:
Hello @manup,

I changed location of raspbee and problem solved a little bit. Thanks for your help.

closed 2019-01-30 by akinomeroglu

sorryusernameisalreadytaken, 2020-09-25:
> Have you also power cycled the light, power off for a few seconds?
>
> As alternative you can also reset lights via the old WebApp:
>
>     * _Phoscon App > Menu > Help > Open Wireless Light Control [...]

oh thank you for this. I paired for testing purposes a HUE_E27 with an v1_base from philips and after delete the HUE_E27 it was not able to pair it again with deconz. Your mentioned reset fixed it.

Is there a reason why this is not in the new UI?