dresden-elektronik/deconz-rest-plugin #1192: Group endpoint reports wrong mode and values for lights

Lokaltog, 2019-01-29:
When querying /groups, the API returns the wrong mode and values for lights. Sometime it corrects itself after updating the light mode (e.g. hs -> ct) or color.

This issue appears related to https://community.home-assistant.io/t/active-light-color-from-deconz-in-hass/34118, #415 and https://github.com/home-assistant/home-assistant/pull/15128. The issue causes deCONZ light groups to be displayed as blue in HASS when the colormode is `ct`.

Response from /groups:

```
  "10": {
    "action": {
      "bri": 0,
      "colormode": "hs",
      "ct": 0,
      "effect": "none",
      "hue": 0,
      "on": true,
      "sat": 127,
      "scene": null,
      "xy": [
        0,
        0
      ]
    },
    "devicemembership": [],
    "etag": "...",
    "id": "10",
    "lights": [
      "7",
      "11"
    ],
    "name": "Office",
    "scenes": [],
    "state": {
      "all_on": true,
      "any_on": true
    },
    "type": "LightGroup"
  },
```

Response from /lights/7 and /lights/11 (members of the group above):

```
{
  "ctmax": 526,
  "ctmin": 133,
  "etag": "...",
  "hascolor": true,
  "manufacturername": "OSRAM",
  "modelid": "CLA60 RGBW OSRAM",
  "name": "Office 1",
  "state": {
    "alert": "none",
    "bri": 1,
    "colormode": "ct",
    "ct": 390,
    "effect": "none",
    "hue": 0,
    "on": true,
    "reachable": true,
    "sat": 0,
    "xy": [
      0.380545,
      0.376913
    ]
  },
  "swversion": "V1.05.10",
  "type": "Extended color light",
  "uniqueid": "..."
}
```

The actual mode for both "office" lights is `ct`, but as the example above shows the /groups response contains default/fallback values in `actions`. HASS appears to interpret the group light color as having a mired value of 0, which translates to a bright blue color.

manup, 2019-01-29:
The action is meant to represent the last action sent to the group, not the state of lights (which might be very individual).

Lokaltog, 2019-01-29:
Thanks for the clarification. In that case it's the implementation in HASS which is wrong, so I'll report the issue there instead.

closed 2019-01-29 by Lokaltog

Lokaltog, 2019-01-29:
It's definitely a tricky problem, I guess the UI could display the actual color if every light has the same color, or a fallback color if there's any difference between the lights. I think the HASS implementation may have to be refactored to use data from the /lights endpoint instead of relying on the state returned from /groups to display light colors.