dresden-elektronik/deconz-rest-plugin #1193: Cannot pair iConnectHue with deCONZ 2.05.57

waywit, 2019-01-29:
Hi

actually I have issues with pairing my raspbee with deCONZ 2.05.57 and iConnectHue. Prior this has worked, but actually I cannot pair it anymore. Does anyone has some tips how to go on and get it work? What can I try? 
I already deleted iConnectHue and original Hue App, after that it will be found with correct IP... but after adding it has errors and cannot be used. After that the reaspbee bridge is not found anymore...

Thanks and best regards!

markbajaj, 2019-03-11:
I am new to deCONZ and am on 2.05.57. I cant get deCONZ to pair with any HUE app or iConnectHue either - iConnectHue finds the bridge but says unknown error. Does anyone know if this is possible to do?

ottelo9, 2019-04-18:
I have a Conbee 2 Stick and use Version 2.05.63 / 3.4.2019. I can find my Base with the Hue App but with iConnectHue I get the same error.

ottelo9, 2019-05-02:
The problem still exists with the newest version

bubffm, 2019-05-31:
I cant connect my RaspBee neither. deConz 2.05.65

closed 2019-09-06 by stale[bot]

alert82, 2023-02-09:
Does it work now ?