dresden-elektronik/deconz-rest-plugin #1194: Problems with Phoscon

Nobbbi, 2019-01-29:
Unfortunately, the switch editor is not reliable.

1st case
I have controlled 3 Ikea lamps with Hue or Ikea remote. There are different scenes on the buttons.
Switching the scenes is very unreliable. Most individual lamps remain in the old state or do not react. Sometimes the lights go off even though they should only change the brightness. Switching the scenes usually confuses everything. Turning everything off and on again is usually not a problem.

2 case
I have an Osram smartplug with a 15 minute timer and controlled with a Xiaomi single button. The scenario envisages that the Osram will be switched on and switch off automatically after 15 min. This does not work in most cases. Either the smart plug stays on for over 15 minutes or does not turn off at all. In very rare cases it works.

manup, 2019-01-29:
Usual first question: which deCONZ version are you running?

The switch editor only sends a command to the group that a scene should be activated, the scene is stored on the bulbs itself.

If this doesn't work well, please check if the scene is working properly in the scene editor. For Ikea lights for now I'd recommend setup each light individually not via group control and then save the scene.

> I have an Osram smartplug with a 15 minute timer and controlled with a Xiaomi single button. The scenario envisages that the Osram will be switched on and switch off automatically after 15 min. This [...]

I don't understand, how is this setup done? The switch editor doesn't support timers.

Nobbbi, 2019-01-29:
Hi Manup

It ist Build 57
Switching in the scene editor ist the same behavior and the lamps are setup individually not as a group. 

My fault not the switch editor, the timer setting in a group was mean

Nobby

manup, 2019-01-29:
Which Ikea light models are you using? Maybe I can recreate the issue, it's been a while since I've testes scenes with Ikea lights.

>My fault not the switch editor, i mean the timer setting in a group.

Can you please provide a screenshot how the switch is configured in the sensor control?

manup, 2019-01-29:
The versions of the Ikea lights would also be interesting, maybe they can be updated.

Nobbbi, 2019-01-29:
The Model of the Ikea Lights are TRÅDFRI Lampe E27 opal 1000lm with  V 1.2.214.
I have the feeling that the commands to the lamps are quite slow if i change the scenes.
Is there a way to check that or to tweak the zigbee network? 

Here some Screenshots of the Group with the Timer

[image]
[image]
[image]
[image]

Nobbbi, 2019-01-30:
Ok thank you for testing !
Did you change the scenes 1x, 2x and 3x klicks with the xiaomi button?

I am rotating the scenes with 2 buttons at a hue switch.
Could this make a differenz?

[image]

How can i check or tweak  that my zigbee network, is working with best results. 
As i wrote before i have the feeling that the communication between the devices are slow or somtimes the lamps are not getting the new seetings. If i look at the networkmap with vnc everything looks ok.

Nobbbi, 2019-01-30:
I read your hint in another bug entry, and changed it already  !

closed 2019-06-06 by stale[bot]