dresden-elektronik/deconz-rest-plugin #1195: Did something change for the websocket notifications

KodeCR, 2019-01-30:
I have an automation in homekit that turn on the lights when detecting motion from a Xiaomi motion sensor and turns them off again after 5 minutes. Using version 2.05.55 the timer seemed to be reset while there was motion and then turned off 5 minutes after there no longer was motion. After updating to 2.05.57 the light turn off 5 minutes after the first motion (even though homekit still shows the sensor detected motion). Did something change in what is sent in the websocket notification that homebridge-hue picks up?

manup, 2019-01-30:
Note the Xiaomi motion sensor issue is discussed in issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1178

2.05.58 arriving soon and has some fixes in that regard.

KodeCR, 2019-02-06:
Thanks @manup. I've installed 2.05.58 for a few days now, but the same behaviour is still happening. The status is correct in homekit in that it still shows it detected motion even if I've there for a while, but the lights still turn off 5 minutes after the initial motion is detected (with the sensor status still detecting motion).

KodeCR, 2019-02-18:
Ok, reverted all the way back to 2.05.50 and it was still happening, so this may be a change on Apple's side. I set up the motion behaviour for phoscon now and that seems to work fine.

closed 2019-02-18 by KodeCR