dresden-elektronik/deconz-rest-plugin #1196: 2.5.58 Enable firmware updates in Phoscon App when deCONZ is running via Hass.io

lubbertkramer, 2019-01-30:
Will this also be available for deconz docker users ?

manup, 2019-01-30:
I don't know, currently I think marthocs container only supports updates as described in the following guide. The Phoscon App update button was disabled for plain docker images since it would only run in cycles.

https://github.com/marthoc/docker-deconz#updating-conbeeraspbee-firmware

Not sure what is different but the Hass.io might have found another way to support the firmware updates in the same manner as native Linux does, maybe this can be adapted in marthocs docker image.

https://github.com/home-assistant/hassio-addons/tree/master/deconz

lubbertkramer, 2019-01-30:
Great and thanks for the reply I will ask him if he knows if this can be done

jamesking420, 2019-02-01:
Is there any value switching from your docker to the other one? Will you continue to maintain your original one?

pvizeli, 2019-02-07:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1221#issuecomment-461166038

closed 2019-06-14 by stale[bot]