dresden-elektronik/deconz-rest-plugin #1201: Hue outdoor motion sensor

MattL0, 2019-01-31:
https://www2.meethue.com/en-us/p/hue-outdoor-sensor/046677541736# 

Who id willing to try theses?  :o

fsandblom77, 2019-02-04:
Hi @MattL0 
I have two of these sensors. Bought them today.
It doesn´t work :) What would you like me to try?

ebaauw, 2019-02-04:
Could you pair one with deCONZ and post the screenshots as described in https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support.

From what I read, they're very much like the regular motion sensors, probably just another _Model Identifier_.

ebaauw, 2019-02-04:
Indeed, this looks all too familiar.  Could you try and read the _SW Build ID_ and _Date Code_ (double-click on the attribute to get a popup window).  Just need to whitelist the model identifier, at 17 places in the code...

fsandblom77, 2019-02-04:
Thanks! 
New C
[image]
luster Info screenshot..

ebaauw, 2019-02-04:
That does seem to be different (newer?) firmware.  My regular Hue motion sensors report 6.1.0.18912.

fsandblom77, 2019-02-04:
Yes, you are correct. I compared firmware version with my other regular motion sensors and I have the same version as you 6.1.0.18912 so it seems like they have a new firmware on the new outdoor motion sensor.

fsandblom77, 2019-02-05:
Just an update..
Updated to latest commit(Deconz)
I can now add the sensors from Phoscon. It is not visable in Phoscon but I can see it in Rest API. Also get presence detection, temperature and brightness.
So I'm happy because I'm using the Rest API.
Thanks @ebaauw for your effort!

ebaauw, 2019-02-08:
Also added support in homebridge-hue v0.11.14.

NedSin, 2019-02-14:
Hello together, 
is there any News or Information about an update for Phoscon who make the new Sensor Running ? 

Phoscon Version: 262F0500 

Best Greetings

SirMaggot, 2019-02-22:
Motion and temp is running just fine, but i cant get the lux to show anything else then 0.
Anyone else having the same issue?
Running DeCONZ 2.05.59 and firmware 26300500

olemr, 2019-02-27:
on .59
Just added the  HUE outdoor sensor from PWA.
Had one regular HUE sensor already added still with its default name: Motion Sensor.
After including the new outdoor sensor, the name of the original was changed to: Motion Sensor (2) and the newly included was not shown.
Had to rename the original and re-add. Now it works fine:
[image]

mrmoritz01, 2019-03-06:
Hi, I can confirm that the Outdoor Sensor is running fine with Phoscon Version: 262F0500 and deconz-2.05.59-qt5.deb from https://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.05.59-qt5.deb.

I updated using Step 4 described here: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

And added the motion sensor from the Phoscon Web App, worked perfect.
It did not work on .58.

closed 2019-07-11 by stale[bot]

caplam, 2019-12-12:
I paired a hue outdoor sensor and luminance value is not read.
The firmware is 6.1.0.25261.
Does someone else has the same firmware number ?

meerios, 2020-02-21:
I have 6.1.0.25261 as Firmware of the HUE outdoor sensor as well and the luminance value is not showing up.

Here is the important part of the GW Config:

```
"apiversion": "1.16.0"
"devicename": "ConBee II"
"fwversion": "0x264a0700"
"modelid": "deCONZ"
"swversion": "2.5.73"
```
Sensor Information:

```
"manufacturername": "Philips",
"modelid": "SML002",
"name": "HUE outdoor motion sensor",
"state": {
"dark": null,
"daylight": null,
"lastupdated": "none",
"lightlevel": null,
"lux": null
},
"swversion": "6.1.0.25261",
"type": "ZHALightLevel",
```

How could that be supported?

Jdruwe, 2020-06-09:
@meerios I have the luminance value also not showing up, did you find a fix?

meerios, 2020-06-10:
@Jdruwe Yes, see my comment in another related case: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1559#issuecomment-590332183

Jdruwe, 2020-06-10:
@meerios for some reason my luminance started working a couple hours later