dresden-elektronik/deconz-rest-plugin #1202: 2.05.58 Can't edit or delete light

ketiljo, 2019-01-31:
Just updated to version 2.05.58 from 2.05.54 to get the Xiaomi Aqara bulbs working correctly, and they do, thanks! But now I can't edit the name or delete any of the lights anymore. For switches and sensors, when you click in them in the Phoscon app under "Devices", you get to another window with some info where you can delete and rename.  That used to apply for lights as well, but it doesn't anymore. Nothing happens when I select the light, except that it blinks when I click the light bulb as usual.

I'm using the Deconz SD card image on a Raspberry Pi. 

I also noticed that the IP address shown in the upper right hand corner isn't correct. The Raspberry is at 192.168.1.13 while the address shown is 192.168.8.1.

ketiljo, 2019-02-04:
Should I try to reinstall? If so, can I just do a backup, reinstall the SD card and restore the backup?

manup, 2019-02-05:
Can you please open the browser developer console (F12) and check if there is any Javascript error shown? You may need to reload the page once the console is open.

Reinstall won't help here, might be just browser cache not up-to-date issue.

ketiljo, 2019-02-06:
I'm using Chrome. I've tried several computers, same result. The console shows one error: 
`Uncaught TypeError: Cannot read property 'indexOf' of null
    at n.addOne (devices-lights.js?b137f7:3)
    at underscore-min.js?2a5156:5
    at Function.m.each.m.forEach (underscore-min.js?2a5156:5)
    at n.forEach (backbone-min.js?cedae4:1)
    at n.initialize (devices-lights.js?b137f7:3)
    at n.e.View (backbone-min.js?cedae4:1)
    at new n (backbone-min.js?cedae4:1)
    at setGui (devices-lights.js?b137f7:3)
    at HTMLDocument.<anonymous> (devices-lights.js?b137f7:3)
    at j (jquery-min.js?0baf29:2)`

And one warning: 
`Query.Deferred exception: Cannot read property 'indexOf' of null TypeError: Cannot read property 'indexOf' of null
    at n.addOne (http://phoscon.lan/pwa/js/devices-lights.js?b137f7:3:5599)
    at http://phoscon.lan/pwa/library/js/underscore-min.js?2a5156:5:1507
    at Function.m.each.m.forEach (http://phoscon.lan/pwa/library/js/underscore-min.js?2a5156:5:2350)
    at n.forEach (http://phoscon.lan/pwa/library/js/backbone-min.js?cedae4:1:774)
    at n.initialize (http://phoscon.lan/pwa/js/devices-lights.js?b137f7:3:5064)
    at n.e.View (http://phoscon.lan/pwa/library/js/backbone-min.js?cedae4:1:14744)
    at new n (http://phoscon.lan/pwa/library/js/backbone-min.js?cedae4:1:22931)
    at setGui (http://phoscon.lan/pwa/js/devices-lights.js?b137f7:3:8934)
    at HTMLDocument.<anonymous> (http://phoscon.lan/pwa/js/devices-lights.js?b137f7:3:13014)
    at j (http://phoscon.lan/pwa/library/js/jquery-min.js?0baf29:2:29999) undefined`

I have also tried clearing the cache, no luck. Thanks.

manup, 2019-02-06:
Thanks that should help to find the error, should be fixed in 2.05.59

manup, 2019-02-08:
Checked the code, the error means the lights doesn't have a `type` attribute.
This will be guarded in next version but still the questions remains why the light doesn't have a type attribute.

ketiljo, 2019-02-08:
I did add a few Xiaomi bulbs right after upgrading. They do behave as they should with this version

manup, 2019-02-15:
2.05.59 is available now

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

closed 2019-02-15 by manup