dresden-elektronik/deconz-rest-plugin #1203: New TRADFRI on/off switch adds an extra space in the end to the default name

tubalainen, 2019-02-01:
When a new remote control switch of the type IKEA On/off is added an extra space in the default name is being added in the end.

It is not only cosmetic, I can possibly cause issues with home assistant via the API and the events functions now with the new slug changes.

`"TRADFRI on/off switch "`

[image]

manup, 2019-02-01:
Hmm interesting, normally we trim modelid strings for exactly that reason. I'll check the code if there is something missing.

tubalainen, 2019-02-01:
> Hmm interesting, normally we trim modelid strings for exactly that reason. I'll check the code if there is something missing.

Can this be because the switch has been joined prior to the .58 release?

Ive got another one in store, will test later today and see if I get the same results. :)

tubalainen, 2019-02-04:
Sorry for late reply ... Job and kids ...

Just tried to add another on/off button and it got the same issue with the extra space in the end.
I added it via "switches - add switch - IKEA - (pushed the round five button one) then 4x tapped the join button.. Waited for it to appear in the background then closed the "add switch popup".

tubalainen, 2019-02-15:
Issue solved in .59 release. Thanks! <3

closed 2019-02-15 by tubalainen