dresden-elektronik/deconz-rest-plugin #1206: sensor integration

banabo, 2019-02-02:
Hi all,
I have an environment sensor containing the following measurement types:
- Pressure
- Humidity 
- Temperature
- Carbon dioxide
- Light Level
- UV index
A body weight should also be transmitted by the same enddevice. 

I want to extend the sensor with a ZigBee interface.
I could imagine deconz as the concentrator software with the rest plugin as interface to my application
running on a gateway, lights should also be controlled with this gateway.

What would be the best way to achieve 
a) the end device becomes whitelisted in the rest plugin 
b) integrate value types (CO2, UV index, body weight ) which are currently not available in HA profile

ebaauw, 2019-02-02:
> I want to extend the sensor with a ZigBee interface.

Cool, but deCONZ won’t be of help here; it _is_ the gateway, exposing Zigbee devices, so they can be controlled over the REST API.  You need an (embedded) application and Zigbee radio hardware that exposes the sensors over Zigbee, so they can connect to deCONZ.  For the ZigBee radio hardware, you could use another RaspBee or ConBee, but it would run your embedded application over the serial interface, not deCONZ.

> the end device becomes whitelisted in the rest plugin

We would need to know the mac address prefix (of the Zigbee radio hardware), and the _Manufacturer Code_ and _Model Identifier_ as exposed by your embedded application.

> integrate value types (CO2, UV index, body weight ) which are currently not available in HA profile

We already support CO2 sensors.  Typically they expose themselves through the _IAS Zone_ cluster.  Thye provide only a binary alarm though, not the CO2 level.  For UV index you could do the same; for body weight you could use an _Analogue Input_ cluster (as for the CO2 and UV index, if you want to expose the level).  We could create additional sensor resource types, or create a ZHAGenericStatus.

ebaauw, 2019-02-02:
No, you would just see a `status` value in the websocket.  If you want to see `weight`, `co2level`, and `uvindex`, we would create separate ZHAWeight, ZHACO2Level, and ZHAUVIndex sensor types (and their CLIP equivalents) with corresponding `state` attributes.

banabo, 2019-02-02:
Sounds great:

> we would create separate ZHAWeight, ZHACarbonDioxide, and ZHAUVIndex 

What cluster id's should we use ?

manup, 2019-02-02:
I'd recommend checking the Zigbee Cluster Library (ZCL) specification if there are some matching clusters. Maybe the Analog Input Cluster is a good fit to provide some of these values.

closed 2019-06-09 by stale[bot]