dresden-elektronik/deconz-rest-plugin #1207: Lights unreachable at start with 2.05.58

wvuyk, 2019-02-02:
Starting with 2.0558 lights are `unreachable` when deCONZ starts? 
I do not know why this change has been done, but this is not nice behaviour with my plugin. Since long the plugin has the ability to restore light settings if they change from being unreachable to reachable. 

So it now happens that lights go on if deCONZ is restarted, where they should not be on at all. Is it possible to start deCONZ with the last known status instead of `unreachable?`

manup, 2019-02-02:
>Starting with 2.05.58 lights are unreachable when deCONZ starts?

Yes this was done to mitigate another problem. Lights were reachable but couldn't be controlled until they were discovered again.

>Is it possible to start deCONZ with the last known status instead of unreachable?

Yes this is already on the 2.05.59 todo list, the issue was reported already.
The restore will be done like for sensors.

wvuyk, 2019-02-02:
That sounds great to me! Thanks!

manup, 2019-02-02:
>So it now happens that lights go on if deCONZ is restarted and resets lights, where they should not be on at all.

Just understood this part now. Big sorry, totally haven't thought about the impact on such features.

MattL0, 2019-02-04:
This is not just that. 

When i restart deconz I have to mannually unplug some hue light before beeing able to control them again.

wvuyk, 2019-02-04:
@MattL0 With a restart of deCONZ it can take a few minutes before the lights can be controlled. I can see the same here, it can take up to 3 minutes here before all lights respond. 
Does your Hue light stay uncontrollable after 5 minutes or so?

MattL0, 2019-02-04:
Yes

WhistleMaster, 2019-02-04:
Something related, somehow, with #1213 ?

MattL0, 2019-02-05:
> @MattL0 With a restart of deCONZ it can take a few minutes before the lights can be controlled. I can see the same here, it can take up to 3 minutes here before all lights respond.
> Does your Hue light stay uncontrollable after 5 minutes or so?

5-8 minutes i think

MattL0, 2019-02-05:
But,...In the phoscon app, some are ''unreachable'' after 25 mins. But i can control them with the grouping function.

MattL0, 2019-02-05:
One hue light is not controlable at all after 25 minutes. This one has no group assigned.

MattL0, 2019-02-05:
What firmware and deconz version do you recommend me to get a stable networks as before ( I do not remember) thanks

MattL0, 2019-02-05:
Thanks manup!!

MattL0, 2019-02-07:
with .58 

Removed the ability of my sensors to report light Level and temps. 

The lights seems to reachable a lot sooner !

closed 2019-06-14 by stale[bot]