dresden-elektronik/deconz-rest-plugin #1208: Device support request: GreenWave Energy monitor

endor-force, 2019-02-02:
Hi,
Just picked up an old Zigbee Greenwave RA11D51-A11-GR
Seems to be manufactured 2013.

Based on TI CC2530.

I found a serial interface that outputs data on startup. 
It does not parse any data from my PC or output anything else.

Serial Output on boot: 
``` 
SN: GR0B-1108-00XXX
IEEE: D4A92802000000D2
Key: 00000000000000000000000000000000
FW: 0.63
HW: 3
CH: 0x6718000
Z-Stack: 2.5.1
P1 type: 1, scale: u, res: u, base: u, count: u
P1 on: 3000 us, poll: 30 ms
P1 IR state: 0, cal: 0, thres: 0
P2 type: 4, scale: u, res: u, base: u, count: u
P2 on: 500 us, poll: 50 ms
P2 IR state: 0, cal: 0, thres: 0
Meter state: 0
```
I managed to reset key from a previous value and now it is just 0:s .
It is showing up i deConz with the IEEE as MAC.

[image]

Data in node info:
[image]

Nothing in Cluster info.

This is all the information i have now. 
Debug logs are not showing very much

```11:31:23:880 Power_Descriptor_req zdpSeq: 211 to 0xD4A92802000000D2
11:31:23:881 APS-DATA.request id: 244, addrmode: 0x02, addr: 0x9dc1, profile: 0x0000, cluster: 0x0003, ep: 0x00 -> 0x00 queue: 1 len: 3 tx.options 0x04
11:31:23:924 ZDP skip fetch, node 0x212EFFFF027781 total unconfirmed zdp request count 2 [2]
11:31:23:000 ZDP skip fetch, node 0x212EFFFF027781 total unconfirmed zdp request count 2 [2]```

```11:32:52:769 aps request id: 244 prf: 0x0000 cl: 0x0003 timeout (confirmed: 0) to 0xD4A92802000000D2 (0x9DC1)
11:32:52:849 ZDP skip fetch, node 0x212EFFFF027781 total unconfirmed zdp request count 2 [2]
11:32:52:849 aps request id: 244 prf: 0x0000 cl: 0x0003 timeout (confirmed: 0) to 0xD4A92802000000D2 (0x9DC1)
11:32:52:927 ZDP skip fetch, node 0x212EFFFF027781 total unconfirmed zdp request count 2 [2]```

Is there any hope for this?

manup, 2019-02-02:
Have you joined it while sensor search is running? This is needed in order to get the descriptors fetched.

endor-force, 2019-02-02:
I have activated sensor search. There is very little documentation for the GreenWave device so i'm not sure how to set it to join.

There is one button and one LED. 
After reset the LED is not blinking by default, pressing the button makes it blink green and after a few min it blinks red once and then stops blinking.. my guess is that this is the pairing mode that initializes.

manup, 2019-02-02:
>A friend has one of these devices and has sniffed data with wireshark.. not sure if there is anything we could extract from that?

Maybe, sniffer logs are often very helpful.

Does the device node blink blue or green when the button is pressed?

endor-force, 2019-02-02:
Sniffer log: 
https://send.firefox.com/download/a3e8f37b84/#0Pm8RbXQvp2Tddob3cKzUQ

I can see a very few blue lights on the node, and also a few red.

manup, 2019-02-02:
It looks like the device only broadcasts some data in the network but never is been queried.
Since the encryption key is unknown I can't see the content of the broadcast.

However, in your setup you can start deCONZ with logging, perhaps something is visible.

    deCONZ --dbg-aps=2

endor-force, 2019-02-06:
It seems like this is a Jannet protocol device. I guess it is not supported by deconz/Conbee?

Found some config details: 
JENNIC_KEY=::D4:A9:28:00:DF:90
JENNIC_CHANNEL=15
JENNIC_PAN_ID=0x7ef3

endor-force, 2019-06-06:
Closing, this device was confirmed to be jannet and would not work with deconz.

closed 2019-06-06 by endor-force