dresden-elektronik/deconz-rest-plugin #1211: Issue with Keen Smart Vent

snjoetw, 2019-02-02:
I’ve been using Keen Smart vent with Deconz and it works great. I’m able to control the vents and view the temperature from Home Assistant. Thanks for adding the support.

However, I noticed the pressure sensor and battery level doesn’t seem to work.
For the pressure sensor, it always show 0 in Home Assistant. But if I go to Zigbee mesh -> Select the node -> view pressure cluster info and click read, it’d fetch pressure value and I can see that from HomeAssistant. So looks like pressure is not fetched automatically like temperature?

As for the battery level, it’s in the similar boat, I only get updated value is I manually hit Read in Zigbee mesh.

Please let me know if you need further info to address this. Thanks!

staraxis, 2019-05-02:
Hi @snjoetw I'll second this issue. I have had my keen vents working well in deconz for almost a year but on 4 out of the 5 nodes I have to go into the deconz gui and manually refresh values for both the temperature and humidity - they remain at zero and show as 'unavailable; in home assistant. Is it possible for this to be looked at? I've waiting a while as I thought it might have been an aberration but removing and re-adding the nodes doesn't yield a permanent fix either.

Is there anything logs/info I can provide for @manup or @ebaauw to assist in debugging? Thanks

ebaauw, 2019-05-02:
Sounds like attribute reporting hasn’t been setup.  As these are battery-powered setting it up during pairing might be challenging, although apparently, they poll their parent quite frequently (or controlling the vent wouldn’t be responsive).  I’m not even sure if the vent supports attribute reporting, maybe we should poll the vent for battery percentage, as we do for the iCasa keypads.

Could you try and manually setup in the deCONZ GUI the bindings from the appropriate cluster to the coordinator, and the attribute reporting configuration?  And see if temperature, humidity, and battery get updated automatically?  Typically temperature and humidity are setup to report every 5 minutes (300 seconds), and battery every 2 hours (7200 settings).  You want to be conservative here, not to drain the battery.  Also, the vent might take measures only at predefined intervals, you wouldn’t want to configure the reporting more frequently.

Ltek, 2019-07-28:
You guys get this working? I just got 6 vents, 3 installed so far. Temp & Battery seem to read fine, Pressure is always Zero

closed 2019-12-02 by stale[bot]