dresden-elektronik/deconz-rest-plugin #1212: Lights not connecting after backup and restore.

OakNinja, 2019-02-03:
Yesterday, I tried to migrate from Windows 10 to another PC running Ubuntu, so I:

- Upgraded deCONZ on Windows 10 to the latest release,  Version 2.05.58 / 1/30/2019, 
- Took a backup from Phoscon
- Installed deCONZ Version 2.05.58 on Ubuntu
- Physically moved my Conbee stick to the new machine

The conbee stick seems to be working, and all the devices is visible both in Phoscon and in deCONZ, but I cannot control any of them (nothing happens when i change the state of any node).

Firmware was one release behind (262F...) but Phoscon said it was up to date so I updated it manually to 26300500, but unfortunately, it made no difference. I've tried rebooting a few times, and I've tried running the HASS.IO image for deCONZ in docker as well, and tried a manual migration via the file system but nothing works.

What am I missing here?

manup, 2019-02-03:
Hmm, strange. Can you please try copy manually the zll.db file from the original machine to the new machine?

>The conbee stick seems to be working, and all the devices is visible both in Phoscon and in deCONZ, but I cannot control any of them (nothing happens when i change the state of any node).

On the Ubuntu machine is the firmware version shown in the Phoscon App Gateway Settings? (meaning connection works).

manup, 2019-02-03:
>The conbee stick seems to be working, and all the devices is visible both in Phoscon and in deCONZ, but I cannot control any of them (nothing happens when i change the state of any node).

Do you mean controlling lights?

Note after starting deCONZ it can take a bit until lights become controllable (they need to be seen by the controller). How long did you let it run?

Are the lights shown in the deCONZ GUI and do the nodes blink blue from time to time?

OakNinja, 2019-02-03:
Yeah, I see the lights (some not available) but not possible to change their state. 

The nodes blink blue from time to time. It seem to me like the mesh does not let the conbee controller ”in”, like there’s a mismatch in keys or something missing. I don't know though. Theres errors ending with hex Numbers, like 0xa7 and others. I left it over night but it didnt help. (My 4 months old baby helped me  write this prose)

OakNinja, 2019-02-03:
It’s a NUC so it could definitely be some interference going on, but wouldn’t it at least be intermittent? I’ll try with an extension cable, but right now I’ll just tuck that windows 10 laptop into the closet and pretend it worked as planned. 

I hoped it was a known issue with a crystal clear solution - thanks for all the help.

manup, 2019-02-03:
>I hoped it was a known issue with a crystal clear solution - thanks for all the help.

Maybe :) for some the USB extension cable really did wonders, chances are good it might help here too.

thomashup, 2019-02-05:
I have the same problem, when i wanted to migrate to the Deconz plugin in Hassos.

I have a RPi2 running Deconz image, and a RPi 3B+ running the latest Hassos.

I did as OakNinja explained, but nothing happend when i tried to use the ConBee in hassos.

Then i moved back the ConBee stick to the Deconz RPi, but after that i couldn't control my light and no matter what i'm doing, nothing happens. I downloaded the image and tried to start from scratch, but my Deconz stille does'nt find any light or any of my Sensors.

Before it worked okay, and the two RPi are next to each other, so i don't believe it has something to do with bad signal.

Sorry if it's not the same problem and i hijacked the threat.

closed 2019-06-12 by stale[bot]