dresden-elektronik/deconz-rest-plugin #1213: Recommended firmware?

roflmao, 2019-02-03:
Which firmware is recommended at this time?
After upgrading my Conbee to:

```Version 2.05.58 / 1/30/2019
Firmware 262F0500```

I have lots of problems with my network, lights not responding(ikea/hue), sensors not responding (xiaomi). Previously i had an old firmware which didnt have these kind of instabilities (but i cant remember the version because of the HEX coding, heh)

Nobbbi, 2019-02-04:
unfortunately I have the same problem.

Oliviakrkk, 2019-02-04:
I have the same issue with Xiaomi Temp and Smoke sensors :/.  But Xiaomi round button started working flawlessly with this conbee firmware.

Nobbbi, 2019-02-04:
For me the biggest Problem are that the lamps (ikea tradfri) are not very reliable, if i switch some scenes. sometimes it works - mostly not. But i am noch using the conbee usb stick. i have the raspberry add on modul. it feels that something with the zigbee network is wrong or a least not as good as should be. 
My mesh network is big enough in a smal flat. 
but i have no idea where the search.

Oliviakrkk, 2019-02-04:
I just looked into deconz GUI via VNC (I am using docker container) and I can confirm that also Philips Hue Iris is dropping. 

For now I have only one Hue device moved from Hue hub...I will move some more light and will see.

WhistleMaster, 2019-02-04:
Same here... Since version 2.05.58 and firmware 0x262f0500, I have also issues with my Philips Hue lights and sensors not responding...

manup, 2019-02-04:
Hmm strange the 0x262f0500 firmware should be way more stable than any previous firmware. It has only one known bug (which is a really bad one and corrupts network parameters).

I'm not sure if the issues mentioned here are related to this bug. If you can send me a backup file to mpi@dresden-elektronik.de I can check what's wrong (please refer this issue in the mail).

The bug will be fixed in the next firmware version (ETA 2.05.60/61). In the next version 2.05.59 the Phoscon App will provide an extra page to repair corrupted network parameters to the previous working state.

MattL0, 2019-02-05:
here running deCONZ_Rpi_0x26300500.bin.GCF , is that a good one?

roflmao, 2019-02-09:
After running for some time it seems like my network has healed itself? Everything is connected now for quite some time, and responding. (i dont have a GUI other than Phoscon)

manup, 2019-02-15:
2.05.59 is available now, which contains Zigbee network diagnose page

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

Closing this issue for now.

closed 2019-02-15 by manup