dresden-elektronik/deconz-rest-plugin #1216: Deletes sensors visible in Home Assistant

Oliviakrkk, 2019-02-04:
Hi

I deleted sensors in Phoscon UI and checked via api (POST /api/<apikey>/sensors) that removed sensors are not visible anymore.

But sensors (cube and vibration) are still shown in home assistant:
[image]

How can I remove those sensors?

Kane610, 2019-02-04:
Have you tried restarting Home assistant?

Oliviakrkk, 2019-02-04:
Yes.  Multiple times. And manually run services deconz.device_refresh. Neither helped. 

I even removed deconz integration and readded it - ghosts are still there.

Kane610, 2019-02-04:
It should be. There are some comments in other issues about removing items from the database

Oliviakrkk, 2019-02-04:
Looks like this solved the issue. Question is why API did not show those entries...

Kane610, 2019-02-04:
It is marked as deleted, but for some reason the Rest API doesn't honor that

manup, 2019-02-05:
The REST-API doesn't return sensors which are marked as deleted. Sensors which are marked as deleted in the database are further not loaded on startup.

So in theory these entries shouldn't make any trouble.

forast3r, 2019-02-06:
@Oliviakrkk You can also delete the old (deleted) sensors in HA's core.device_registry

Be careful!

Oliviakrkk, 2019-02-06:
Good point. If I encounter another ghost entries I will try that.

Thanks!

snjoetw, 2019-02-06:
@forast3r I had to do this previously when I remove a device from deconz. 

But I had to also modify deconz DB to remove stale entries as well. I think this was fixed in recent release?

Oliviakrkk, 2019-02-07:
I guess not, because I am on most recent one and I had issues.

lbschenkel, 2019-02-07:
@Kane610: I can confirm that I have deleted devices showing in HASS (lights, sensors and switches) after I deleted them from deCONZ. I checked that they're not visible in the deCONZ REST API (confirmed with `curl`), they're not present in the deCONZ DB (rows for sensors were forcefully removed from the DB), I have restarted HASS multiple times, and yet these deleted devices show in the Config/Integrations page and remain in `core.device_registry`.

Should I file a bug in HASS?

P.S.: I tried to find the code that removes devices which are no longer exposed in the REST API from the HASS device registry to better understand the logic but I was not able to find it in https://github.com/home-assistant/home-assistant/tree/master/homeassistant/components/deconz. Did I miss it somewhere?

ottelo9, 2019-05-13:
How can I delete old ghost groups/lights fro mthe DB? I use home assistant and deconz plugin.

ottelo9, 2019-05-17:
Ok I deleted it via the REST_API (followed the tutorial [here ](http://dresden-elektronik.github.io/deconz-rest-doc/) and [here](http://dresden-elektronik.github.io/deconz-rest-doc/sensors/) -> DELETE Sensor)

closed 2019-09-21 by stale[bot]