dresden-elektronik/deconz-rest-plugin #1217: OTAU only works after restart

lbschenkel, 2019-02-04:
This is probably the same issue reported on https://github.com/dresden-elektronik/deconz-rest-plugin/issues/947.

I have recently updated all my power-connected IKEA devices to latest test firmware (from `fw.test.ota.homesmart.ikea.net`) and I had the chance to retest the OTAU functionality from deCONZ GUI.

I found out that no matter what I did (pressing "query" and/or selecting the right firmware file and pressing "update"), I could only perform a firmware upgrade after a deCONZ restart. Without a restart, or once a device is upgraded, deCONZ OTAU functionality stays on what looks like a "stuck" state. I noticed the following on the logs when this happens:

```
Send query next image response: OTAU_NO_IMAGE_AVAILABLE (sensors busy)
```

Once this shows up on the logs, only a restart can restore the proper OTAU functionality. I did not find this string anywhere in this repository, so I'm assuming it comes from the closed-source deCONZ binary and not the REST plugin.

When I see this, and I restart, either the firmware upgrade starts automatically by itself or immediately once I press "query" or select the file + "update". That's how I ended up managing to update the devices (but I had to restart deCONZ once per device).

I tried playing with `--otau-notify-delay` which I found at https://github.com/dresden-elektronik/deconz-rest-plugin/blob/1cc9e0f80670a42d8498a1737c3feae2c45b8f0d/de_otau.cpp#L51, but I saw no apparent differences when I tried the values of `5` and `10` compared to the default of `60`.

Also, no matter what I do, I cannot trigger my battery-powered IKEA devices to upgrade. I tried triggering it in deCONZ and taking the battery out and putting it back in, but no luck so far.

Any tips on how to troubleshoot this further, @manup?

closed 2019-06-11 by stale[bot]

realwax, 2020-02-26:
Can we reopen this please?

popy2k14, 2020-05-18:
Same issue here with current stable deconz.
After updating an ikea blind, the otau plugin seems stucked and no query/update works.
After restarting deconz-gui, i can update the next one...

manup, 2021-10-13:
The OTA plugin is currently in rework for various improvements and will be updated by one of the 2.13.x updates.
For end devices it's more tricky since these sleep most of the time, but mains powered devices should trigger the update within seconds.