dresden-elektronik/deconz-rest-plugin #1219: Can't get Gledopto bulbs to detect at all!

mredz, 2019-02-05:
I have 4 x GL-B-001Z candle bulbs. I am trying to connect them to a Conbee. 
My firmware is this:
Version 2.05.20 / 4/8/2018
Firmware 261F0500 

I try to scan and I don't get any response from any of the globes. 
I have the conbee on a USB cable right on top of the light. 
Also I have tried debugging on Windows and there is no feedback in the console?
deCONZ -platform minimal --dbg-info=2 --dbg-aps=1 --dbg-error=1 --http-port=80

How can I dubug this please? I have a computer science background and I'm happy to assist with any debugging or compatibility issues.

ebaauw, 2019-02-05:
Make sure the bulb is in factory new state; reset it if needed.  Search for new lights from Phoscon and then power cycle the bulb.  When the device joins the network, a new node should appear in the deCONZ GUI, but sometimes its outside the current view or hidden behind another node.  Check the _Node Info_ panel to be sure.  Once the REST API resource is created, the node name in the GUI should change to `Light 1` or something.

Don’t use _Permit Join_ from the GUI: the device will join the network, but no REST API resources will be created.

manup, 2019-02-05:
You need to reset the GLEDOPTO controller while light search is running.

But before that, please update to a recent version 2.05.58 of deCONZ:

https://github.com/dresden-elektronik/deconz-rest-plugin#precompiled-deconz-packages

mredz, 2019-02-05:
Update: Still no joy. I have updated to: 2.05.58. I have tried both a power cycle and a reset and there is nothing registering. 
[image]
The Node info is as above. I presume that is the conBee? The Node list only shows one item.

mredz, 2019-02-05:
Could I please get a pointer on how to debug please? I understand there is a zigbee utility for this?

manup, 2019-02-05:
Debugging on windows is tricky. How do you reset the GLEDOPTO controller?

Here is the reset sequence shown in a video:
https://www.youtube.com/watch?v=YbeT72nTmn0

Note light search needs to be started in the Phoscon App before starting the sequence.

mredz, 2019-02-06:
Thanks yes I have tried this thanks. How can I test the conbee is working?

manup, 2019-02-06:
When the blue note is shown as in your screenshot, ConBee is operating and the network is up.

We can check the network settings if they are valid *Edit > Network Settings*

mredz, 2019-02-06:
[image]
Thanks. I'm on a second PC now so should be default settings.

mredz, 2019-02-06:
Ok I have booted into Ubuntu for testing and I'm running in debug mode. Here is a capture of the debug output: https://pastebin.com/HqHzqtLu
Can you please advise where I go from here?

manup, 2019-02-06:
The settings are not valid, wondering how that happened.

Please try to reset the gateway as described here: https://doc.phoscon.de/app/doc.html#settings-gateway-advanced-en

This should restore some valid settings.
If that doesn't work we can also do that manually.

mredz, 2019-02-06:
Ok thanks Manuel! Big leap forward. I reset the controller as instructed and I have the lights deteted now though the Manufacturer and model id seems to be unknown. Light 4 in the diagram is not a Gledopto it is a zigbee relay I had lying around so please ignore this. I don't seem to have controls for the lights yet. Where to from here please? 
[image]

mredz, 2019-02-07:
Ok I just went back to a fresh install on the depolyment server and it now seems to be sweet. Thanks guys for gettign me up and running. Sorry for all the beginers questions. 
[image]

closed 2019-02-15 by manup