dresden-elektronik/deconz-rest-plugin #1220: Strange behaviour with Trådfri motion sensors

ademobile, 2019-02-05:
I have been having an issue where my Trådfri motion sensors stop working (batteries have been changed) and they keep appearing like this:

[image]

Currently I am using 2.05.58, however I had the issue before this release too. Has something changed to report these in the GUI as I am sure when I delete these entries, this is when they stop being reported via the API?

manup, 2019-02-05:
Hmm this should only happen when the sensors are reset back to factory settings.

The sensors broadcast to a group, which is captured by deCONZ and linked to belong to the sensor.
If that group changes (this should only happen with due a reset; pressing the sensor button 4 times) deCONZ does release the former group. It won't be deleted but ends up displayed as normal group as shown above.

So I guess we need to compare the REST-API output for the sensor (here the `config.group` attribute) and if these groups are related to it.

ademobile, 2019-02-05:
Being a bit slow, do you need more information from me, and if so, what would you like?

ademobile, 2019-02-08:
I looked at the old version of the web interface and I notice in that I have 4 sensors:
[image]
[image]
[image]

Yet only 3 exist. I had an issue today where one of the sensors was no longer being reported in the API or Phoscon, however it was still shown alive and well in deCONZ interface. I also had two Aqara devices doing the same. I ended up deleting them and re-adding (after a lot of grief from the other half over things now working AGAIN!)

The really strange one is I have one Aqara switch which is shown as out of service in Phoscon, however reports on the API and if I click the group it will switch!

I really need to know what more you need so I can resolve this issue before I get in to more trouble!

manup, 2019-02-08:
As mentioned above we nee to check the REST-API output to compare sensor owned group ids and left over groups.

As alternative you may send me a backup file to mpi@dresden-elektronik.de so I can check the records.

I'll do some tests tomorrow with my Ikea motion sensor, to see if the groups are changed just by pull out/in the battery .

ademobile, 2019-02-08:
I have just created a backup and emailed you. Thank you so much

manup, 2019-02-08:
Hmm looked in the database file, it's weird.
Did you ever manually edit the zll.db file, e.g. like deleting entries?

For example "Tradfri motion sensor 8" in your screenshot references a sensor with id 8. There is one sensor with id 8 but that's a Xiaomi switch.

The Ikea motion sensors have ids 1, 4 and 10 and there are no entries of deleted motion sensors. Unless these were deleted manually from the database this is really strange.

ademobile, 2019-02-08:
No, I’ve never edited the database. All has been done via phoscon apart from the virtual sensor that I added via an api call. 

Would this explain why I am having so many strange issues?

ademobile, 2019-02-15:
Lost a sensors again this morning, and noticed that 2.05.59 was out, so upgraded. Thought I would remove (via web interface) all the IKEA motion sensors and re-add. I have noticed that there are still 3 entries shown on the older web interface that I cannot delete:

[image]
[image]
[image]

How to I clean up my system so it truly reflects what I have installed?

manup, 2019-02-15:
These should be the groups controlled by the sensor directly, they shouldn't be deleted. Note the old web interface isn't maintained anymore. In the Phoscon App you shouldn't see these groups.

ademobile, 2019-02-15:
Sorry, I should have made it clear, these still existed when NO IKEA motion sensors were added!

ademobile, 2019-02-15:
Also they seem to break through to the Phoscon interface as per my original posting

ademobile, 2019-02-15:
And within 12 hours I have lost a motion sensor, had to delete and re-pair. I am getting grief over this as it was no problem until after 2.05.55 update and my mother has no issue with hers (who is on that version). Either I need something cleaning up on my installation or some change is causing this. Please can you give me so things to try?

manup, 2019-02-15:
You can edit the zll.db file with sqlitebrowser and delete entries for the sensors and groups (close deCONZ first). 

However I don't think the sensors are the problem here. More that they somehow loose connection to their parent node or even that they are connected but the parent doesn't forward the signal.

When I recall correctly you have Xiaomi router devices in the mix. I suggest you watch to which parent node the sensor is connected in the deCONZ GUI (green line), so we can get a better picture on what happens here. Screenshot would also be helpful.

ademobile, 2019-02-16:
Sorry for the delay, had a few things to do with it being the weekend :-)

Here are the parent nodes for each sensor:

The porch sensor (Its parent is the raspbee):
[image]

The en-suite sensor (Its parent is a IKEA CT lamp):
[image]

The cloak room sensor (Its parent is also a IKEA CT lamp):
[image]

Yes you are right I am using Xiaomi router devices in the the mix, they are either 1 or 2 gang battery switches, and 1 or 2 gang main powered switches (with neutral as like most houses built after 2000 in the UK DO have neutral at the switch box). I also have IKEA 5 button switches, IKEA lamps and Osram lamps. I also have a Osram smart switch mini, which pairs but does not show up in Phoscon and I have not got round to raising a support request for!

I hope the above helps

ademobile, 2019-02-16:
I have also just noticed the porch sensor that I lost yesterday, after deleting and re-pairing has now appeared in the Phoscon app as the old ID as an empty group:

[image]

ademobile, 2019-02-21:
This morning I lost the en-suite motion sensor. It did work once as an event was picked up, however it just stopped. I fired up the GUI and captured this:

[image]

The dot beside the grey bar flashes blue (especially if I try to read the basic cluster) however I am not detecting any events via the API or Phoscon (I have the device next to my, waving my hand over it):

[image]

This sensor is causing me the most grieve when it 'stops' working as my other half goes mad!

ademobile, 2019-02-22:
This morning lost the en-suite remote switch - which I lost yesterday after the motion sensor but did not report as I was waiting for a reply (this caused a massive argument, and because of the log in issues as well, it was hard to resolve the situation).

It was as if the remote had lost its bindings as it would not turn the group of lights on that it should even without the RPi running (I powered it down to take it out of the equation), unlike the other two remotes, which turned the lights on or off that they are bound to.

There is a common theme with my issues - IKEA remotes and sensors. They are shown in the deCONZ GUI as the screen shots provided above, however it is like they are not being listened to.

I really need some help in resolving this as it cannot continue and I have invested too much time and money just to give up over something that should just 'work'.

manup, 2019-02-22:
>There is a common theme with my issues - IKEA remotes and sensors. They are shown in the deCONZ GUI as the screen shots provided above, however it is like they are not being listened to.

The line between sensor and light indicates it still operating. Do the sensor and remote nodes blink blue when they are triggered by motion or button press?

ademobile, 2019-02-22:
Yes, I waved my hand to 'wake' the sensor up to read the cluster info as I am aware that they sleep. For the remote I cannot remember as the 'argument' was clouding my fault finding to get things working, however I would say yes

manup, 2019-02-22:
Looks like Zigbee side is working.
I think this might be an older still unresolved bug, which doesn't fire events when time configuration on the system is different from the one in the deCONZ application. Looking forward to fix this in 2.05.60.

In the meantime can you check how the system timezone is configured?

    cat /etc/timezone

ademobile, 2019-02-22:
Europe/London

I have noticed in the past disparity between the time deCONZ / Phoscon shows and my local time, if that is any help (i.e. one hour different)

ademobile, 2019-02-23:
The en-suite remote stopped again this morning. We have been testing several times yesterday and this morning at around 6:50 (UK time!) it worked, then at around 7:15 nothing. Later I got up the GUI with the idea of recording the blue dot flashing etc. It was still not working at around 8:50, so I hit the record button and pressed the remote and the lights turned on. As I cannot upload the video all I can do is describe that the dot flashed blue then turned green!

ademobile, 2019-02-24:
This morning (6:55ish UK time) the motion sensor in the en-suite was not working again, got up, had breakfast and just went to retrieve it to see if it is still alive in the GUI (like the remote above) and it worked without me doing anything. So there is definitely a problem in deCONZ with events not being forwarded correctly.

ademobile, 2019-02-24:
Also any thoughts on my comments about logging in to Phoscon that I have mentioned here and issue #1238? as it is very annoying just to log in to turn lights on!

ademobile, 2019-02-27:
After a couple of days of no noticeable issues, this morning the remote stopped responding again, so I turned the lights on via Phoscon (going through the rigmarole of logging in just to do this 👎 ) and then the remote started responding. I have noticed that there has been little or no response over the past few days, has something happened?

ademobile, 2019-02-28:
Again this morning the IKEA remote is un-responsive (yet still shown as 'alive' in the GUI). As I have got the Xiaomi remote switch working yesterday, this is in the same room and it worked fine. This is now pointing to my thoughts that there is an issue with IKEA battery items 'sleeping' whilst connected to deCONZ.

closed 2019-07-05 by stale[bot]