dresden-elektronik/deconz-rest-plugin #1222: Heiman SmartPlug new version

d-litvinov, 2019-02-05:
I've get new version of the Heiman SmartPlug and it dosn't show the power consumption information, but turning on/off works. I have an older version of the same plug and it works like a charm.

I use the latest version of deCONZ (V2_05_58).
Looks like it should be added in the whitelist, but I don't know how.

[image]
[image]
[image]
[image]
[image]
[deconz.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2832268/deconz.log)

d-litvinov, 2019-02-05:
logs here https://github.com/dresden-elektronik/deconz-rest-plugin/files/2832268/deconz.log

d-litvinov, 2019-02-12:
I've just checked the master branch, and the plug now reports the power consumption. 
Thank you so much for such a quick fix!

closed 2019-02-12 by d-litvinov

arnonel, 2019-10-10:
Hi - wasnt sure where to ask this

I have the Heiman HS2SK (UK Version)

How do you put it into pairing mode so that app can find it?