dresden-elektronik/deconz-rest-plugin #1224: Unable to add Xiaomi Door Sensor

Oliviakrkk, 2019-02-05:
Hi

I am unable to add Xiaomi Door Sensor to deconz. 

1. I enter pairing mode
2. I reset the sensor (led flashes 3 times). 
3.  Nothing happens...

Log from deconz container:
19:11:11:003 Set sensor check interval to 1000 milliseconds
19:11:14:289 don't close database yet, keep open for 900 seconds
19:11:25:462 Current channel 11
19:11:25:510 Device TTL 2921 s
19:11:32:447 GW firmware version: 0x262f0500
19:11:32:447 GW firmware version is up to date: 0x262f0500
19:11:44:404 send permit join, duration: 59
19:11:49:201 0x001788010122C44A error APSDE-DATA.confirm: 0xE9 on task
19:11:54:179 0x001788010122C44A error APSDE-DATA.confirm: 0xA7 on task
19:11:59:030 0x001788010122C44A error APSDE-DATA.confirm: 0xA7 on task
19:12:02:125 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
19:12:02:125 ZCL attribute report 0x00158D0001E52DDB for cluster 0x0400, ep 0x01
19:12:02:188 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
19:12:02:188 ZCL attribute report 0x00158D0001E52DDB for cluster 0x0406, ep 0x01
19:12:02:903 Set sensor check interval to 100 milliseconds
19:12:09:511 don't close database yet, keep open for 900 seconds
19:12:22:113 0x001788010122C44A error APSDE-DATA.confirm: 0xE9 on task
19:12:25:473 Current channel 11
19:12:25:505 Device TTL 2861 s
19:12:26:659 0x001788010122C44A error APSDE-DATA.confirm: 0xA7 on task
19:12:31:909 0x001788010122C44A error APSDE-DATA.confirm: 0xA7 on task
19:12:43:464 Search sensors done
19:12:44:403 send permit join, duration: 59
19:12:50:222 0x001788010122C44A error APSDE-DATA.confirm: 0xE9 on task
19:12:50:894 0x001788010122C44A error APSDE-DATA.confirm: 0xE9 on task
19:12:52:191 0x001788010122C44A error APSDE-DATA.confirm: 0xE9 on task
19:12:57:057 0x001788010122C44A error APSDE-DATA.confirm: 0xA7 on task
19:13:01:955 0x001788010122C44A error APSDE-DATA.confirm: 0xA7 on task
19:13:11:213 don't close database yet, keep open for 900 seconds

Any ideas what I am dowing wrong?

manup, 2019-02-05:
From the logs it doesn't look like the sensor tries to join. Have you tried pressing the setup button a bit longer like 10 seconds?

Further are you using a ConBee; if yes do you have a USB extension cable?

Oliviakrkk, 2019-02-05:
Ok. so I tried once more and:

1. Pairing process in Phoscon failed. Message "Failed to connect"
2. Sensors is visible in deconz GUI, but not completely detected:
[image]

3. Logs:
19:42:58:718 0x00158D000171E9E0 skip Xiaomi attribute 0x0000
19:42:58:780 FP indication 0x0104 / 0x0000 (0x00158D000171E9E0 / 0xB3D2)
19:42:58:781                       ...     (0x00158D000171E9E0 / 0xB3D2)
19:42:58:782 ZCL attribute report 0x00158D000171E9E0 for cluster 0x0000, ep 0x01
19:42:58:785 [1] get node descriptor for 0x00158d000171e9e0
19:42:58:812 FP indication 0x0104 / 0x0000 (0x00158D000171E9E0 / 0xB3D2)
19:42:58:812                       ...     (0x00158D000171E9E0 / 0xB3D2)
19:42:58:813 Remember Xiaomi special for 0x00158D000171E9E0
19:42:58:813 ZCL attribute report 0x00158D000171E9E0 for cluster 0x0000, ep 0x01
19:42:58:813 0x00158D000171E9E0 extract Xiaomi special attribute 0xFF02
19:42:58:813    64 on/off 1
19:42:58:813    01 battery 2985 (0x0BA9)
19:42:58:813    03 unsupported index (data type 0x21)
19:42:58:813    04 unsupported index (data type 0x24)
19:42:58:813    05 unsupported index (data type 0x21)
19:42:58:813    06 unsupported index (data type 0x20)
19:42:58:818 wait response fastEnddeviceProbe() 0x00158D000171E9E0, elapsed 33 ms
19:43:01:434 don't close database yet, keep open for 900 seconds
19:43:06:589 APS-DATA.confirm id: 54 status: transaction expired (8 seconds)
19:43:08:426 saved node state in 4 ms
19:43:08:577 sync() in 150 ms
19:43:18:293 0x001788010122C44A error APSDE-DATA.confirm: 0xE9 on task
19:43:23:351 0x001788010122C44A error APSDE-DATA.confirm: 0xA7 on task
19:43:25:464 Current channel 11
19:43:25:497 Device TTL 1001 s
19:43:32:446 GW firmware version: 0x262f0500
19:43:32:446 GW firmware version is up to date: 0x262f0500
19:43:35:693 0x001788010122C44A error APSDE-DATA.confirm: 0xE9 on task
19:43:40:383 0x001788010122C44A error APSDE-DATA.confirm: 0xA7 on task
19:43:43:932 Search sensors done
19:43:45:105 0x001788010122C44A error APSDE-DATA.confirm: 0xA7 on task
19:43:45:404 send permit join, duration: 59
19:43:48:601 no button map for: lumi.weather ep: 0x01 cl: 0x0402 cmd: 0x0A pl[0]: 000
19:43:48:601 ZCL attribute report 0x00158D0002725634 for cluster 0x0402, ep 0x01
19:43:48:664 no button map for: lumi.weather ep: 0x01 cl: 0x0405 cmd: 0x0A pl[0]: 000
19:43:48:664 ZCL attribute report 0x00158D0002725634 for cluster 0x0405, ep 0x01
19:43:48:693 no button map for: lumi.weather ep: 0x01 cl: 0x0403 cmd: 0x0A pl[0]: 000
19:43:48:694 ZCL attribute report 0x00158D0002725634 for cluster 0x0403, ep 0x01
19:43:50:676 0x001788010122C44A error APSDE-DATA.confirm: 0xA7 on task
19:43:59:438 don't close database yet, keep open for 900 seconds
19:44:07:405 0x001788010122C44A error APSDE-DATA.confirm: 0xE9 on task
19:44:12:286 0x001788010122C44A error APSDE-DATA.confirm: 0xA7 on task
19:44:17:457 0x001788010122C44A error APSDE-DATA.confirm: 0xA7 on task
19:44:22:467 0x001788010122C44A error APSDE-DATA.confirm: 0xA7 on task
19:44:25:476 Current channel 11
19:44:25:508 Device TTL 941 s
19:44:27:178 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0400 cmd: 0x0A pl[0]: 000
19:44:27:178 ZCL attribute report 0x00158D0001E52DDB for cluster 0x0400, ep 0x01
19:44:27:209 no button map for: lumi.sensor_motion.aq2 ep: 0x01 cl: 0x0406 cmd: 0x0A pl[0]: 000
19:44:27:209 ZCL attribute report 0x00158D0001E52DDB for cluster 0x0406, ep 0x01
19:44:27:902 Set sensor check interval to 100 milliseconds
19:44:40:091 0x001788010122C44A error APSDE-DATA.confirm: 0xE9 on task
19:44:44:403 send permit join, duration: 59
19:44:44:621 Search sensors done
19:44:45:213 0x001788010122C44A error APSDE-DATA.confirm: 0xA7 on task
19:44:45:621 Search sensors done
19:44:45:855 Content not completely loaded (got 0 of 2), wait 20ms
19:44:50:432 0x001788010122C44A error APSDE-DATA.confirm: 0xA7 on task

19:45:10:906 0x001788010122C44A error APSDE-DATA.confirm: 0xA7 on task
19:45:16:449 no button map for: lumi.weather ep: 0x01 cl: 0x0405 cmd: 0x0A pl[0]: 000
19:45:16:451 ZCL attribute report 0x00158D0002725634 for cluster 0x0405, ep 0x01
19:45:16:512 no button map for: lumi.weather ep: 0x01 cl: 0x0403 cmd: 0x0A pl[0]: 000
19:45:16:512 ZCL attribute report 0x00158D0002725634 for cluster 0x0403, ep 0x01
19:45:25:472 Current channel 11
19:45:25:504 Device TTL 881 s
19:45:31:283 0x001788010122C44A error APSDE-DATA.confirm: 0xE9 on task
19:45:36:470 0x001788010122C44A error APSDE-DATA.confirm: 0xA7 on task
19:45:41:480 0x001788010122C44A error APSDE-DATA.confirm: 0xA7 on task
19:45:43:403 send permit join, duration: 59
19:45:45:918 Search sensors done
19:45:46:919 Search sensors done
19:45:54:346 don't close database yet

Oliviakrkk, 2019-02-05:
And I forgot to answer:

I have conbee attached directly do USB port on miniITX board. So there shouldn't be any issues with power.

manup, 2019-02-05:
>I have conbee attached directly to USB port on miniITX board. So there shouldn't be any issues with power.

Power is not the issue, but when connected directly in some cases the radio signal is very bad.

The logs show that the requests don't come through reliable, hence the 0xE9, 0xA7 and transaction expired status codes.

Since you have only end-devices (screenshot) connected you may suffer also from another issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1190, which will be fixed in deCONZ version 2.05.59.

Oliviakrkk, 2019-02-05:
Ok. So, I removed sensor from the doors and tried pairing ~5cm from the Conbee. Did not work.

I also paired like 10 sensors today without any issues (Aqara Temp, Xiaomi and Aqara Motion).  I only cannot pair Xiaomi Window and Aqara Window (tried both). Only those two are not working. I think if it was range issue I wouldn't be able to pair anything... Also deconz uses only 7% of CPU and it is steady. 

As to Conbee sitck in the USB...I will get extension cord and recheck. Thanks!

I joined one Hue Iris (so now I have one device not battery operated). Will see how it works after some time.  I have many more, but decided to ditch Xiaomi hub first, then take care of Hue. 
[image]

manup, 2019-02-05:
Ah ok didn't see the Hue light, this light keeps the other issue from triggering :)

Do you have more lights to bring in the network? Currently you have 13 end-devices connected directly to the gateway. I need to check the code it shouldn't hit the limit, but not 100% sure yet.

Oliviakrkk, 2019-02-06:
I added 3 more Hue Lights. Those are now used as relays... 
Now I am struggling to add anything else ... 

20:15:40:682 0x00178801021D61FE error APSDE-DATA.confirm: 0xE9 on task
**20:15:49:117 new node - ext: 0x00158d000192d220, nwk: 0x3BEE
20:15:49:190 reuse dead link (dead link container size now 1)**
20:15:51:471 Current channel 11
20:15:51:503 Device TTL 1620 s
20:15:53:568 CTRL skip polling while permit join is set
20:15:53:776 0x00178801021D61FE error APSDE-DATA.confirm: 0xA7 on task
20:15:54:384 don't close database yet, keep open for 900 seconds
20:15:56:475 GW firmware version: 0x262f0500
20:15:56:476 GW firmware version is up to date: 0x262f0500
20:15:59:433 saved node state in 5 ms
20:16:02:600 sync() in 3166 ms
20:16:02:608 CTRL skip polling while permit join is set
20:16:05:093 0x00178801021D61FE error APSDE-DATA.confirm: 0xE9 on task
20:16:05:687 CTRL skip polling while permit join is set
20:16:05:702 CTRL skip polling while permit join is set
20:16:07:094 0x00178801029AA1D7 error APSDE-DATA.confirm: 0xE9 on task
20:16:11:753 CTRL skip polling while permit join is set
20:16:17:804 CTRL skip polling while permit join is set
20:16:18:139 0x00178801021D61FE error APSDE-DATA.confirm: 0xA7 on task
20:16:23:886 CTRL skip polling while permit join is set
20:16:29:403 send permit join, duration: 59
20:16:29:937 CTRL skip polling while permit join is set
20:16:36:003 CTRL skip polling while permit join is set
20:16:36:387 0x00178801029AA1D7 error APSDE-DATA.confirm: 0xE9 on task
20:16:42:072 CTRL skip polling while permit join is set

I found in logs:
"20:15:28:021 verified group capacity: 24 and group count: 1 of LightNode 0x00178801021d61fe"

This is device limit you mentioned?

Oliviakrkk, 2019-02-06:
UPDATE: I read in some other issue about powering off conbee. I power off nas and disconnected power supply for 5 mins - all LEDs on the board went away...And then I powered everything back up.

After that I was able to add:
-2 xiaomi buttons
-2 aqara motion sensors
-aqara double wall switch
-aqara water sensor

But still I am fighting with xiaomi and aqara door sensors :/.

Oliviakrkk, 2019-02-07:
I think Manup was right. I had to have more reolays in my network. After adding Hue lights and power cycle I was able to pair both Xiaomi and Aqara Window sensors.

Thank!.

closed 2019-02-07 by Oliviakrkk