dresden-elektronik/deconz-rest-plugin #1225: After restart of Rasp, no update, no sensor

HeyJonny, 2019-02-05:
Hey!

I have a Raspberry 3 with the Raspbee module. After I reboot the Rasp, the software update is somehow "lost". Before the restart, however, the update was installed successfully, no problems!
I have tried updating several times, via deCONZ, via Phoscon, via Console ... nothing works ... I am really confused.
Additionally I loose all lamps / switches. I think, these two issus are somehow connected.
Could somebody help me? I have absolutly no idea what the problem is.

Greez, Jonny

manup, 2019-02-05:
Can you please provide some details:
* deCONZ version
* Firmware version
* Operating system / environment
* Headless, Docker or deCONZ GUI version

HeyJonny, 2019-02-17:
Sorry for my late reply...

Sure:
-  deCONZ 2.05.59 / 14.02.2019
- Firmware: "not connected" - update to 0x262f0500
- Raspbian 9.6 stretch | 4.14.79-v7+ GNU/Linux
- Headless

Maybe a nother point: After the update, the lamps are back again, without need to restore the backup

manup, 2019-02-17:
Hm interesting, so it works now?

HeyJonny, 2019-02-17:
No not really. As I said, after a reboot of the device, all lamps/witches and the firmeware is "gone" ... I have to update the firmeware of the device every restart.
When I manually install the update (only works via console), the lamps are back and everything works ... Until I restart the next time ...

manup, 2019-02-17:
>When I manually install the update (only works via console), the lamps are back and everything works ... Until I restart the next time ...

That's strange, maybe there are multiply instances of deCONZ running.
Can you post the output of:

    ps ax | grep deCONZ

Further please check that the serial interface is configured properly as described here:
https://github.com/dresden-elektronik/deconz-rest-plugin#software-requirements

HeyJonny, 2019-02-17:
The output:
  291 ?        Ss     0:07 /bin/bash /usr/bin/deCONZ-update2.sh
  373 ?        Ssl    2:13 /usr/bin/deCONZ -platform minimal --http-port=80
  374 ?        Ss     0:03 /bin/bash /usr/bin/deCONZ-WIFI2.sh
14128 pts/0    S+     0:00 grep --color=auto deCONZ

I am quite sure, that I configured correctly when I set up the system. However, I double-checked and the serial interface is configured as discribed.
After the reboot: Same problem :/

closed 2019-06-24 by stale[bot]