dresden-elektronik/deconz-rest-plugin #1226: All light lost connection, sponanously!

posixx, 2019-02-05:
Hello,

I was running 2.05.58 for some days now, and an hour ago no light can be controlled anymore! All connections are gone somehow. Taps were working as i saw the socket events coming in on my homeseer. I tried downgrading to .54 but now i only have 1 connection left: one motion sensor is working, but all my lights are disconnected. What to do? 

Frans

posixx, 2019-02-05:
[image]

This is the state currently, it found one motion sensor again. All hue lights.

manup, 2019-02-05:
Have you tried power on/off the gateway? (not just reboot)

posixx, 2019-02-05:
no will try soon, building latest from source now to see it that helps.. keep you posted

manup, 2019-02-05:
I think the firmware is stuck not higher level deCONZ , easiest way to check this is to power on/off.

manup, 2019-02-05:
Odd, can please create a fresh backup and send it to mpi@dresden-elektronik.de all network changes are stored in there.

I can check if this is a known bug (can be fixed manually).

posixx, 2019-02-05:
really appreciated! just sent 2 backups, one form junu 2018 and one from a half hour ago. let me know if you got them as i don't know if .dat files are accepted at your end

l-mb, 2019-02-05:
I think I'm seeing the same. All lights are unavailable, there's nothing in the UI and the web UX reports "not connected". I'd appreciate if you'd share how this can be fixed manually?

posixx, 2019-02-05:
@i-mb: i'm seeing all lights in the web-ui and dezonz-gui, just no online one, all unreachable. don't know if this is the same as your problem..

manup, 2019-02-05:
@posixx I've mailed you a workaround to fix the configuration, hope it works.

manup, 2019-02-05:
>I think I'm seeing the same. All lights are unavailable, there's nothing in the UI and the web UX reports "not connected". I'd appreciate if you'd share how this can be fixed manually?

Not connected would be another problem. Can you please provide more details on your setup (versions/firmware)?

posixx, 2019-02-05:
@manup you are a lifesaver. Your workaround works, currenly i see the network rebuilding connections again. Thanks for your quick help, i can shut down lights again when i go to bed:-)

smallgreyrock, 2019-02-07:
Can you share the workaround? I'm seeing a similar or identical issue with my lights that just started last night.

Sensors/switches are still visible in the deconz web interface, but no lights are displayed or controllable, and they can't be re-added. The devices are still present in zll.db. Unfortunately I have no db backup to revert to.

This is with 2.05.57 (and .58 I guess) and firmware 0x26300500.

l-mb, 2019-02-13:
Okay, same issue then - not connected might have been the wrong word. I can see them, but they flag as unavailable and I cannot perform any actions on them.

manup, 2019-02-15:
2.05.59 is available now and contains a Zigbee configuration page to inspect and repair broken configuration.

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

fillilutten, 2019-02-16:
I tried going back 2 versions when I know everything worked. But unfortunately it didn't help. Still can't reach some of my lights. The hue works great but the IKEA ones are still unreachable.

What I did was - revert back 2 versions to when it worked > rebooted my pi > turned on and off my IKEA bulbs > Still greyed out 

Hm. Anything I'm missing?

manup, 2019-02-16:
>rebooted my pi

No need to reboot.

>Hm. Anything I'm missing?

When the Hues work but IKEA don't it is a indication that network settings are okay now but the IKEA lights might be on the wrong channel.

If you have a IKEA bulb near to the gateway you me check where it is:

* Phoscon App > Help > Open Wireless Light Control (2016)
* Menu > Settings > Scan for Devices

If the light can be found you can see on which channel/panid it is and if it is different from your current channel.

[image]

Note Hue lights detect a channel change on their own after a power cycle, IKEA lights sadly don't.

fillilutten, 2019-02-16:
Thanks! You where right, it is on the wrong channel and also network id. Tried pushing the reset and then paired it again, it shows up as connected now but when changing brightness nothing happens.

manup, 2019-02-16:
>it shows up as connected now but when changing brightness nothing happens.

It should work soon after the light joins. May take a few seconds in worst case minutes.

fillilutten, 2019-02-16:
Just after a minut it's now back to greyed out. To me it seemed like the reset button didn't do anything, when I did reset and then scanned again for devices the bulb hade the same wrong network id and channel.

manup, 2019-02-16:
Yeah IKEA reset can be a bit tricky.

I usually use the reset sequence, since it's most reliable:
* Start light search in Phoscon App
* Reset light as shown in the video https://www.youtube.com/watch?v=mJm9YpPrGzk
  (import toggle on/off quickly)
* When the light flickers, toggle on/off once more

fillilutten, 2019-02-16:
Thanks so much @manup! It's now all up and running again. 
Wow you saved me so much time and effort, have a great weekend!

manup, 2019-02-16:
:) cool glad that it works now.

fillilutten, 2019-02-16:
Some deconz god doesn't like me :(

Was away a couple of hours. When I come back the whole apt lights up when I arrive, just as I've programmed. But after adding my newly bought Innr 120 I realize that all my IKEA bulbs are now gone from the UI. They work if I send a scene request but they are totally gone.

Can see them in HA but they don't do anything and my newly added power plug doesn't show up within HA.

smallgreyrock, 2019-02-17:
> I think the firmware is stuck not higher level deCONZ , easiest way to check this is to power on/off.

Unplugging and reinserting the ConBee fixed things for me.

manup, 2019-02-17:
>But after adding my newly bought Innr 120 I realize that all my IKEA bulbs are now gone from the UI.

Do you mean not visible in the deCONZ Node View or Phoscon App shown as not reachable?

By the way, which firmware version are you using?

fillilutten, 2019-02-17:
> Do you mean not visible in the deCONZ Node View or Phoscon App shown as not reachable?
> 
> By the way, which firmware version are you using?

I'm running HassIO and the deCONZ addon and they are not visible in Phoscon Web UI. Not sure but I don't think there are any deCONZ Node View when using the deCONZ addon.
Running fw version 262F0500 conbee.

manup, 2019-02-17:
>and they are not visible in Phoscon Web UI

I see, but do you mean not listed at all or they are listed but shown as reachable?

If they aren't listed at all there is something wrong with you configuration, since this should always work regardless of any Zigbee traffic. The lights, sensors and switches are just loaded from the database and should be shown even when no ConBee/RaspBee is attached.

 Is this running on a Raspberry Pi?

fillilutten, 2019-02-17:
> I see, but do you mean not listed at all or they are listed but shown as reachable?
>
> If they aren't listed at all there is something wrong with you configuration, since this should always work [...]

Not listed at all, they are just gone, not in the old 'Open Wireless Light Control (2016)' or the "default mode". Haven't had the time to try and redo the reset > repair I did yesterday when all started working again.

Also tried to reinstall deconz addon and then backup from .dat save file but didn't make any difference.

Yes running on a Raspberry Pi 2.

manup, 2019-02-17:
>Not listed at all, they are just gone, not in the old 'Open Wireless Light Control (2016)' or the "default mode". Haven't had the time to try and redo the reset > repair I did yesterday when all started working again.

>Also tried to reinstall deconz addon and then backup from .dat save file but didn't make any difference.

Something is not working here with the database, if devices are gone without doing anything. Note the backup can just work if it was created when everything still worked and devices were listed.

To me the question is, where is the data gone? Hard to tell. Possible reasons in the past:
* sd-card failure
* Multiple instances of deCONZ running (unlikely in default Hass.io install)

You may send me the backup file to mpi@dresden-elektronik.de I can check if at least device data was in there or as it looks like it is empty.

rich710se, 2019-02-20:
Today suddenly all my zigbee devicesw stopped working, nothing happens, I have rebooted my rpi several times, pulled out my conbee stick. Powered my bulbs on and off, but nothing happens except that some xiaomi endpoints get connected. How can I find out what to do, my wife is very angry now ;) I have deconz 2.05.59 since the day it was released, have worked flawless since then.
[image]

manup, 2019-02-20:
@rich710se  please check if the network configuration parameters are stable in the new Zigbee configuration page.

To open the page, follow the instructions from the release note: https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

rich710se, 2019-02-21:
The page looks like this, nothing have happened over night still no connections. Are there more too look at, that is interesting for you?
[image]
[image]

rich710se, 2019-02-21:
Nothing happened when i took "Load" on the first entry, but I think that is because I tried to reinstall 2.05.59 yesterday after I lost it. So maybe I reverted back to that position. But when i took Load on my second entry in this list, every bulp started talking again!! :) Thanks alot!!.

posixx, 2019-02-21:
Exactly, your PANID was changes, reverting back to the orginal one did the trick.

manup, 2019-02-21:
>Exactly, your PANID was changes, reverting back to the orginal one did the trick.

This. 

Next firmware update should fix this issue to happen in first place.

closed 2019-06-28 by stale[bot]