dresden-elektronik/deconz-rest-plugin #1227: Philips Hue "Being" Ceiling Lamp supported?

VGS1703, 2019-02-06:
Hallo,

I'm using FW 2.05.58. I am trying to connect a Philips Hue lamp ("Being", Model LTC001) to deconz, but it failed. It took me litterally ages to get it back to factory setting and not to connect back to Hue bridge automatically. As I managed that, deconz was still not able to find it.
I just want to know: Is it me (than I will keep on trying...) or is this particular lamp just not supported yet?

Thanks in advance.

alex0411, 2019-02-06:
Hello, 

my Hue Ceiling Lamp LTC011 "Cher" is working proberly with deConz. Is there anything in the log when you are trying to connect your lamp to deconz?

manup, 2019-02-06:
>and not to connect back to Hue bridge automatically.

If it is still controllable in Philips hue bridge, then it is not reset to factory settings.

When  the light is still connected to the hue bridge (and controllable) try the following:

* Start light search in Phoscon App
* Delete the light in the Philips Hue App
* The lights will be reset by the Hue bridge and should join the Phoscon network

VGS1703, 2019-02-07:
I feel a little ashamed that I couldn't figure that out by myself:

> When the light is still connected to the hue bridge (and controllable) try the following:

* Start light search in Phoscon App
* Delete the light in the Philips Hue App
* The lights will be reset by the Hue bridge and should join the Phoscon network

But it worked! I'm happy and glad that I asked, it saved me a lot of time trying to reset the light with the dimmer switch or entering the SN in the Hue App over and over again...

THANK YOU!

closed 2019-02-07 by VGS1703