dresden-elektronik/deconz-rest-plugin #1229: add support for Blaupunkt SCM-S1 roller shutter

steveknoefler, 2019-02-07:
Hello,
please add support for the Blaupunkt roller shutter scm-s1.

thanks
Steve

[image]
[image]
[image]
[image]
[image]
[image]
[image]

ebaauw, 2019-02-09:
@steveknoefler It is already exposed by the REST API as `/lights/26`.I don’t think there’s anymore we can do, unless there something useful in cluster 0x0100.

ebaauw, 2019-02-09:
0x0100 is _Shade Configuration_.  No cluster-specific commands, but a number of cluster-specific attributes:

Id | Name | Type | Range | Access | Default | M/O
-- | -- | -- | -- | -- | -- | --
0x0000 | PhysicalClosedLimit | uint16 | 0x0001 – 0xfffe | Read Only | - | O
0x0001 | MotorStepSize | uint8 | 0x00 – 0xfe | Read Only | - | O
0x0002 | Status | map8 | 0b0000 xxxx | Read/Write | 0b0000 0000 | M
0x0010 | ClosedLimit | uint16 | 0x0001 – 0xfffe | Read/Write | 0x0001 | M
0x0011 | Mode | enum8 | 0x00 – 0xfe | Read/Write | 0x00 | M

I don't think exposing them through the REST API would be of much value, but it would make sense to include them in general.xml, so they can be queries and set through the deCONZ GUI.

steveknoefler, 2019-02-10:
oh sorry my mistake its now working :)

closed 2019-02-10 by steveknoefler

sammy98, 2020-01-07:
I have a hard time to get it working. 
It is paired as light but i cannot trigger up and down and dont see the current position of the device.

[image]

Any hints for me?