dresden-elektronik/deconz-rest-plugin #1231: upgraded from 2.05.52 to 2.05.58 - now it's a mess

ErnstEeldert, 2019-02-08:
I recently upgraded from 2.0.52 to 2.05.58 as I was looking for some improved handling of the Eurotronics thermostat.

However, since then I am experiencing the following issues:

1. 30% of  my Xiaomi Motion sensors are dead/missing/inactive
2. The ones who are not dead/missing/inactive are not responsive in 25% of the cases.
3. My only Hue Motion Sensor blinks red consistently (normally only intermittently)
4. My IKEA Tradfri lights blink every now and then (are they trying to tell me something?)

What I would like to know is how to get my dead sensors back online, without resetting their /sensors/nn address as I don't want to have to fix my rules.

Am I running on a 'somewhat' stable deconz/fw version, or should I pick another recommended combo?

This is what PWA tells me:

Version 2.05.58 / 1/30/2019
Firmware 262F0500

Thanks.

kugelkopf123, 2019-02-08:
Interesting! My Tradfri e14 flashes now and then too. But only if its on. Had already a connection with deconz suspected. I'm still on .57.
If they work, my Aqara PIR sensors triggering a bit slower as they did befor.
Have currently that feeling it would run worse with each new version.

manup, 2019-02-08:
@ErnstEeldert can you please create a Backup file and send it to mpi@dresden-elektronik.de I'd like to check if this is a known error.

manup, 2019-02-08:
>Interesting! My Tradfri e14 flashes now and then too. But only if its on. Had already a connection with deconz suspected. I'm still on .57.

I've seen that too for my GU10 Ikea spots, but I don't think we can do much about the lights flickering. Currently this looks like an light firmware issue.

>If they work, my Aqara PIR sensors triggering a bit slower as they did befor.

The performance issue is fixed now; will be part of 2.05.59

jurriaan, 2019-02-08:
Since the last few releases it seems that the blinking increased, have seen it before as well, but looks like it happens more often now.

I also have had issues where some lights don't respond until I turn them off and on again. Even reset my whole network (setting up a new deconz setup from scratch).

I definitely see more issues with IKEA devices the latest releases.

manup, 2019-02-08:
The thing that changed in recent releases is that lights get polled quite fast until reports are received. Maybe Ikea lights don't like polling too much and some internal error is triggered; perhaps the light reboots.

Considerations:
* In theory we can switch to an attribute reporting only model for Ikea lights
* Reporting works great once it is configured and polling isn't needed 
* Bad news is that reporting configuration must be redone on each power-cycle of a Ikea light

If polling is the problem, we can to improve and speedup reporting configuration after a light gets powered (device announce is received).

I wonder how the Ikea gateway deals with this. Can't check it since min currently doesn't communicate with the Ikea app at all :/

Also how is the situation with the Philips Hue bridge? It completely relies on fast polling. I guess only the Mgmt_Lqi_req are missing, which deCONZ does to build and maintain the mesh gui.

manup, 2019-02-11:
@ErnstEeldert thanks I did check the database in the backup. The configuration looks good, no parameters were changed. Looks like the radio module might be in a invalid state.

Have you tried to cut power off the gateway completely (not just restart) in order to reboot the radio?

joubbi, 2019-02-15:
My Ikea lights are blinking as well.
The blinking happens quite often and it seems to be random.
Yesterday one of them was blinking at least once per minute.
This is quite annoying. Is there anything I can help with getting rid of this problem?
I didn't have this problem with 2.05.52 and the older firmware I was running two weeks ago.

manup, 2019-02-15:
2.05.59 is available now and contains a few changes for IKEA lights.
Note after update please power-cycle the lights once.

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

ErnstEeldert, 2019-02-15:
Thanks! Will try and report the results here.

joubbi, 2019-02-15:
I have been running 2.05.59 some hours now. No flickering.
Thank you!

closed 2019-06-22 by stale[bot]