dresden-elektronik/deconz-rest-plugin #1232: Xiaomi PIR dead in deconz

Gnaget2, 2019-02-08:
My Xiaomi PIR seems to be inactive in decons gui. no line at all and dont responding when i try the signal monitor to it.

it is working and is active in phoscon web gui and to Home Assistant but from time to time (a few time a day its not triggering on motion so i guess its related) and thats wy i reporting this

Conbee
Version 2.05.58 / 1/30/2019
Firmware 262F0500

I have tried to remove it and re add and then it works for some day but then the same.

Below is a filtered log from all line from the device today

```
08:25:19:812 ZCL attribute report 0x00158D0001E7E144 for cluster 0x0406, ep 0x01
08:26:21:350 ZCL attribute report 0x00158D0001E7E144 for cluster 0x0406, ep 0x01
08:29:28:291 max transmit errors for node 0x00158D0001E7E144, last seen by neighbors 66227 s
08:29:38:422 max transmit errors for node 0x00158D0001E7E144, last seen by neighbors 66237 s
08:29:48:680 max transmit errors for node 0x00158D0001E7E144, last seen by neighbors 66247 s
08:30:04:621 max transmit errors for node 0x00158D0001E7E144, last seen by neighbors 66263 s
08:44:28:611 max transmit errors for node 0x00158D0001E7E144, last seen by neighbors 67127 s
08:44:37:798 ZCL attribute report 0x00158D0001E7E144 for cluster 0x0406, ep 0x01
08:53:07:600 ZCL attribute report 0x00158D0001E7E144 for cluster 0x0000, ep 0x01
08:53:07:600 0x00158D0001E7E144 extract Xiaomi special attribute 0xFF02
08:53:17:890 0x00158D0001E7E144 error APSDE-DATA.confirm: 0xD0 on task
09:06:07:407 ZCL attribute report 0x00158D0001E7E144 for cluster 0x0406, ep 0x01
09:10:27:925 ZCL attribute report 0x00158D0001E7E144 for cluster 0x0406, ep 0x01
09:20:56:449 ZCL attribute report 0x00158D0001E7E144 for cluster 0x0406, ep 0x01
09:47:27:647 ZCL attribute report 0x00158D0001E7E144 for cluster 0x0000, ep 0x01
09:47:27:647 0x00158D0001E7E144 extract Xiaomi special attribute 0xFF02
09:47:37:824 0x00158D0001E7E144 error APSDE-DATA.confirm: 0xD0 on task
09:48:52:583 ZCL attribute report 0x00158D0001E7E144 for cluster 0x0406, ep 0x01
10:42:24:234 ZCL attribute report 0x00158D0001E7E144 for cluster 0x0000, ep 0x01
10:42:24:234 0x00158D0001E7E144 extract Xiaomi special attribute 0xFF02
10:42:34:559 0x00158D0001E7E144 error APSDE-DATA.confirm: 0xD0 on task
10:54:48:756 ZCL attribute report 0x00158D0001E7E144 for cluster 0x0406, ep 0x01
10:57:22:399 ZCL attribute report 0x00158D0001E7E144 for cluster 0x0406, ep 0x01
10:58:22:815 ZCL attribute report 0x00158D0001E7E144 for cluster 0x0406, ep 0x01
10:59:24:720 ZCL attribute report 0x00158D0001E7E144 for cluster 0x0406, ep 0x01
11:00:25:248 ZCL attribute report 0x00158D0001E7E144 for cluster 0x0406, ep 0x01
11:01:26:482 ZCL attribute report 0x00158D0001E7E144 for cluster 0x0406, ep 0x01
11:04:25:728 ZCL attribute report 0x00158D0001E7E144 for cluster 0x0406, ep 0x01
11:07:11:959 ZCL attribute report 0x00158D0001E7E144 for cluster 0x0406, ep 0x01
```
please let me know what/if you need more info to be able to trubleshoot

manup, 2019-02-08:
This seems to be the Xiaomi Mi version of the sensor (without light sensor).

Is the version shown in the sensor details on the Phoscon App > Devices > Sensors?

According to the logs deCONZ tries to read something from the sensor, likely the datecode, but no route can be found to the device. Which means not parent node.

There is a chance that the gateway itself is the parent but doesn't now about it if the sensor doesn't do any mac data polling.

I need to check the code perhaps this case can be handled.

Gnaget2, 2019-02-08:
Hi!

yes this is the version withot lux/temp

when i added it it was conected to a hue light 4-5 m away with a green line (several other hue ligtstrips are within 2-3m always on in the same room)

it could be connected direct to the GW, as its just 4-5m away but on the other side of a wall

[image]

manup, 2019-02-08:
One thing to note here is that the parent node should be one which is always powered, since Xiaomi end-devices are very stubborn and often don't select a new parent node when the old on is powered off.

Not sure if it is possible with the Mi version but Aqara devices can be forced to make a signal check when pressing the small button quickly 5 times.

Gnaget2, 2019-02-08:
i can try turning of all of the few nodes i have that is beeing turned of from time to time forsing it to connect to one ofe the nodes that are close.

M203, 2019-02-08:
I have a xiaomi motion sensor with lux, I can see it reports lux everytime I step in front of it but motion is only reported sporadically. This sensor has worked flawlessly before but after sd failure I set everything up againg (phoscon and ha backup) and then this sensor didnt work as before.

manup, 2019-02-08:
>I have a xiaomi motion sensor with lux, I can see it reports lux everytime I step in front of it but motion is only reported sporadically. This sensor has worked flawlessly before but after sd failure I set everything up againg (phoscon and ha backup) and then this sensor didnt work as before.

Can you please provide the REST-API output of the sensor?

Two things I'd like to check:
* Does the ZHAPresence resource exist
* Are presence reports received

For reports we need to check logs when the sensor triggers, please start deCONZ as follows:

    deCONZ --dbg-info=1 --dbg-aps=2 --http-port=80

M203, 2019-02-08:
Ok, I´d love to help but need some pointers.

Running the official home assistant deconz addon. What to do to provide you with what you need?

> Can you please provide the REST-API output of the sensor?
>
> Two things I'd like to check:
>
> * Does the ZHAPresence resource exist
> * Are presence reports received
>
> For reports we need to check logs [...]

Ok, I´d love to help but need some pointers.

Running the official home assistant deconz addon. What to do to provide you with what you need?

manup, 2019-02-08:
For hassio it's a bit more tricky to debug, you need to enable and connect via SSH:

https://developers.home-assistant.io/docs/en/hassio_debugging.html

In there you need to attach to the deCONZ Docker container:

docker exec -it [container-id] bash

M203, 2019-02-08:
> For hassio it's a bit more tricky to debug, you need to enable and connect via SSH:
> 
> https://developers.home-assistant.io/docs/en/hassio_debugging.html
> 
> In there you need to attach to the deCONZ Docker container:
> 
> docker exec -it [container-id] bash

Ok! I will try this monday!

Gnaget2, 2019-02-08:
I have added a second PIR (same type) and deleted and re-added the first PIR, this was 45 min ago, they are now online and respnd to signal monitor test and is working in phoscon, but still no green lines to them, also noticed that they lack manufactor data and buffer size information

[image]

ErnstEeldert, 2019-02-08:
Is there a way to force an end device to connect via a specific router?

Gnaget2, 2019-02-08:
> 
> 
> Is there a way to force an end device to connect via a specific router?

i wonder the same, woulr be nice to force some sensors to connect to specific node/router

manup, 2019-02-08:
>Is there a way to force an end device to connect via a specific router?

In theory yes, it would be technically possible to force the initial join through a specific router.
However once a device has joined a network it might pick any router it wants.

manup, 2019-02-08:
>also noticed that they lack manufactor data and buffer size information

I guess the node descriptor hasn't been queried.

Can you delete and rejoin the sensor with logging enabled:

    deCONZ --dbg-info=2 --dbg-aps=2 --http-port=80

Hopefully it brings more insight.

Gnaget2, 2019-02-08:
ok, i have now restarted deconz with the to logs on level 2, booth my XIOMI PIR is gone from deconz gui but visible in phoscon but not reacting ni motion.

vill delete them and re-pair them

Gnaget2, 2019-02-08:
one device re-adde, but what shoudl i filter/llok for in the loogs, quit a lot there now

Gnaget2, 2019-02-08:
now when i have re-paird both of them and the mesh is heald i have green lines to booth of them from the GW, but non of them respond to signal monitor test.

i noticed a third sensor (door/window magnet) that also connected directly to the GW and it behaves the same way

manup, 2019-02-08:
>but non of them respond to signal monitor test.

The signal monitor is only usable with Router devices, battery devices won't respond since they sleep most of the time.

Can you provide the log file? Important sections are the joining and when deCONZ tries to query some descriptors.

M203, 2019-02-08:
@manup So in a last attempt to try to fix this before the weekend I removed the xiaomi motion sensor via the phoscon ui. Then I added the sensor to the xiaomi hub. It worked perfectly with the xiaomi hub. I then removed the sensor from the xiaomi hub via the mi home app. Then I added the xiaomi sensor back to deconz via the phoscon ui. It is now working lightning fast again. So wont be able to provide any log in this instance but something happens when adding to the xiaomi hub and then adding it back to deconz.

manup, 2019-02-08:
>It is now working lightning fast again. So wont be able to provide any log in this instance but something happ

Note after setup the sensor reports very fast for some time until it enters normal state and only reports once per minute (if motion stays).

Gnaget2, 2019-02-10:
ok, i have now added one of them again with extened logging. as i dont know what to look for i have everything from the log during.

i started the "search for new sensors at 14:00 and just a few seconds later i resetted the PIR and it was found rather fast

`
13:59:59:001 Poll APS request 18 to 0x0017880103F4E618 cluster: 0x0008
14:00:00:028 APS-DATA.request id: 18, addrmode: 0x03, addr: 0x0017880103f4e618, profile: 0x0104, cluster: 0x0008, ep: 0x01 -> 0x0B queue: 1 len: 5 tx.options 0x00
14:00:00:090 Poll APS confirm 18 status: 0x00
14:00:00:090 Erase task req-id: 18, type: 19 zcl seqno: 76 send time 0, profileId: 0x0104, clusterId: 0x0008
14:00:00:090 APS-DATA.confirm id: 18, status: 0x00 SUCCESS
14:00:00:090 APS-DATA.confirm request id: 18 -> erase from queue
14:00:00:094 read attributes of 0x0017880103F4E618 cluster: 0x0300: [ 14:00:00:094 0x0001 14:00:00:094 0x0003 14:00:00:094 0x0004 14:00:00:094 0x0007 14:00:00:094 0x0008 14:00:00:094 0x4000 14:00:00:094 0x4001 14:00:00:094 0x4002 14:00:00:094 ]
14:00:00:094 add task 1513541 type 19 to 0x0017880103F4E618 cluster 0x0300 req.id 19
14:00:00:094 Poll APS request 19 to 0x0017880103F4E618 cluster: 0x0300
14:00:00:122 APS-DATA.indication srcAddr: 0x0017880103f4e618, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -52
14:00:00:122 	asdu: 184c0100000020fe
14:00:00:122 APS-DATA.request id: 18 erase from queue
14:00:00:123 Node data 0x0017880103f4e618 profileId: 0x0104, clusterId: 0x0008
14:00:00:123 update ZCL value 0x0008/0x0000 for 0x0017880103F4E618 after 16 s
14:00:00:127 APS-DATA.request id: 19, addrmode: 0x03, addr: 0x0017880103f4e618, profile: 0x0104, cluster: 0x0300, ep: 0x01 -> 0x0B queue: 1 len: 19 tx.options 0x00
14:00:00:147 Node 0x00158D00021792F9 is known by 0 neighbors, last seen 0 s
14:00:00:202 Poll APS confirm 19 status: 0x00
14:00:00:202 Erase task req-id: 19, type: 19 zcl seqno: 77 send time 0, profileId: 0x0104, clusterId: 0x0300
14:00:00:202 APS-DATA.confirm id: 19, status: 0x00 SUCCESS
14:00:00:202 APS-DATA.confirm request id: 19 -> erase from queue
14:00:00:227 aps request id: 19 finished, erase from queue
14:00:00:250 APS-DATA.indication srcAddr: 0x0017880103f4e618, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0300, lqi: 255, rssi: -52
14:00:00:250 	asdu: 184d01010000208603000021167404000021b668070000216901080000300200400021172101400030020240002000
14:00:00:251 Node data 0x0017880103f4e618 profileId: 0x0104, clusterId: 0x0300
14:00:00:251 update ZCL value 0x0300/0x0001 for 0x0017880103F4E618 after 16 s
14:00:00:251 update ZCL value 0x0300/0x0003 for 0x0017880103F4E618 after 16 s
14:00:00:251 update ZCL value 0x0300/0x0004 for 0x0017880103F4E618 after 16 s
14:00:00:251 update ZCL value 0x0300/0x0007 for 0x0017880103F4E618 after 16 s
14:00:00:251 update ZCL value 0x0300/0x0008 for 0x0017880103F4E618 after 16 s
14:00:00:252 update ZCL value 0x0300/0x4000 for 0x0017880103F4E618 after 16 s
14:00:00:252 update ZCL value 0x0300/0x4001 for 0x0017880103F4E618 after 16 s
14:00:00:252 update ZCL value 0x0300/0x4002 for 0x0017880103F4E618 after 16 s
14:00:00:311 poll node 00:17:88:01:03:61:d3:5e-0b
14:00:00:311 Poll light node LiamF?nster
14:00:00:313 read attributes of 0x001788010361D35E cluster: 0x0006: [ 14:00:00:313 0x0000 14:00:00:313 ]
14:00:00:313 add task 1513545 type 19 to 0x001788010361D35E cluster 0x0006 req.id 25
14:00:00:313 Poll APS request 25 to 0x001788010361D35E cluster: 0x0006
14:00:00:327 APS-DATA.request id: 25, addrmode: 0x03, addr: 0x001788010361d35e, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x0B queue: 1 len: 5 tx.options 0x00
14:00:00:410 Poll APS confirm 25 status: 0x00
14:00:00:410 Erase task req-id: 25, type: 19 zcl seqno: 78 send time 0, profileId: 0x0104, clusterId: 0x0006
14:00:00:410 APS-DATA.confirm id: 25, status: 0x00 SUCCESS
14:00:00:410 APS-DATA.confirm request id: 25 -> erase from queue
14:00:00:442 APS-DATA.indication srcAddr: 0x001788010361d35e, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -58
14:00:00:442 	asdu: 184e010000001000
14:00:00:442 APS-DATA.request id: 25 erase from queue
14:00:00:443 Node data 0x001788010361d35e profileId: 0x0104, clusterId: 0x0006
14:00:00:443 update ZCL value 0x0006/0x0000 for 0x001788010361D35E after 16 s
14:00:00:618 APS-DATA.indication srcAddr: 0x00158d000217ace6, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -55
14:00:00:618 	asdu: 18bf0a00001000
14:00:00:618 Node data 0x00158d000217ace6 profileId: 0x0104, clusterId: 0x0006
14:00:00:619 update ZCL value 0x0006/0x0000 for 0x00158D000217ACE6 after 300 s
14:00:00:619 no button map for: SP 120 ep: 0x01 cl: 0x0006 cmd: 0x0A pl[0]: 000
14:00:00:619 ZCL attribute report 0x00158D000217ACE6 for cluster 0x0006, ep 0x01
14:00:00:619 	payload: 00001000
14:00:00:628 Node 0x00158D0002178FC9 is known by 0 neighbors, last seen 0 s
14:00:00:760 APS-DATA.indication srcAddr: 0x000d6ffffeca833a, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 239, rssi: -61
14:00:00:760 	asdu: 08270a00001000
14:00:00:760 Node data 0x000d6ffffeca833a profileId: 0x0104, clusterId: 0x0006
14:00:00:760 update ZCL value 0x0006/0x0000 for 0x000D6FFFFECA833A after 300 s
14:00:00:760 ZCL attribute report 0x000D6FFFFECA833A for cluster 0x0006, ep 0x01
14:00:00:760 	payload: 00001000
[... cut ...]
14:00:46:151 APS-DATA.confirm id: 105, status: 0x00 SUCCESS
14:00:46:151 APS-DATA.confirm request id: 105 -> confirmed, timeout 1549807246
14:00:46:183 APS-DATA.indication srcAddr: 0x00178801029c9179, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -52
14:00:46:183 	asdu: 18a30100000020e5
14:00:46:184 Node data 0x00178801029c9179 profileId: 0x0104, clusterId: 0x0008
14:00:46:184 update ZCL value 0x0008/0x0000 for 0x00178801029C9179 after 17 s`

the device i added is 0x00158d0001a8d3a0

manup, 2019-02-10:
Thanks that brings more insights what happens.

So it looks like the sensor doesn't respond to any request (which the Aqara version does).
This causes a bit trouble since at least hourly reports will invoke reading datecode which never suceeds (hence the 0xD0 status codes). This can be fixed and also shouldn't cause the main issue here.

Another assumption in regard to the missing green lines is that the device might not do mac polling on regular base. And some routers tend to remove end-devices which don't poll. Which might also result in presence signals not being forwarded to the gateway.

This should be not problem when the sensor is directly connected to the gateway but it likely is a problem if it routes through a light like Ikea, innr or similar.

Gnaget2, 2019-02-10:
One of them is now connected to the GW and the other one to a hue light close. 

Booth now have green lights. I will monitor and see what happens in the next days

Gnaget2, 2019-02-19:
ok now a week later and they are working as expected, i think the problem was thay they first connected to a node that from time to time could be out of power.

after i turned of all of those noods during start of deconz and all sensors connected to a stable node or GW they have been working fine even when the rest of the light have been turned on.

main problem still exist but from my point of view this can be closed as i have a working routin to not have the problem

manup, 2019-02-19:
>i think the problem was thay they first connected to a node that from time to time could be out of power.

Indeed that's problematic with Xiaomi battery powered devices they are quite stubborn and try to use the parent node even if that is not available anymore. I haven't found a robust solution for that but will do some tests with the Xiaomi gateways to check how it is handled there.

closed 2019-06-26 by stale[bot]