dresden-elektronik/deconz-rest-plugin #1234: Long delay in refire time on light with motion sensors before a new intervall is starting

Ronoon, 2019-02-08:
Just testet what should be a minimal issue but standing in the dark is not so nice.
When you using, in this case, an Hue motion sensor an 2 Aqara motion detector to start the light in your very dark cellar an after the lights go out because the was no motion an then you have to wait for 12 sec. before the next motion is used to start the lights again looks like it is a bit much of delay.
And yes the motion is detected by the sensors but the Phoscon app/ DeConz is not responding
for the time being.
If the lights are in the dimming phase and there is a new motion detection, then the "timer" will be fired up again and the lighting cycle is complete reset. Not a bad thing in my opinion but why the delay at the end of the lightphase before firing up again?

manup, 2019-02-08:
>Not a bad thing in my opinion but why the delay at the end of the lightphase before firing up again?

The idea behind this was a sanity when the motion sensor state machine enters the `off`state. Mainly when the off state was triggered by an action other when timeout like turining off a group via app or a switch. So for example when turning off the lights via switch while still being in the room, they shouldn't be turned on again a second later, there should be enough time to leave the room.

I'll check the state machine, it will be better to enter idle state in case off a timeout and sanity state in case of switch/app off action.

Ronoon, 2019-02-09:
Well thats where the snake bites the snail :-)
Maybe adding a option to the motion sensor would do it in that case, like fire up immediatly or delayed by xx sec after last motion through the phoscon app ??
so the user can switch the behavior of the motion sensor and therefor the lighting

ebaauw, 2019-02-09:
Please note that the behaviour of the motion sensor is determined by the motion sensor's hardware and firmware, not by deCONZ.  In the wonderful world of Zigbee standardisation, different types of motion sensors behave differently.  You could try contacting the manufacturer for a firmware update (assuming the sensor's firmware is upgradable), but it's probably easier to buy another type of motion sensor.

Some sensors, like the Trådfri motion sensor, only send a message when detecting motion.  To clear `state.presence`, the deCONZ REST API plugin provides a courtesy timer, `config.duration`.  Afaik all motion sensors take some time to "recharge", before sending the next motion detected message.  If you set `config.duration` too short, you're left in the dark.

Other sensors, like the Hue motion sensor, send a message when detecting motion, and another one when no longer detecting motion (after the sensor has "recharged").  The deCONZ REST API plugin uses the first message to set `state.presence`, and the second message to clear it.  The (additional) time to wait before sending the second message is configurable on the sensor, the API plugin exposes this as `config.delay`.

I also ran into an issue with my Hue motion sensors, when using rules to turn on lights only when it's dark.  The rule has two conditions, one on `state.presence` and one on `state.dark`.  The latter is updated a bit later than the former, so when the next motion detected message arrives too early, `state.dark` is still false (from when the lights were on, depending on `config.tholddark`), and the rule won't fire to turn on again the lights.

Ronoon, 2019-02-09:
@ebaauw 
I think you missunderstand this  because this is not a problem on the zigbee side but a programming solution done by manup.
The zigbee side is okay but the delay programmed by manup should be different according to manup (see above his comment) was a solution to allow leaving the room and preventing the restarting of the lights for an amount of time.
But this is not every time useful in the way it is now, so i asked it to be a user solution if it  is used or not by programming the delay to be switched on/off by the user.

I think your right about the zigbee side but thats not the discussion  in here i think.

bluemoehre, 2019-03-26:
I think I ran into the same issue with my Philips Motions Sensors + IKEA Spots and was trying for weeks to fix it via configuration. 
The issues I noticed so far:
- lights go off, new motion does not trigger for a long while (approx. 1min)
- while lights are fading out, new motion does nothing
- lights are fading out, new motion stops fading - brightness stucked
- lights are fading out, new motion brings them up again, after some seconds lights get turned off completely
- light are fading out, getting stucked on lowest brightness - never turn off

Some of these issues could easily be race conditions. If I can help you guys test some behaviour let me know. Just upgraded to latest relase right now.

Ronoon, 2019-03-26:
Well thats what i was talking about.
Since the change to .60  i think there is a better behavior in this case (maybe also done by the firmware upgrade).
The lights are not stucked at the moment but this could also an issue with switching my "normal" lights from INNR GU 10 Type RS 125 to Paulmann GU10 lights.
Allover i think there has to be more investigating in this issue cause i´m not alone :-)

bluemoehre, 2019-04-02:
With the current new version I experienced these two issues the past days only:
- light are fading out, getting stucked on lowest brightness - never turn off
- while there is continuous movement, after some minutes lights fade out and turn off.

closed 2019-08-07 by stale[bot]

bluemoehre, 2019-09-16:
This is still an issue.. I can see in the sensors state "motion detected" but lights keep dimming and switching off. How can I help fixing this issue?

thisisbenwoo, 2020-05-01:
I agree that it's still an issue. Here's a great example ...
The period of time after 8:26 there was activity, and motion should have continually be set as detected.

[image]

Ronoon, 2020-05-01:
Well my solution to this problem is to use the Deconz only as a gateway to my lights and do all the programming for timers, delays etc. through Home Assistent (Hassio). Because you have no options in timing the delays with the software/userinterface of Deconz/Phoscon App this seems to be one solution to the timing probs. Also there is the use of FHEM or any other managment software with programming abilities for your use if you are willing to do the programming by yourself.

thisisbenwoo, 2020-05-01:
I am using Hassio to do all the programming (through Node-RED), but the problem is that Philips Hue Motion Sensor isn't detecting motion after a while, and reports no activity.

Ronoon, 2020-05-01:
Well .... 
1. did you do the firmware update from Phillips?
2. did you configure the delay times under attributes under 10 at 0x0010
3. last of my probs was an nearly empty battery ( my wife got a bit angry about that :-( )
and for you to read here a part of my log from today with a 20 sec delay between detection

[image]

thisisbenwoo, 2020-05-01:
How do you update the firmware of the sensor without using the Hue hub/Hue app? I'm only using deCONZ. Also, not sure how to do the configuration either. as for the battery, it's at 82%

Ronoon, 2020-05-01:
Okay thats a problem without a proper bridge and allmost imposible to update to the latest firmware from phillips.
to configure the delay time look here :
[image]

double click on the 12 and the dialog shows up like these:

[image]