dresden-elektronik/deconz-rest-plugin #1235: Only one Tradfri Bulb is stuck

fbrinker, 2019-02-08:
This may be a little bit off topic, but I'm not sure what causes the problem.

The Tradfri bulbs are not very reliable, yeah. But I have a weird problem now and don't have any idea to debug the current problem. Any ideas would be great.

My setup: 
* Raspi 3 with Conbee Stick and Deconz Docker-Container
* 1 Tradfri Wall Plug
* 1 Tradfri Dimmer
* 4 Tradfri 1000lm E27 bulbs
* 1 Tradfri GU10 bulb

Everything worked fine since one 1000lm E27 suddenly stopped responding. It went to 100% brightness and decided to stay there forever. I've tried resetting it but I can't see any response. I've even built a reset-box with an Arduino and a 230V transistor and can reset every other Tradfri device on the first try (very nice, struggled often with the reset haha) but I can't see any reaction with the described bulb.

I've also tried to reset the entire Gateway. It found every Tradfri device on the first startup, but not the crazy hung bulb.

Is there any chance to reset that thing?
Or to find out what's wrong?

manup, 2019-02-08:
By reset you mean the reset sequence?

https://www.youtube.com/watch?v=mJm9YpPrGzk

* Start light search
* Quickly 6 times on/off
* Bulb flickers
* Once again off/on

Usually this works quite well, I found it important to execute the sequence quickly.

Note also if ConBee is directly connected to the Raspberry a USB extension cable often improves the signal a lot.

fbrinker, 2019-02-08:
Yes, the reset sequence with my Arduino and SSR (800ms off, 500ms on x6) works like a charm for every Tradfri bulb I own, but not for the problematic one since it got stuck at 100% brightness.

I guess, since I also can pair all of the other bulbs with the Conbee stick, the problem is that I can't reset the bulb anymore via reset sequence and I can't find another working sequence either.

Thank you for the hint with the extension cable, added it to my setup :)
Still now luck with the one bulb but it may prevent other issues in the future

Seems like a problem with the bulb itself since I can't reset it anymore neither Phoscon can find it :/

Is there any way to "sniff" for Zigbee singals from that lamp with the Conbee stick?

closed 2019-02-08 by fbrinker

manup, 2019-02-09:
>Is there any way to "sniff" for Zigbee singals from that lamp with the Conbee stick?

Yes sniffing can be done with ZSHARK, but note that ConBee can't be coordinator and sniffer at the same time.

https://www.dresden-elektronik.de/funktechnik/products/software/zshark/?L=1

fbrinker, 2019-02-09:
Ah good to know :)

But since I can't reset the bulb, I guess there couldn't be anything helpful to sniff for?
Or is it possible to "force pair" a bulb to the USB stick? :thinking:

manup, 2019-02-09:
>Or is it possible to "force pair" a bulb to the USB stick?

I'm afraid no, reset sequence is usually the last resort.