dresden-elektronik/deconz-rest-plugin #1237: Problems pairing Trådfri E27 1000lm

joubbi, 2019-02-09:
I bought a Trådfri set with a 1000lm bulb and a rotating dimmer.
I'm able to pair the dimmer with Phoscan without problems.
I'm able to pair the dimmer with the bulb without problems.
I'm NOT able to pair the bulb with deCONZ.
I have spent several hours with this now.

Any ideas?
The bulb should be OK since it works with the dimmer.
I have read this thread and done everything according to the instructions:
https://github.com/athombv/com.ikea.tradfri/issues/4

I have a TRÅDFRI bulb E27 CWS opal 600lm and a FLOALT panel WS 30x30 paired without problems.
These were paired with earlier software and firmware versions though.

deCONZ:
Commercial Version -  2.05.58
Build with Qt 5.5.1
Commit a9a32542be9eb582635e6776b238cc5fc43c6afd
Firmware version 0x262F0500

manup, 2019-02-09:
You need to reset the bulb

https://www.youtube.com/watch?v=mJm9YpPrGzk

* Start light search
* **Quickly** 6 times on/off
* Bulb flickers
* Once again off/on

joubbi, 2019-02-09:
I have done that several times.
The bulb never flickers when I do the reset.
So I'm not sure that it resets.
The bulb flickers however when I pair it with the dimmer.

Reading the other thread there seems to be others with the same kind of issue with a bulb that doesn't flicker when resetting. I have tried with both short and long off periods when trying to reset it.

Smanar, 2019-02-09:
Have you check the youtube video ? https://www.youtube.com/watch?v=mJm9YpPrGzk

manup, 2019-02-09:
Ah ok, hmm have heard about this before but honestly I don't know why the bulb won't reset.

You may try to keep the light un-powered for a few minutes so that the electronic (mainly capacitors) can calm down.

Earlier today I couldn't join the Ikea outlet, which just worked now.

joubbi, 2019-02-09:
I've tried it so many times now with different timings. This is very frustrating.
What does the dimmer do in order to pair without a reset of the bulb?

manup, 2019-02-09:
>What does the dimmer do in order to pair without a reset of the bulb?

It forces the bulb to join a specific network via Touchlink. This is different from the way deCONZ pairs devices, which is named classical join – usually supported by all devices.

Momo592, 2019-02-09:
I have the same problem. I can not connect my Ikea TRÅDFRI bulbs with my Phascon App on a Raspberry 2nd gen with Rasbian, RaspBee and deCONZ (headless). My first bulb (small E14) do not flickers by reset, so I thought the reset is not suggest. But today I brought a second bulb (big E14 with color). This one flickers by after the reset. But in Phoscon is no light found. :(

Momo592, 2019-02-09:
Update: With the big E14 bulb it is done. After many times with reset it has connect.

manup, 2019-02-09:
@Momo592 do you use latest version 2.05.58?

SirMaggot, 2019-02-09:
I had the same issue, but solved it by going to "Phoscon ui -> help".
You will see a link to the "Old WebApp version", Click on it.
Within the old ui, click on "settings -> scroll down" and click on "Scan for device"
My light was on the wrong channel, weird, but clicking on "reset", then open network, fixed my problem.
(Flip the powerswitch once after the network is open)

Paired 3 Ikea bulbs this way.
On version 2.05.58.

joubbi, 2019-02-10:
Thank you @SirMaggot !!
I was just about to return the lightbulb to Ikea and ask them to shove it up someplace dark.
The "Scan for devices" in the old UI did the trick. It was straight forward just as @SirMaggot explained.
The light was on channel 11 and my gateway on channel 20. Pairing worked without a problem in Phoscon APP after I did a reset from the old UI.

Shouldn't there be the same kind of button in Phoscon App?

Momo592, 2019-02-10:
@meanup yes I use this version. 

@SirMaggot: Nice to know! I will test this with my other Ikea bulb. :)

loz7777333, 2019-02-12:
@SirMaggot you are a god amongst men, thank you for this work around

manup, 2019-02-16:
Guess this can be closed now.
Note to my self: add wiki page on various lights joining procedures + reset sequences.

closed 2019-02-16 by manup

Dinth, 2019-11-24:
> I had the same issue, but solved it by going to "Phoscon ui -> help".
> You will see a link to the "Old WebApp version", Click on it.
> Within the old ui, click on "settings -> scroll down" and click on "Scan for device"
That's "Touchlink reset" right?
Have you performed 6x power cycle while doing it? Have you kept the gateway close to your bulb?