dresden-elektronik/deconz-rest-plugin #1238: Login required every time on iOS

saurabh984, 2019-02-09:
Phoscon App requires Login using password every time on iOS. This has let to over 200 API keys being created. 

Is there a way to set a single API key for Phoscon App rather than login to create a new key?

manup, 2019-02-09:
Normally the API key is stored in browser cache and is reused. Are you using private mode or something similar which clears the cache / localStorage?

Larsn1, 2019-02-09:
I have the same problem and I am not using different browser settings as before. Thank you for reporting this.

saurabh984, 2019-02-10:
Same thing happens on macOS but less compared to safari on iOS. I’m not in privacy mode. 

Safari on iOS  does warn about the site being insecure when the login page appears. Could that be the reason why session store is not persisted?

stevenwfoley, 2019-02-11:
I am also experiencing this same issue on my Android phone: Google Pixel 2 XL. I'm not in privacy mode, and have not changed any browser settings. I first noticed this in v2.05.57.

ademobile, 2019-02-11:
It is an absolute pain having to log in nearly every time on iOS or MacOS. Can we have a tick box that does away with authorisation please?

manup, 2019-02-11:
I'll do some tests, there seems to be a bug.
As mentioned before it shouldn't be needed to login every time since the login details are stored in the local browser cache.

daaaaa, 2019-02-20:
I have just updated to `V2_05_59`, but this still happen on macOS using chrome.
The session is not even kept, if I open a new tab. Maybe are you using session storage rather than local storage?

manup, 2019-02-20:
>The session is not even kept, if I open a new tab. Maybe are you using session storage rather than local storage?

Both :)

2.05.59 does add new parameter `gwid` to the url to fix the issue when session storage expires. In this case the gwid will be used to load the configuration from the local storage.

URLs in Phoscon App should now look like:

    http://192.168.2.4:8090/pwa/devices-lights.html?_v=0f1f703d5d57e997131c7ae41ce847256c493e70,1,3,17&gwid=00212EFFFF00C5FB

daaaaa, 2019-02-20:
I confirm that my url is in the form showed by you. This is mine:
`http://192.168.1.6/pwa/index.html?1550171194000&gwid=00212EFFFF0279AF#control/group/1`

Session is maintained only if I save the bookmarks with the `gwid` in the url.
After a login, If I open a new tab with this url `http://192.168.1.6/pwa/`, the browser redirect me to the login. Is this the expected behavior?

ademobile, 2019-02-21:
As I mentioned before, it would be a good if there was a way to switch of authentication as I am still being asked every time to log in (Safari, iOS and Firefox) and especially with my issue of the motion sensors stopping working (as one did this morning).

Smart control is supposed to be just that, however designers don't think about speed of control. With a light switch, you press it and job done. When things are not working as expected, you have to unlock your smart device, call up a web page, log in, find what you need, by which time everyone is having a go at you because a switch just works!

If you must have authentication for the web page then how about light control (i.e on/off, brightness, colour/temperature, scenes) being default and unprotected (you don't password protect a switch). Anything else then a log in is required (i.e. add new devices, change configuration, edit scenes etc.)

closed 2019-06-29 by stale[bot]