dresden-elektronik/deconz-rest-plugin #1239: ConBee not connected

jannnfe, 2019-02-09:
Hello,
I have a ConBee and want to use it with deConz in conjunction with an Ubuntu server system. Installation of deCONZ worked but in the Phoscon app, "Not connected" can be found under Gateway. I disabled deconz-gui because I do not have a GUI on my server system anyway. How do I get the ConBee to work?

Second question. How can I permanently assign a new IP address to the whole deCONZ system?

Thanks!

manup, 2019-02-09:
The user which runs deCONZ needs to be in the group `dialout`.

    sudo gpasswd -a <user> dialout

A restart (or relogin) is needed afterwards.

To start headless version of deCONZ use the deconz.service instead of deconz-gui.service as described here: https://github.com/dresden-elektronik/deconz-rest-plugin#headless-support

>Second question. How can I permanently assign a new IP address to the whole deCONZ system?

The IP is just used as configured in the operation system so you need to check Ubuntu network settings.

jannnfe, 2019-02-09:
Hey @manup 
thanks for your quick reply. Unfortunately, this proposal has not changed anything. I also pulled out the ConBee again and plugged it into another USB port, unfortunately without success. Do you have another idea?

> The IP is just used as configured in the operation system so you need to check Ubuntu network settings.
Sorry I did not mean IP but Port. Actually, another application is currently running on port 80, which is why I would like to permanently put deCONZ on a different port.

Thanks! :)

jannnfe, 2019-02-10:
I have tested the ConBee on a Windows computer and get the following network settings:
[image]

Output of`ls -l /dev/ttyUSB0`:
`crw-rw---- 1 root dialout 188, 0 Feb 10 12:25 /dev/ttyUSB0`

And Output of  `groups`:
`root dialout`

I also have an RFLink on my Ubuntu server via USB. Is something similar to RFXcom. Already tested connected without the RFLink, unfortunately, it does not work then?

Another interesting point is that I can find on my Windows PC the Philips Hue Bridge that is connected to the network in Phoscon but not on my Ubuntu server system?
Maybe this new information will help solve the problem?

manup, 2019-02-10:
>Sorry I did not mean IP but Port. Actually, another application is currently running on port 80, which is why I would like to permanently put deCONZ on a different port.

The `--http-port=<port>` can be used for that. Most easiest is to override the systemd service file via:

    sudo systemctl edit deconz

And override startup variables from the default deconz.service file.

>I also have an RFLink on my Ubuntu server via USB. Is something similar to RFXcom. Already tested connected without the RFLink, unfortunately, it does not work then?

Maybe some other process is accessing the file. What is the output of:

    sudo fuser -v /dev/ttyUSB0

jannnfe, 2019-02-10:
Hey @manup 
thanks for answering my issue and try to help me :)

While ConBee is connected via USB the command: `sudo fuser -v /dev/ttyUSB0` provide no output, so no other process is using it?
When I disconnect ConBee and try to execute the command the output is:
`Specified filename /dev/ttyUSB0 does not exist.`

Is there anybody who uses ConBee with Ubuntu Server? :(
Maybe @manup still has an idea? Thanks :)

jannnfe, 2019-02-10:
I tried to install [the drivers](https://www.ftdichip.com/Drivers/D2XX.htm) for the FTDI ConBee USB Stick according [to this manual](https://www.youtube.com/watch?v=jynlynjOOek&feature=youtu.be). Unfortunately, without success. I am really desperate that there is no one who may already use my System configuration?

manup, 2019-02-10:
Usually on Ubuntu it just works no need to install any driver.

For further debugging please do:

    sudo GCFFlasher_internal -l

This should print all detected devices which could be used by deCONZ.

Check that only one instance of deCONZ is running:

    ps ax | grep deCONZ

Further start deCONZ with logging enabled (stop other deCONZ instances):

    deCONZ --dbg-error=1 --dbg-info=2

jannnfe, 2019-02-10:
Hey,

> sudo GCFFlasher_internal -l

```
GCFFlasher V2_11 (c) dresden elektronik ingenieurtechnik gmbh 2017/12/10

       1  FTDI device found

 device | vendor | product | serial  | description
--------|--------|---------|---------|----------------------
   0    | 0x0403 | 0x6015  | DO01MI8M | FT230X Basic UART
```

> ps ax | grep deCONZ

```
 1049 ?        Ss     0:03 /bin/bash /usr/bin/deCONZ-update2.sh
39466 ?        Rsl    0:00 /usr/bin/deCONZ -platform minimal --http-port=8081
39544 pts/0    S+     0:00 grep --color=auto deCONZ
```

After `service deconz stop`:
```
 1049 ?        Ss     0:03 /bin/bash /usr/bin/deCONZ-update2.sh
39544 pts/0    S+     0:00 grep --color=auto deCONZ
```

> deCONZ --dbg-error=1 --dbg-info=2

How to use this command? Where do I have to execute this?

manup, 2019-02-10:
>How to use this command? Where do I have to execute this?

Just run in in the terminal it will output quite some debug messages.
I forgot since this is a headless setup better use:

    deCONZ --dbg-error=1 --dbg-info=2 -platform minimal

jannnfe, 2019-02-10:
Okay @manup 
Starting deCONZ with this Command worked and a new Gateway is created in that is the ConBee available!!!
`deCONZ --dbg-error=1 --dbg-info=2 --http-port=8081 -platform minimal`
But when i stop the process and start again with `service deCONZ start` ConBee is "Not connected" again!
Strange. Something with my service file seems to be wrong? Maybe because of the headless setup?

Output of `deCONZ --dbg-error=1 --dbg-info=2 --http-port=8081 -platform minimal`:

```
23:13:36:533 HTTP Server listen on address 0.0.0.0, port: 8081, root: /usr/share/deCONZ/webapp/
23:13:36:544 CTRL. 3.11.023:13:36:577 COM: /dev/ttyUSB0 : FT230X Basic UART
23:13:36:577 COM: /dev/ttyACM0 : 0042
23:13:36:577 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt
23:13:36:656 parent process bash
23:13:36:656 gw run mode: normal
23:13:36:656 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version
23:13:36:657 DB sqlite version 3.11.0
23:13:36:657 DB PRAGMA page_count: 38
23:13:36:657 DB PRAGMA page_size: 1024
23:13:36:657 DB PRAGMA freelist_count: 0
23:13:36:657 DB file size 38912 bytes, free pages 0
23:13:36:657 DB PRAGMA user_version: 6
23:13:36:657 DB cleanup
23:13:36:657 DB create temporary views
23:13:36:658 DB view [0] created
23:13:36:658 DB view [1] created
23:13:36:658 DB view [2] created
23:13:36:658 DB view [3] created
23:13:36:658 sql exec SELECT apikey,devicetype,createdate,lastusedate,useragent FROM auth
23:13:36:658 sql exec SELECT key FROM config2
23:13:36:658 sql exec SELECT key,value FROM config2
23:13:36:658 Load config from db.
```

jannnfe, 2019-02-10:
Started deCONZ with `service deCONZ start`.
Both have the same Groups and Settings and both with ConBee "Not connected" (Inside the Settings "Reset Gateway" not possible, it says Error:
[image]

Stopped deCONZ service and started deCONZ with `deCONZ --dbg-error=1 --dbg-info=2 --http-port=8081 -platform minimal`.
Only one Gateway available with ConBee connected as it should but a "fresh" Gateway without the groups?:
[image]

manup, 2019-02-11:
>But when i stop the process and start again with `service deCONZ start` ConBee is "Not connected" again! And now there are two Gateways in the Phoscon Login Page?!

Please check the service that it uses your `user id` (not root).
If the user has not the id 1000 (default user id in the deconz.service) this likely causes the issue.

jannnfe, 2019-02-11:
Hey @manup 
thanks for your feedback again!
The "service deCONZ start" is startet by user jannnfe with user id 1000 and that's not working.
When I start deCONZ with "eCONZ --dbg-error=1 --dbg-info=2 --http-port=8081 -platform minimal" user root is starting and root has id 0 and with root its working. So I have to change the user that starts deCONZ. 
Maybe the best way to delete deCONZ completely and reinstall again? How to do that or do you have an idea to change the user?

manup, 2019-02-11:
>Maybe the best way to delete deCONZ completely and reinstall again?

No that won't change the configuration.

To me it looks that user jannnfe is not in the group dialout.

Did you exec the needed command to put it in the group?

    gpasswd -a jannnfe dialout

(restart or logout/login afterwards)

jannnfe, 2019-02-11:
> To me it looks that user jannnfe is not in the group dialout.

You are the best. Thank you for your help. Now its working! :)

closed 2019-02-17 by jannnfe

bramschats, 2019-08-17:
I have the same problem with my received Conbee II. I connected the Conbee II to my ubuntu 18.04 instance (installed on my NUC) but it is not discovered. All other USB devices (also on that port) are discovered but not the Conbee II stick. the device is not connected within the Deconz instance. What can be the problem here and how can it be solved?

Smanar, 2019-08-17:
Hello, have you tried all the classic tests ? > https://www.phoscon.de/en/support#conbee2-connection-issues.

bramschats, 2019-08-17:
Hello Smanar,

Thanks for the reply, with the extension cable the dongle is recognised. 

Hope It will work now!

Bram Schats
bram.schats@gmail.com
+31611417660

bramschats, 2019-08-19:
> bram.schats@gmail.com <mailto:bram.schats@gmail.com>

Hello,

I am still having problems pairing some devices:
HUE bloom		I do not have any remote so I have to do delete the HUE via the app. But even when I do that I cannot see the bloom HUE lights to show in Deconz….?
Aqara transmitter.   Transmitter 2-gangWXKG02LM Rev.2 https://github.com/dresden-elektronik/deconz-rest-plugin/issues/744#issuecomment-522298910 <https://github.com/dresden-elektronik/deconz-rest-plugin/issues/744#issuecomment-522298910>?
Aqara smoke detector.   I have the sensor as shown in the image beneath and It can connects to Deconz. The problem is that is shows up in Hassio but it does not show under Deconz. However it looks like it is working because it sends it’s state to Hassio?

All other things seem to work just fine, it would be nice to solve these latest issues!

Regards,

Bram Schats
bram.schats@gmail.com
+31611417660

Smanar, 2019-08-19:
And for the HUE bllom without remote, you can try the touchlink feature on the old webapp (in phoscon/help/Old WebApp version) but I have never used it.

bramschats, 2019-08-20:
unfortunately also the touchlink option in the old web app isn't working for my Philips Hue lights! Can you also confirm the smoke detector is working without being shown within the phoscon device list?

ebaauw, 2019-08-20:
Be sure to power cycle the light.  Hue lights only respond to touchlink for a limited time after powering on.  Keep the light and the RaspBee/ConBee close to each other (< 1m).