dresden-elektronik/deconz-rest-plugin #1240: Aqara / Xiaomi Wall Switch Pairing

MornLum, 2019-02-10:
Hallo,

I have bought a one button and a twi button wall switch from Aqara / Xiaomi. It is the one, which can also switch one and off from the devices.

[https://www.aqara.com/en/ctrl-neutral.html](url)

I am using the current version of DeCONZ / Phoscon and have a Raspberry Pi with Zigbee device. All other devices from Aqara/Xiaomi and also Ikea and Hue are working well.

 I tried everything I found in the internet to get them connected. They were not found. I tried paring in Phoscon as sensor, switch and also light. Never found anything. I reduced the distance and put it directly in front of the Raspberry Pi. No sucecess. The switching itself is working.

manup, 2019-02-10:
>I am using the current version of DeCONZ / Phoscon and have a Raspberry Pi with Zigbee device.

* Which version/firmware version?
* RaspBee or ConBee?
* Did you reset the device while switch search is running?

   When I recall correctly to reset the switch a button must be pressed for >10 seconds and the LEDs should start blinking.

MornLum, 2019-02-10:
- I am using a RaspBee
- Sorry to ask, but how to read the firmware of the RaspBee (It could not be old, because I started with it this year)
- I pressed the button very long:
     After some second, the two blue lights getting red and are blinking.
     Then, the connected light bulb is going of
     Then, the two lights are blinking blue

In Phoscon, should I seach for a sensor or a switch?

MornLum, 2019-02-10:
I just found the solution after trying it again with the two button switch.
Besides I only was searching for switch and sensor, there appiered two addition lights. These lights are the two buttons of the wall switch. A little bit confusing is that the one button wall switch seems also to create two lights, only one having a function.

ebaauw, 2019-02-10:
> A little bit confusing is that the one button wall switch seems also to create two lights, only one having a function.

Yes.  It has the same Zigbee fingerprint as the two-button variant.  The second (non functional) light endpoint still needs to be ignored by the REST API plugin.

manup, 2019-02-10:
The version is shown in the gateway settings.

[image]

closed 2019-06-17 by stale[bot]