dresden-elektronik/deconz-rest-plugin #1241: Halting Trådfri dimming once started

PremiumUsername, 2019-02-10:
I would like to know how the dimming is working behind the scenes for IKEAs bulbs. If you use transitiontime in a rest command is the bulb given one command and left to take care of the dimming by itself or is it deconz that produces many small changes in brightness that produces the smooth dimming?

If it is the former, is it possible to halt the dimming process somehow after it has started?

This is the use case I had in mind. Using app-daemon in HA,  I would like to use an IKEA trådfri remote to just start dimming with the button-down event and stop it using the button up event. This would result in the smoothest transition I imagine. I just don't know how to halt the dimming (which would go to max or min unless it is halted) without setting a new brightness. Is there a halt command?

lbschenkel, 2019-02-11:
Yes you can, that is what my motivated my PR to HASS:
https://github.com/home-assistant/home-assistant/pull/17722

That's exactly so I could do it via AppDaemon by sending `bri_inc` commands.

PremiumUsername, 2019-02-11:
Brilliant! Thank you! I would never have discovered that without your help. I can't see any mention of 'bri_inc' in the deconz rest API help pages. 

It works perfectly, except for one thing. It seems that when you use bri_inc for a light group (not a single light) HA no longer gets any status updates for brightness. have you noticed the same thing?

lbschenkel, 2019-02-11:
Not that I can recall, no. Could you give step-by-step details so I can try to reproduce?

PremiumUsername, 2019-02-11:
Ok. this is an appdaemon test app that shows the problem if you run it with a light group.

```
class HelloWorld(hass.Hass):

    def initialize(self):
        self.log("Hello from AppDaemon")
        self.log("You are now ready to run Apps!")

        self.selections = {'entity' : 'light.zigbeegroupkitchenisleall', 'field' : "/action"}
        self.selection_three = {'entity' : 'light.zigbeegroupkitchenisleonlythree', 'field' : "/action"}
        self.selection = {'entity' : 'light.kitchen1', 'field' : "/state"}     
        self.timedelay = 5
 
        

        self.log("Resetting")
        self.call_service("deconz/configure", **self.selection, data = {"on": True, "bri": 1})
        self.run_in(self.test, 1) 
        time.sleep(0.1)
        self.log("starting in 1 sec. brightness is now: " + str(self.get_state(entity = self.selection.get('entity'), attribute = 'brightness')))
    
    def test(self, entity):

        self.log(str(self.timedelay *2) +" second  increase of brightness starting")
        self.call_service("deconz/configure", **self.selection, data = {"on": True, "bri_inc": 254, "transitiontime": (self.timedelay*2*10)})
        self.run_in(self.stop, self.timedelay) 

    def stop(self, entity):
        self.log(str(self.timedelay) +" seconds have passed, trying to stop")
        self.call_service("deconz/configure", **self.selection, data = {"on": True, "bri_inc": 0})
        self.run_in(self.result, self.timedelay +1)
        
    def result(self, entity):
        self.log(str(self.timedelay *2) + " seconds have elapsed. brightness is: " + str(self.get_state(entity = self.selection.get('entity'), attribute = 'brightness')))
 
```

PremiumUsername, 2019-02-11:
To be clear both work as in the light slowly increases. It's only get.state that doesn't work. Same in HA panel. 

With single light:

```
INFO hello_world: starting in 1 sec. brightness is now: 1
INFO hello_world: 10 second  increase of brightness starting
INFO hello_world: 5 seconds have passed, trying to stop
INFO hello_world: 10 seconds have elapsed. brightness is: 127
```

With Light group: (114 is the last bri that was set in HA) In reality it does start from 1 and ends somewhere in the middle in terms of brightness

```
INFO hello_world: starting in 1 sec. brightness is now: 114
INFO hello_world: 10 second  increase of brightness starting
INFO hello_world: 5 seconds have passed, trying to stop
INFO hello_world: 10 seconds have elapsed. brightness is: 114
```

closed 2019-06-19 by stale[bot]

KennethLavrsen, 2019-09-19:
A long time later and I just implemented dimming using the deconz configure service in Home Assistant as described in https://community.home-assistant.io/t/here-is-a-good-method-for-dimming-lights-in-deconz/138115

I am posting here to say that when I use this method, the brightness state is not updated instantly. But after a few seconds it is. So maybe things changed since this bug report was axed by the stale bot. I post this here so people finding this bug report do not fear using the method. For all practical use by a human it works fine. It is not important that apps and displays from Home Assistant takes a few seconds to update the brightness.