dresden-elektronik/deconz-rest-plugin #1245: NYCE Door/Window Sensor NCZ-3011-HA

stevenwfoley, 2019-02-11:
Hello,
According to issue #548, it looks like this device was suppose to gain Phoscon support in 2.05.57. I'm running 2.05.58, but I cannot get this to show in Phoscon. I can see it in deconz. Is there a special joining procedure? Or for those that have successfully discovered this device with Phoscon, do you have any advice or suggestions?

Thanks!

bdickhaus, 2019-02-11:
Issue 548 was only for the ceiling motion sensor NYCE NCZ-3043-HA. Each of the NYCE sensors has to be defined in deCONZ. There is also the garage door tilt sensor NYCE NCZ-3014-HA. I have one of those but haven't taken the time to capture all the info yet.

stevenwfoley, 2019-02-11:
That's how 548 started, but one of the posts asked about support for the 3011 device and the release notes for 2.05.57 say it added support for both NYCE devices, so I assumed it was because of 548. The supported devices wiki also says its supported.

manup, 2019-02-11:
Can you please provide the device info as described in the  wiki:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

Currently there is code for the following two sensors:

    { VENDOR_NYCE, "3011", emberMacPrefix }, // NYCE door/window sensor
    { VENDOR_NYCE, "3043", emberMacPrefix }, // NYCE ceiling motion sensor

This indicates the NCZ-3011-HA should work but I can't test it here since I don't own the sensor.

wvuyk, 2019-02-11:
> Hello,
> According to issue #548, it looks like this device was suppose to gain Phoscon support in 2.05.57. I'm running 2.05.58, but I cannot get this to show in Phoscon. I can see it in deconz. Is [...]

How does the sensor show in deCONZ? As 0xnnnn or as "name nn"?

stevenwfoley, 2019-02-12:
Here is the requested info from deCONZ. Please let me know if you need any further info. And thanks for adding support!

[image]
[image]
[image]

bdickhaus, 2019-02-12:
This is for the tilt sensor NYCE NCZ-3014-HA

[image]
[image]
[image]
[image]
[image]
[image]
[image]

manup, 2019-02-12:
Thanks the NCZ-3014-HA garage door/tilt sensor is added now.

@stevenwfoley unfortunately the basic cluster doesn't show the needed attributes like modelid. Can you make another screenshot after reset and pairing the sensor again. This should populate the attributes.

stevenwfoley, 2019-02-13:
So after resetting the NCZ-3011-HA door/window sensor device a couple times, it joined the network and showed up in Phoscon as expected. I've included updated screenshots for completeness.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

closed 2019-06-20 by stale[bot]