dresden-elektronik/deconz-rest-plugin #1247: Xiaomi not meshing, almost at all

TroLoos, 2019-02-11:
Hi, I've started to add my Xiaomi Wall Switches to Conbee/Deconz managed network, and as for this moment - their behaviour seems rather strange to me...

As you can see on below mesh visualisation, none of the switches connect via other router. The only deviced that connected indirectly is Aqara Temp/Humid/Press sensor.

[image]

I'm not sure what is the limit for direct Conbee connections, but I've already added 12 devices, and this number will change to ~30 in near future. If they don't start meshing, I'm sure I will hit some limit :-( What is also surprising - some of the switches have long way to enforce direct connection... and they still prefer to do that rather than use other, stronger path via other router.

Do you observe similar things in your networks?

nunoary, 2019-02-11:
Yes. I have 12 Xiaomi smart plugs added and none of them behaves like a router, as it would be supposed. This behavior occurs since I installed the latest version (V.58).
I do not know if this problem is only with Xiaomi equipment, but I have an Ikea lamp and this works fine.

manup, 2019-02-11:
>I do not know if this problem is only with Xiaomi equipment, but I have an Ikea lamp and this works fine.

Believe me they mesh, the reason the network lines aren't visually populated is that the requests to Xiaomi router devices have been drastically reduced to not bring them in some error state.

We've seen problems where Xiaomi switches have been gone offline until they were power-cycled.

nunoary, 2019-02-11:
Thank you for the clarification. It's a pity not to see the mesh, but what matters is that they do. I can live without the lines :)

closed 2019-02-14 by manup