dresden-elektronik/deconz-rest-plugin #1248: Wrong group type for groups created by sensor

Korhm, 2019-02-11:
Hello,

I have a group that contains 2 Hue bulbs. This groups is also associated to a Philips Dimmer Switch.

The group with id 4788 was created automaticaly and has a "devicemembership" which matches my Dimmer Switch.
But This group is also of type "LightGroup"

So the API returns the bulbs + the switch.

`curl -s -X GET "http://xxxxx:80/api/xxxxx/groups/4788"`
`{
  "action": {
    "bri": 128,
    "colormode": "hs",
    "ct": 0,
    "effect": "none",
    "hue": 0,
    "on": false,
    "sat": 128,
    "scene": null,
    "xy": [
      0,
      0
    ]
  },
  "devicemembership": [
    "3"
  ],
  "etag": "xxxxxxxxxxxxxxxx",
  "id": "4788",
  "lights": [
    "3"
  ],
  "name": "RWL021 5",
  "scenes": [],
  "state": {
    "all_on": false,
    "any_on": false
  },
  "type": "LightGroup"
}
`

I known that with Hue Bridge API it was not present. Maybe because with Hue, the type for group of bulbs is Room ?

- Is there a way to exclude from a GET on /group result that have a devicemembership not null ?
- Or is there a way to change the type of group to "Room", with API or direct into the database ? Will it break the webpage ?

Thank you

manup, 2019-02-11:
Technically it is just a group of lights, which happens to be controlled by the Dimmer Switch. Does this cause any problems?

If your application doesn't need these groups you can filter them out based on the `devicemembership` array length.

Korhm, 2019-02-11:
I understand that technicaly it's a group of lights controlled by a Dimmer Switch. But for me it's a "duplicate" of the classic group of lights created on phoscon app
Maybe I miss something to understand it properly.

I'll manage it on application side.

manup, 2019-02-11:
Not a duplicate, but the group is special in one way, that is it can be controlled by the gateway and the Dimmer Switch directly. So if you put light in the group the switch will control them (with basic function on/off and dimming) even when the gateway is not powered.

Korhm, 2019-02-11:
Ok, I understand now. Thank you

lbschenkel, 2019-02-13:
@manup: I have a follow up question. I don't have this Philips switch, but I do have IKEA remotes and sensors that have their own groups as well. I would like to understand a bit better how it works at the Zigbee layer.

My (possibly flawed) understanding is that these devices broadcast commands to a "built-in group", is that how it works? This "built-in group" that these devices broadcast to, is it randomized on each device reset? Otherwise, when the device joins a network, how does it pick a group id that is not already being used, potentially sending commands to random devices?

And how does deCONZ figure out that these groups exist? Does deCONZ query the devices (are there commands for that?) or does it just listen to the network, and once it sees commands sent to a group that it does not recognize, then it automatically creates it (I mean in the REST layer) on the spot? Does that mean that deCONZ can only know such a group exists after the device has sent at least one command to it?

And to finalize: are there any devices in which you can control the "built in group" in a way that you can set up two different devices (let's say remotes) to be sending commands to the same group?

closed 2019-06-20 by stale[bot]